luni64/TeensyStep #2: Get Current Position in Steps

Jerware, 2017-09-27:
I don't see a way to get the current position of the steppers. Is this possible?

luni64, 2017-09-27:
You get the current position by calling the getPosition() member of the stepper class. Here a short snippet:

```c++
ma.setTargetAbs(-15000);  // 15000 steps ccw
mb.setTargetAbs(5000);    // 5000 steps cw
controller.moveAsync(ma, mb);

// display positions while moving
 while (controller.isRunning())  
 {
      int posA = ma.getPosition();
      int posB = mb.getPosition();
      Serial.printf("motor a: %d | motor b: %d\n",posA, posB);
      delay(200);
 }
 // display final position 
 int posA = ma.getPosition();
 int posB = mb.getPosition();
 Serial.printf("motor a: %d | motor b: %d\n", posA, posB);
```
Output:
```
motor a: -1 | motor b: 0
motor a: -287 | motor b: 96
motor a: -687 | motor b: 229
motor a: -1087 | motor b: 362
motor a: -1487 | motor b: 496
motor a: -1887 | motor b: 629
motor a: -2287 | motor b: 762
motor a: -2687 | motor b: 896
motor a: -3087 | motor b: 1029
motor a: -3487 | motor b: 1162
motor a: -3888 | motor b: 1296
motor a: -4288 | motor b: 1429
motor a: -4688 | motor b: 1563
motor a: -5088 | motor b: 1696
motor a: -5488 | motor b: 1829
motor a: -5888 | motor b: 1963
motor a: -6288 | motor b: 2096
motor a: -6688 | motor b: 2229
motor a: -7088 | motor b: 2363
motor a: -7488 | motor b: 2496
motor a: -7888 | motor b: 2629
motor a: -8288 | motor b: 2763
motor a: -8688 | motor b: 2896
motor a: -9089 | motor b: 3030
motor a: -9489 | motor b: 3163
motor a: -9889 | motor b: 3296
motor a: -10289 | motor b: 3430
motor a: -10689 | motor b: 3563
motor a: -11089 | motor b: 3696
motor a: -11489 | motor b: 3830
motor a: -11889 | motor b: 3963
motor a: -12289 | motor b: 4096
motor a: -12689 | motor b: 4230
motor a: -13089 | motor b: 4363
motor a: -13489 | motor b: 4496
motor a: -13889 | motor b: 4630
motor a: -14290 | motor b: 4763
motor a: -14690 | motor b: 4897
motor a: -15000 | motor b: 5000
```

Jerware, 2017-09-28:
Whoa, how did I miss that. Thank you!

closed 2017-11-15 by luni64