luni64/TeensyStep #4: Issue with Teensy 3.2

JohnPunnett, 2017-10-25:
Hi Luni64, I am currently trying to implement your TeesnyStep Library on a Teensy 3.2 with 2 X Steppers driven by DRV8825 boards. I am trying to run the simple example & keep getting lots of errors. I am not too experienced with Arduino etc. so pls forgive my ignorance if I am making a silly mistake. Can you help.  Many thanks,  John

[new 1.txt](https://github.com/luni64/TeensyStep/files/1414555/new.1.txt)

steinerlein, 2017-10-25:
Hey John, you might be better off trying to get support in the [PJRC Forum](https://forum.pjrc.com/)
This place here is intended for technical bugs in the library. Your issue seems to be another unrelated one. Have you installed teensyduino? You can find it here:
https://www.pjrc.com/teensy/teensyduino.html

JohnPunnett, 2017-10-25:
Hi Luni64 - many thks for that. Yes teensyduino is all installed & I have run/ compiled several progs before just seem to be having issues with this. I will touchbase with PJRC & see what they suggest - thks again & really looking forward to getting this working.  John

luni64, 2017-10-25:
The error message suggests that there is something wrong with your build environment. Are you sure that you selected the right board in the tools menu? Can you try to install / reinstall the latest Teensyduino?
Here a link to a similar problem: [https://forum.pjrc.com/threads/46990-TeensyStep-build-errors?p=156488&viewfull=1#post156488](https://forum.pjrc.com/threads/46990-TeensyStep-build-errors?p=156488&viewfull=1#post156488). It was solved by reinstalling teensyduino,

JohnPunnett, 2017-10-25:
Hi Luni64 - Excellent you solved it - I have now updated to latest Arduino IDE & TeensyDuino & now all works as expected - many thanks for all support - John

closed 2017-10-25 by JohnPunnett

luni64, 2017-10-25:
You're welcome. Have fun with the library and don't hesitate to ask if you need help