luni64/TeensyStep #5: moveAsync for an array of steppers

eshkrab, 2017-10-27:
Hello! The readme says that a controller can control up to ten motors but I'm confused about the usage of the function. I thought, if motors is an array of stepper pointers, the way to call the function would be controller.moveAsync(motors); but that causes the Teensy to crash. What's the correct way to do this? 
Thanks!

luni64, 2017-10-27:
@eshkrab
Can you post your code? I can have a look at it tomorrow

luni64, 2017-10-28:
I added an [example](https://github.com/luni64/TeensyStep/tree/master/examples/StepperArray) which shows how to use arrays of pointers to steppers, hope that helps.

eshkrab, 2017-10-28:
thank you! I'm not obviously doing something different so perhaps it's crashing for another reason, but the example helps!

closed 2017-10-28 by eshkrab