luni64/TeensyStep #6: Limit Switch Stopping

kenbo017, 2017-11-02:
I'm trying to figure out a way to handle stopping at limit switches.  I have a minimum limit switch and a maximum limit switch for each motor.  It appears that the emergency stop function cannot be called for a particular motor.  Since I will have multiple motors (up to 13 motors) moving at one time various ones can hit either limits.

A few scenarios would be:

When turning on the machine and wanting to "home" where we go to min fairly slowly and stop when it hits the limit switch.  Then goes forward until it is off the limit switch and then zeros the counter.  This way all counters will start at the exact point when the motor comes off the limit switch.  It is possible that when the machine turns on the position is already on the min limit.  In that case we would want to have it move forward until it comes just off the limit and set the count to zero.

When multiple motors are moving and one of them hits a limit.  That motor needs to stop right away while others can happily go to their position.  Various motors could hit limits at different times while others need to continue.

I believe I should have the limit switches trigger ISR's when they hit the limit.  In the case of hitting the min limit the ISR could include the move forward to get just off the limit.

What are your suggestions?

luni64, 2017-11-03:
> When turning on the machine and wanting to "home" where we go to min fairly slowly and stop when it hits the limit switch. Then goes forward until it is off the limit switch and then zeros the [...]

Perfect, that's the usual way to home steppers. If you need very high precision you can improve the case when the position is already at the min limit. -> move out of the switch some way and simply start the normal homing sequence from that point. This ensures that homing is always done under same conditions regardless of the starting position. 

>When turning on the machine and wanting to "home" where we go to min fairly slowly and stop when it hits the limit switch.

It is a good idea to place the switches such that the motor can move a couple of mm past the switching point. In that case you can move much quicker to the home position and do a regular stop when you reach it. Moving out of the sensor should be done slowly of course.  I usually prefer optical switches over mechanical for this purpose. 

>When multiple motors are moving and one of them hits a limit. That motor needs to stop right away while others can happily go to their position. Various motors could hit limits at different times while others need to continue.

Of course I don't know your setup, and much of the following depends on what you want to achieve. Here the usual way to handle limit switches:  
Normally you exactly know the position of the motor by its counter value. So,  if you hit a limit switch without expecting it you either have a firmware bug or your  motor somehow lost steps. In any case this is a major malfunction and I would do an emergency stop of the complete system. If you use the limit switches in that way (homing and emergency only)  you can simply wire them in series. (one series for min and one series for max) which makes the design much simpler and you only need two inputs for your 26 limit switches. 

> In the case of hitting the min limit the ISR could include the move forward to get just off the limit.

I wouldn't do that. Homing is only necessary after starting the system.  As mentioned, hitting a limit switch when you did not expect it. is probably a major system malfunction.

luni64, 2017-11-03:
>Since I will have multiple motors (up to 13 motors) 
BTW: What kind of project is this? Sounds interesting. Do you have any links, videos etc?

kenbo017, 2017-11-03:
Hi Lutz,

Thank you so much for your reply!

I am developing a fitness machine.  It is very advanced and completely
controlled by a computer.  I had built a working prototype and then hired
an engineering company to take my prototype and make complete commercial
electronics.  They convinced me that they should design complete special
hardware.  I paid them $200K in development costs and about three months
ago (after 3+ years of waiting for them) they came to me and said they
couldn't do it.  I'm out the money and don't have more to spend.  Hopefully
someday I can sue them but right now I need to try to move forward.  The
hardware I used in my original prototype is obsolete and I need to build a
new electronics system that I can ultimately get made into a commercially
viable solution.  I'm far from an expert in the electronics.  I have taught
myself enough about coding and electronics to get the original working but
starting over is not easy.  I started with the Arduino Mega to control the
motors using accelstepper.  As I'm not an expert in coding dealing with
acceleration on the motors is something I couldn't do so I'm stuck with
what I can piece together from learning on the Internet.  I discovered that
I can't get fast enough pps with accelstepper when I have multiple motors
so I got a bunch of Mega boards thinking I could daisy chain them.  Then I
got some new motors that are servo and I need much higher pps and found
your solution.  I just bought a teensy 3.6 and it seems like your solution
will work but I ran into the issue of the limit switches.  Any and all help
would greatly be appreciated!!!

I understand your thoughts on the limit switches and having all motors stop
if you hit one because any problem (which you would be having if you hit
one when you didn't anticipate hitting one) is something that all motors
need to stop.  That being said, I'm not sure how to write the code for the
homing.  In the initialization (when we setup a machine for the first time
or make mechanical adjustments where the machine needs to be "calibrated")
we will want to run each motor to setup its min and max limits.  We'll want
to save the max limit to the database so we can know it at anytime - even
if we reboot the system so it is lost we can call to the database of
machine settings.  Once that is done, then we will have a case where the
power goes out or they restart the machine.  At that point, we'll want to
do a homing procedure of any number of motors (could be all or one).  If it
is multiple motors we'll want them to run at their own "homing speeds"
simultaneously, but asynchronously, until all motors are homed and we reset
each counter as it finishes the homing procedure.  There may be a case
where we have a controller issue so we need to stop, notify the system of
the error, reset the controller and then just home that motor again.  As a
note, we don't want to automatically home the machine on boot.  We will
want to get a "ready" message to know all is okay after bootup.  Then,
after we send a message to the user and they respond it is okay to move,
then we'll home.  So we'll need to get a "home" command.

Are you an expert with this kind of embedded control?  I wish I were better
at it.  I actually have some potential customers but I need to be able to
get this new hardware working.  Below is a photo of the machine.  You'll
see it is quite big and there are a lot of moving axis (13).  You'll see
the ball screws on the machine.  I would welcome any thoughts you have.  I
also have to get the teensy/Arduino to control my magnetic particle brake
while taking into consideration the position (optical encoder) and the
force (measuring a load cell).  I need to read the load cell/encoder to
determine direction, position, and load so I can adjust how much resistance
the brake is giving at any time.  This will need a PID loop.  Again, I'm
not an expert at this.  I have also written a VB application for my
prototype that I will need to adapt to this new Arduino/Teensy hardware.
One thing that is difficult is that I need to record precisely that
feedback from the sensors (encoder, load cell, motors currents, etc.) and
transmit them to the embedded computer that will be talking to the
Android/Teensy board.  This will then all get saved to a local server and
the cloud/Azure.  Getting very fast communication between the boards and
the embedded computer is also proving difficult for me.

I will look forward to your thoughts!

Regards,

KS

Ken Schneider
Schneider Sales and Marketing
312 West Hillcrest Blvd.
Inglewood, CA 90301
ph: 310-680-9544
cell: 310-722-8827
fax: 310-861-5798
e-mail: kens@ssmcentral.com

CONFIDENTIALITY NOTICE. This electronic mail transmission may contain
privileged, confidential and/or protected information and is intended only
for the review of the party to whom it is addressed. Any unauthorized use
or disclosure of the information contained herein may be a violation of
federal law. If you have received this transmission in error, please
immediately return it to the sender, delete it and destroy it without
reading it. Unintended transmission shall not constitute the waiver of the
attorney-client or any other privilege.

luni64, 2017-11-04:
Sent you an email

closed 2017-11-04 by luni64