luni64/TeensyStep #8: stutter motion during continuous updates

egradman, 2017-11-09:
I'm not sure if this is an issue with the library or with how I'm trying to use it.

I have a desktop program streams absolute motor positions for six motors from a timeline-based curve editor over serial to a Teensy.  When each serial line arrives, I call moveAsync on all six motors at once.  The motors obey, but they stutter.

I maxed out the configured velocity, pull-in acceleration, and acceleration because the curves I'm sending from the desktop already limit the max velocity and acceleration.

My coworkers insist that maxing out these values constants means I'm doing something very wrong.  Are they right?

luni64, 2017-11-10:
Can it be that you are sending a new moveAsync while the controller handles the current one? For async movements it is important to check that the controller finished the current movement before you send a new one. Typically you would do something like
```c++
while(controller.isRunning()){              // wait until the movement is finished
    delay(10);                     
} 
```
before starting a new movement.

egradman, 2017-11-13:
I am definitely sending a new moveAsync while the controller handles the current one; I expect my robot to be in continuous motion.

My expectation was that if I send a second moveAsync, and the robot is already moving in the correct direction then teensystep would just update the final location without starting a new ramp.

Is there another way to accomplish this?

luni64, 2017-11-13:
You can't do that with TeensyStep. Did you have a look at [grbl](https://github.com/grbl/grbl)? It takes standard g-code and has an input buffer, movement planner etc. Might be a better solution for your problem.  I once did a board where a Teensy fed an integrated "Uno" (marked red)  which had grbl running. Works perfectly.

[image]

egradman, 2017-11-13:
haha, I actually implemented this already using GRBL but I was looking for an excuse to try TeensyStep so I switched over before I ironed the bugs out of my implementation.  I guess I'll switch back.

I'll find another opportunity to use TeensyStep.  Thanks for writing it, and thanks for the assistance.

closed 2017-11-13 by egradman