luni64/TeensyStep #9: Can this library run in speed near or even below 1Hz ?

Hooch180, 2017-11-19:
Hello.
Can this library run motor in about 1Hz (1 step per second) or even less?

I need it to move axis in lapping machine. Old driver died and with 1/4 step division I need to run it at about 1.5 steps per second. But slower speeds would be nice too. Like 1 step per 2 seconds.

I would like to use library as I also need fast movements without loosing track of steps.

luni64, 2017-11-19:
Yes, that works. Limit is the maximum period of the interval timer which (depending on board type and bus frequency) is about 90s.

luni64, 2017-11-19:
Sorry, that was wrong information. Unfortunately the stepper class expects speed values as unsigned int, therefore the slowest speed you can set is 1 Hz. Technically this is not necessary but you have to change the code to use it. If you just want to test it, the relevant code is in stepcontrol.h. 

[image]

In line 308 I calculate the timer reload value cMax. If you change this line to `cMax = F_BUS / 1.5; ` or any other frequency you need it should work. If it works and you need variable speed you can change the code so that it accepts floats instead of the uint32_t values for the speed.

closed 2017-11-23 by luni64

Hooch180, 2017-11-23:
Thank you. I'll provide pull request when I will have working and tested code.

luni64, 2017-11-23:
Perfect. 
Thanks