luni64/TeensyStep #10: how to change the step pulseWidth

Steamerzone, 2017-12-06:
Hello,

The driver I use needs a longer pulseWidth (optocoupler), I changed the value in StepControl.h, but is there a function call to change it and leaving the library unchanged?

btw Thanks for the great library, I've tested a 4.567khz step freq, and the result on my freq counter was 4.566khz! 

Emile

luni64, 2017-12-06:
Hi Emile, 

StepControl has two template parameters to set the pulswidth and the acceleration update interval (both in microseconds), The pulsewidth defaults to 5µs, the acceleration update interval to 5000µs. 

If you want to use a pulsewidth of 10µs and an acceleration update every 8ms you can simply write

```
StepControl<10, 8000> controller  // pulsewidth 10µs, acceleration update 5ms
```
Lutz

Steamerzone, 2017-12-06:
Hi Lutz,

Shouldn't the following rotate the stepper clockwise and counterclockwise:

  controller.stop();
  stepper3.setMaxSpeed(5000); 
  controller.rotateAsync(stepper3);
  delay(3000);
  controller.stop();
  delay(3000);
  stepper3.setMaxSpeed(-5000); 
  controller.rotateAsync(stepper3);
  delay(3000);

Both are executed as clockwise, and if I change the first speed to -5000 and the second to 5000 both execute as counterclockwise.

Emile

luni64, 2017-12-06:
Emile, 

you are right, this definitely is a bug. It used to work nicely, I must have messed it up with one of the latest updates. I'll fix the bug as soon as possible. Unfortunately I'll be quite busy for the next two weeks. 

Internally the rotate commands simply translate to normal move commands with very large positive/negative targets.  You can achieve the same effect by setting those targets manually.

-> The following workaround should do the trick while I look for a bugfix. 

```
void setup()
{
  stepper3
  .setAcceleration(50000)
  .setMaxSpeed(5000); 

  stepper3.setPosition(0);  
  stepper3.setTargetAbs(INT_MAX);
  controller.moveAsync(stepper3);  
  delay(3000);
  controller.stop();

  delay(3000);

  stepper3.setPosition(0);
  stepper3.setTargetAbs(INT_MIN);
  controller.moveAsync(stepper3);
  delay(3000);
  controller.stop();
}
```

luni64, 2018-01-18:
fixed

closed 2018-01-18 by luni64