luni64/TeensyStep #13: Continuous Rotation - cant change direction

bluepuppy, 2018-01-17:
Great job on this library!  I'm having a problem though..   changing motor.setMaxSpeed to negative or positive is not changing the direction of the motor.

Here's the code I'm converting from AccelStepper to TeensyStep..

I've only programmed the Blue and Yellow buttons so far.. pressing Blue spins the motor CCW.. then pressing Yellow also spins the motor CCW.. 

`#include <Adafruit_GFX.h>       // Include core graphics library for the display
#include <Adafruit_SSD1306.h>   // Include Adafruit_SSD1306 library to drive the display
#include <Wire.h>
#include <ResponsiveAnalogRead.h>
#include <AnalogMultiButton.h>
#include <StepControl.h>

//===============================
//button setup
//===============================

const int BUTTONS_PIN = A0;       // define the analog pin for buttons
const int BUTTONS_TOTAL = 4;      // button count
const int BUTTONS_VALUES[BUTTONS_TOTAL] = {12, 344, 512,614}; // assign button codes
// assign button names
const int BUTTON_BLUE = 0;
const int BUTTON_YELLOW = 1;
const int BUTTON_GREEN = 2;
const int BUTTON_RED = 3;
AnalogMultiButton buttons(BUTTONS_PIN, BUTTONS_TOTAL, BUTTONS_VALUES);

//===============================
//Pot setup
//===============================

const int knob = A9;
ResponsiveAnalogRead analog(knob,true);
int potval = 0;

Adafruit_SSD1306 display;       // Create display

// Define stepper & pins. 
Stepper stepper(4,3);           // Define stepper & pins: motorName(step pin, direction pin)
StepControl<> controller;       // Construct Controller object

int led = 13;               // Built-in LED pin
int val = 0;
int StepSize = 500;
int MaxSpeed = 30000;
int MaxAccel = 200000;
int long SliderPosIn = stepper.getPosition();
int long SliderPosOut = stepper.getPosition();
int long SliderPosHome = stepper.getPosition();
int long MotorPos = stepper.getPosition();

void setup()
{
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // Initialize display with the I2C address of 0x3C
  Wire.setClock(1000000);                     // Teensy Steroid injection for the OLED
  display.clearDisplay();  // Clear the buffer
  display.setTextColor(WHITE);  // Set color of the text
  display.setRotation(0);  // Set orientation. Goes from 0, 1, 2 or 3
  display.setTextWrap(false);  // By default, long lines of text are set to automatically “wrap” back to the leftmost column.
  display.dim(0);  //Set brightness (0 is maximun and 1 is a little dim) 
  pinMode(led, OUTPUT);
  stepper.setMaxSpeed(MaxSpeed);
  stepper.setAcceleration(MaxAccel);
  //Serial.begin(9600); 
}

void loop()
{
    analog.update();
    buttons.update();
    potval = analog.getValue();                     // Pull Analog value
    
    if(buttons.isPressedAfter(BUTTON_BLUE,500))    // Hold Down to move
    {
      stepper.setMaxSpeed(-30000);
      controller.rotateAsync(stepper);
      while (controller.isRunning() && analogRead(BUTTONS_PIN)!=1023)
      {
      display.clearDisplay();
      display.setFont();
      display.setCursor(1,5);
      display.println("Moving LEFT");
      MotorPos=stepper.getPosition();
      display.println(MotorPos);
      display.display();
      }
      controller.stopAsync();
    }
    
 
    if(buttons.onReleaseBefore(BUTTON_BLUE,500))    // Quick click set LEFT 
    {
       SliderPosIn = MotorPos;
       display.clearDisplay();
       display.setFont();
       display.setCursor(1,5);
       display.println("IN Point Set");
       display.println(SliderPosIn);
       display.display();
       FlashLed(1);

    }
//
    if(buttons.isPressedAfter(BUTTON_YELLOW,500))     // Hold Down to move
      {
      stepper.setMaxSpeed(30000);
      controller.rotateAsync(stepper);
      while (controller.isRunning() && analogRead(BUTTONS_PIN)!=1023)
      {
        display.clearDisplay();
        display.setFont();
        display.setCursor(1,5);
        display.println("Moving RIGHT");
        MotorPos=stepper.getPosition();
        display.println(MotorPos);
        display.display();
      }
      controller.stopAsync(); }
//
//    if(buttons.onReleaseBefore(BUTTON_YELLOW,500))    // Quick click to set 
//    {
//     SliderPosOut = MotorPos;
//     Serial.println("Right Point Set");
//     FlashLed(1);
//    }
//
//    if(buttons.isPressed(BUTTON_RED))          // Home Motor
//    {
//     Serial.println("Homing");
//     stepper.runToNewPosition(SliderPosHome);
//     MotorPos=SliderPosHome;
//    }
//
//    if(buttons.onRelease(BUTTON_YELLOW))          // Run Motor IN to OUT
//    {
//     Serial.print("Running IN to OUT");
//     stepper.runToNewPosition(SliderPosIn);
//     delay(500);
//     stepper.setMaxSpeed(10000);
//     stepper.setAcceleration(5000);
//     FlashLed(2);
//     stepper.runToNewPosition(SliderPosOut);
//     Serial.println("....Done");
//     stepper.setMaxSpeed(MaxSpeed);
//     stepper.setAcceleration(MaxAccel);
//     MotorPos=SliderPosHome;
//     FlashLed(3);
//
//     
//    }

//    if(analog.hasChanged())
//    {
//      StepSize = map(potval,2,999,5,1000);   // remap pot to StepSize
////      stepper.runToNewPosition(MotorPos);
//      FlashLed(1);
//    }

}

void FlashLed(int blinks)
{
  digitalWrite(led,LOW);
  for (int x = 1; x<= blinks; x++) {
    digitalWrite(led,HIGH);
    delay(100);
    digitalWrite(led,LOW);
    delay(100);
  }
  

}
`

luni64, 2018-01-17:
Sorry for that. I'm afraid that's a known bug. I'll fix that as soon as I find some time but that may take a couple of weeks...  Fortunately there is a very simple workaround: 

Instead of writing 
```c++
if(buttons.isPressedAfter(BUTTON_BLUE,500))    // Hold Down to move
{
    stepper.setMaxSpeed(-30000);
    controller.rotateAsync(stepper);
    while (controller.isRunning() && analogRead(BUTTONS_PIN)!=1023)
    {
      //....
    }
    controller.stopAsync();
}
```
you could do 

```c++
if(buttons.isPressedAfter(BUTTON_BLUE,500))    // Hold Down to move
{
    stepper.setTargetAbs(INT_MAX);   // <cw rotoation---------
    // stepper.setTargetAbs(INT_MIN);   // <ccw rotation---------
    controller.moveAsync(stepper);     // <--------- 
    while (controller.isRunning() && analogRead(BUTTONS_PIN)!=1023)
    {
      //....
    }
    controller.stopAsync();
}
```
Technically this is doing the same as the rotate command is supposed to. Make sure to include <limits.h> for INT_MIN / INT_MAX or just use some other large values. 

Let me know if that works for you.

bluepuppy, 2018-01-17:
Lutz,

Thanks for getting in touch.. unfortunately, that's not working.  I get
exactly the same behaviour.

If I press the Yellow button, to turn CW.. then press the Blue.. the motor
continues to turn CW.. does so until I reset the teensy..   similarly, if I
press the Blue button first, for CCW.. then the Yellow.. it continues to
turn CCW.

Apparently, whatever direction I select first, subsequent button presses
continues the motor in that direction.. regardless of which button I press.

Here's my code, with your suggested changes:

    if(buttons.isPressedAfter(BUTTON_BLUE,500))    // Hold Down to move
    {
      stepper.setMaxSpeed(10000);
      stepper.setAcceleration(150000);
      stepper.setTargetAbs(INT_MIN);
      controller.moveAsync(stepper);
      while (controller.isRunning() && analogRead(BUTTONS_PIN)!=1023)
      {
      display.clearDisplay();
      display.setFont();
      display.setCursor(1,5);
      display.println("Moving LEFT");
      MotorPos=stepper.getPosition();
      display.println(MotorPos);
      display.display();
      }
      controller.stop();
    }

    if(buttons.onReleaseBefore(BUTTON_BLUE,500))    // Quick click set LEFT
    {
       SliderPosIn = MotorPos;
       display.clearDisplay();
       display.setFont();
       display.setCursor(1,5);
       display.println("IN Point Set");
       display.println(SliderPosIn);
       display.display();
       FlashLed(1);

    }
//
    if(buttons.isPressedAfter(BUTTON_YELLOW,500))     // Hold Down to move
      {
      stepper.setMaxSpeed(10000);
      stepper.setAcceleration(150000);
      stepper.setTargetAbs(INT_MAX);
      controller.moveAsync(stepper);
      while (controller.isRunning() && analogRead(BUTTONS_PIN)!=1023)
      {
        display.clearDisplay();
        display.setFont();
        display.setCursor(1,5);
        display.println("Moving RIGHT");
        MotorPos=stepper.getPosition();
        display.println(MotorPos);
        display.display();
      }
      controller.stopAsync(); }

I really hope you can help.. TeensyStep is like a Holy Grail for me,
solving an issue with display updating while moving a motor that I've been
struggling with for weeks.

Thanks again,

Adrian

-- 

[image: Liquid Light Digital] <http://www.liquidlightdigital.com>

Adrian Lopez / CEO.Director.Producer
vfxwolf@gmail.com / JAM: 876.527.6898 USA: 954.621.3775

Liquid Light Digital
876.978.6418
2a Hillcrest Ave.Kingston 6, Jamaica
www.liquidlightdigital.com

[image: Twitter]  <https://htmlsig.com/t/000001C36H4Q> [image: LinkedIn]
<https://htmlsig.com/t/000001C49NQM> [image: Skype]
<https://htmlsig.com/t/000001C03QW6> [image: Youtube]
<https://htmlsig.com/t/000001C3VY0P> [image: Vimeo]
<https://htmlsig.com/t/000001C93NWY>

luni64, 2018-01-17:
Don't worry, that will definitely work.   I'll have a closer look at it this evening.  Can you try to use +/- 500000 instead of the INT_MAX / INT_MIN meanwhile?

bluepuppy, 2018-01-17:
> Now THAT worked!   Strange, why wouldnt INT_MIN and INT_MAX work?

The only issue now is.. how do I get TeensyStep to take the motor to a
specific position if I cant use .setTargetAbs() without an absurdly large
-/+number?

-- 

[image: Liquid Light Digital] <http://www.liquidlightdigital.com>

Adrian Lopez / CEO.Director.Producer
vfxwolf@gmail.com / JAM: 876.527.6898 USA: 954.621.3775

Liquid Light Digital
876.978.6418
2a Hillcrest Ave.Kingston 6, Jamaica
www.liquidlightdigital.com

[image: Twitter]  <https://htmlsig.com/t/000001C36H4Q> [image: LinkedIn]
<https://htmlsig.com/t/000001C49NQM> [image: Skype]
<https://htmlsig.com/t/000001C03QW6> [image: Youtube]
<https://htmlsig.com/t/000001C3VY0P> [image: Vimeo]
<https://htmlsig.com/t/000001C93NWY>

luni64, 2018-01-17:
> Now THAT worked!   Strange, why wouldnt INT_MIN and INT_MAX work?

Don't know yet why INT_MIN / MAX didn't work but they are borderline of course. I'll check this. 

> The only issue now is.. how do I get TeensyStep to take the motor to a
> specific position if I cant use .setTargetAbs() without an absurdly large
> -/+number?

No, no, the large number was only meant to simulate a continuous move. Of course you can use "normal" small numbers for positioning the motor. That is the normal use case and should work without an issue.

bluepuppy, 2018-01-17:
Lutz, setTargetAbs() does indeed work with the smaller numbers.. however,
there seems to be a miscount somewhere.. I'm testing a simple move from 0
to 150000 steps.. printing out the final end position with
stepper.getPosition().. and I consistently get a final position of 149997.
 What could be causing this?

The motor is moving very slowly.. so I doubt its missing steps.

Another thing..  part of my project calls for the motor movement to be
controlled joystick.. the greater the deflection, the higher the speed.. is
there a function for changing the speed of a running motor in a continuous
move?

Thanks..

-- 

[image: Liquid Light Digital] <http://www.liquidlightdigital.com>

Adrian Lopez / CEO.Director.Producer
vfxwolf@gmail.com / JAM: 876.527.6898 USA: 954.621.3775

Liquid Light Digital
876.978.6418
2a Hillcrest Ave.Kingston 6, Jamaica
www.liquidlightdigital.com

[image: Twitter]  <https://htmlsig.com/t/000001C36H4Q> [image: LinkedIn]
<https://htmlsig.com/t/000001C49NQM> [image: Skype]
<https://htmlsig.com/t/000001C03QW6> [image: Youtube]
<https://htmlsig.com/t/000001C3VY0P> [image: Vimeo]
<https://htmlsig.com/t/000001C93NWY>

luni64, 2018-01-17:
> Lutz, setTargetAbs() does indeed work with the smaller numbers.. however,
> there seems to be a miscount somewhere.. I'm testing a simple move from 0
> to 150000 steps.. printing out the final end [...]

Are you sure that you read out the position AFTER the motor stopped? Note that StopAsync() will only start the deceleration and return before the motor actually stopped. If you want to make sure the motor stopped you can use Stop() instead. 

> The motor is moving very slowly.. so I doubt its missing steps.

The internal step counter would not know that anyway. 

> Another thing..  part of my project calls for the motor movement to be
> controlled joystick.. the greater the deflection, the higher the speed.. is
> there a function for changing the speed of a running motor in a continuous
> move?

No, this is not implemented. Would be a bit tricky since you also need to take the acceleration ramp into account.

luni64, 2018-01-17:
Regarding the getPosition() here ["https://github.com/luni64/TeensyStep/issues/2"] (https://github.com/luni64/TeensyStep/issues/2) you find some example code showing how to read out the current position.

bluepuppy, 2018-01-17:
> TeensyStep/issues/2"] (#2 <https://github.com/luni64/TeensyStep/issues/2>)

Lutz,

There's definitely something strange going on with the step counts..  I've
modified the code to stop() rather than stopAsync().. as you suggested.
I've also printed out the starting position of the motor when the sketch
starts.. and again when the motor stops and after a 2 second delay to be
SURE its stopped.

I get a starting position of 0, as expected.. but a 150003 ending point in
the positive swing and -149997 in the negative swing..

Seems like a +3 step count shift towards the positive...

Here's the code..

// Bounce Motor Display tester for TeensyStep
//
// Make a single stepper bounce from one limit to another REPS many times,
then print the duration to complete
//
//

#include <Adafruit_GFX.h>       // Include core graphics library for the
display
#include <Adafruit_SSD1306.h>   // Include Adafruit_SSD1306 library to
drive the display
#include <StepControl.h>
#include <Wire.h>
#include <Encoder.h>

Adafruit_SSD1306 display;       // Create display
Encoder myEnc(10, 11);
Stepper stepper(4,3);           // Define stepper & pins: motorName(step
pin, direction pin)
StepControl<> controller;       // Construct Controller object

  int32_t elapsed    = 0;
  int32_t starttime  = 0;
  int32_t distance   = 0;
  int32_t targetPos  = 50000;
  int     reps       = 0;
  const int32_t CW   = -1000000;
  const int32_t CCW   = 1000000;
  int     encRead;
  int     encSpeed;

void setup()
{
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // Initialize display with
the I2C address of 0x3C
  Wire.setClock(1000000);                     // Teensy Steroid injection
for the OLED
  display.clearDisplay();  // Clear the buffer
  display.setTextColor(WHITE);  // Set color of the text
  display.setRotation(0);  // Set orientation. Goes from 0, 1, 2 or 3
  display.setTextWrap(false);  // By default, long lines of text are set to
automatically “wrap” back to the leftmost column.
  display.dim(0);  //Set brightness (0 is maximun and 1 is a little dim)

// Motor Setup

  stepper.setMaxSpeed(15000);
  stepper.setAcceleration(150000);
//  stepper.setTargetAbs(CCW);

}

void loop()
{

    display.clearDisplay();
    display.setFont();
    display.setCursor(1,5);
    display.print("motor pos START: ");
    display.print(stepper.getPosition());
    display.setCursor(1,15);
    display.display();                          //  Print starting position
    delay(2000);
    starttime  = millis();
    do
    {
      stepper.setTargetAbs(-(-150000));
      controller.moveAsync(stepper);
      while (controller.isRunning())
        {
              display.clearDisplay();
              display.setFont();
              display.setCursor(1,5);
              display.print("motor pos: ");
              display.print(stepper.getPosition());
              display.setCursor(1,15);
              display.print("Reps: ");
              display.print(reps);
              display.setCursor(1,25);
              display.print("Speed: ");
              display.display();
        }
    controller.stop();
    reps ++;
    }while (reps < 2);

    display.clearDisplay();
    display.display();

    delay(2000);                            // pause to ensure motor is at
rest

    int32_t endtime = millis();
    elapsed = (endtime - starttime)/1000;

    display.clearDisplay();
    display.setFont();
    display.setCursor(1,5);
    display.print("Time to complete:");
    display.setCursor(1,15);
    display.print(elapsed);
    display.setCursor(1,25);
    display.print("current motor position");
    display.setCursor(10,35);
    display.print(stepper.getPosition());        // print position at stop

    display.display();
    delay (30000);            // Hold execution for 30 sec. to note time

}

-- 

[image: Liquid Light Digital] <http://www.liquidlightdigital.com>

Adrian Lopez / CEO.Director.Producer
vfxwolf@gmail.com / JAM: 876.527.6898 USA: 954.621.3775

Liquid Light Digital
876.978.6418
2a Hillcrest Ave.Kingston 6, Jamaica
www.liquidlightdigital.com

[image: Twitter]  <https://htmlsig.com/t/000001C36H4Q> [image: LinkedIn]
<https://htmlsig.com/t/000001C49NQM> [image: Skype]
<https://htmlsig.com/t/000001C03QW6> [image: Youtube]
<https://htmlsig.com/t/000001C3VY0P> [image: Vimeo]
<https://htmlsig.com/t/000001C93NWY>

luni64, 2018-01-17:
I removed all display related things from the code you posted above to check if TeensyStep is doing something wrong and end up with the following: 

```c++
#include "Arduino.h"
#include "TeensyStep\StepControl.h"

Stepper stepper(4, 3);           // Define stepper & pins: motorName(step
StepControl<> controller;       // Construct Controller object

int     reps = 0;

void setup()
{
    Serial.begin(0);
    while (!Serial);

    stepper.setMaxSpeed(15000);
    stepper.setAcceleration(150000);  
}

void loop()
{   
    do
    {
        Serial.println("Enter do while");
        stepper.setTargetAbs(150000);
        controller.moveAsync(stepper);
        while (controller.isRunning())
        {
            Serial.println(stepper.getPosition());
            delay(500);
        }
        //  controller.stop();
        Serial.println(stepper.getPosition());

        reps++;
    } while (reps < 2);

    delay(30000);            // Hold execution for 30 sec. to note time
}
```
1) Each time you enter the do/while loop you set the absolute target value to 150000 -> after the first loop the motor should not move any further since it already sits at 150000.  This is probably not what you wanted? 
2) The controller.stop() after the while(controller.isRunning()) is not needed. You will reach that line of code only if controller.isRunning() returns false, i.e. if the motor is already stopped. I commented it out. 

First let the code run with reps = 1; Here the result: 

[image]

I.e. exactly what you would expect. 

Now increase reps to 2 ->  I'd expect no further movement but I get: 
[image]

Setting the absolute target to the same value again should result in no movement but generates 3 steps.
**Which is definitely a bug.** and needs to be fixed. 
-----
If you can explain what you want to achieve with your code, I can try to do a working example for you. (But as mentioned above I can not change the speed during a movement)

bluepuppy, 2018-01-17:
Lutz,

Thanks for confirming that there does seem to be a step count bug, my
faulty coding logic notwithstanding.  As to what I want to achieve, this
was just a test of TeensyStep's accuracy - I'm ultimately coding a
programmable motion controlled camera rig, that will have 3 motors for the
pan, tilt and sliding axes.  The rig needs to be able to repeat motions
accurately down to the microstep, while updating the display in realtime.
Accelstepper was unable to do this.

The reason for asking about being able to update the motor speed while in
motion was because this rig has to programmed by a camera operator moving a
joystick to control the motion, which will be recorded by the system - and
can be played back as many times as needed.

I was wondering if I could do this with setTargetRel().. where the relative
amount is set by the joystick and updated in a loop..  would this work?  If
the Teensy recorded the motor position each millisecond, I could then
playback the values to the motor.

Thats the pseudocode of it all..

Is this something TeensyStep could handle?

-- 

[image: Liquid Light Digital] <http://www.liquidlightdigital.com>

Adrian Lopez / CEO.Director.Producer
vfxwolf@gmail.com / JAM: 876.527.6898 USA: 954.621.3775

Liquid Light Digital
876.978.6418
2a Hillcrest Ave.Kingston 6, Jamaica
www.liquidlightdigital.com

[image: Twitter]  <https://htmlsig.com/t/000001C36H4Q> [image: LinkedIn]
<https://htmlsig.com/t/000001C49NQM> [image: Skype]
<https://htmlsig.com/t/000001C03QW6> [image: Youtube]
<https://htmlsig.com/t/000001C3VY0P> [image: Vimeo]
<https://htmlsig.com/t/000001C93NWY>

luni64, 2018-01-17:
> Thanks for confirming that there does seem to be a step count bug, my
> faulty coding logic notwithstanding.  

You already found the second bug in 24h which is great. I'll hire you as software tester :-) 

> As to what I want to achieve, this
> was just a test of TeensyStep's accuracy - I'm ultimately coding a
> programmable motion controlled camera rig, that will have 3 motors for the
> pan, tilt and sliding [...]

Positioning a stepper motor accurately to one step is of course the core functionality of any stepper driver code. Well, at least if there are no bugs... TeensyStep was developed to achieve high step rates thus it is quite efficient and allows for other things done while the motors move. 

> The reason for asking about being able to update the motor speed while in
> motion was because this rig has to programmed by a camera operator moving a
> joystick to control the motion, which will be recorded by the system - and
> can be played back as many times as needed.

Playback of some movements would not be the problem but the motor will necessarily stop between each partial movement. TeensyStep as it is programmed can not update any movement parameter on the fly. 

> I was wondering if I could do this with setTargetRel().. where the relative
> amount is set by the joystick and updated in a loop..  would this work?  If
> the Teensy recorded the motor position each millisecond, I could then
> playback the values to the motor.

Same problem, the movement will stop between two partial movements. 

So TeensyStep is definitely not the right tool for your task. If you want we can continue talking about possible solutions per mail. You can reach me at lutz.niggl@lunoptics.com

luni64, 2018-01-18:
Fixed with #14

closed 2018-01-18 by luni64