luni64/TeensyStep #19: How to reset counter?

jeffmann, 2018-04-23:
Hi - I'm looking for a way to reset the position counter to zero after homing, similar to AccelStepper command setCurrentPosition(). Is it possible?

I'm using the TMC2130 SilentStepStick for sensorless homing. When the motor hits a physical end stop it sets a pin, which should generate an interrupt on the Teensy, and do an emergency (or very fast) stop. Then I need to set that as the zero position.

Thanks...

luni64, 2018-04-23:
Assuming you have a stepper m1 you can simply use 
```c++
m1.setPosition(0);    // to set the counter of m1 to 0. 
m1.setPosition(1234); // to set the counter of m1 to 1234. 
```

jeffmann, 2018-04-23:
Oh, great, thanks. I was searching the readme documentation but didn't find it - actually there's a typo:
`motor_1.setPostion(pos);             // set a new value of the step counter `
"setPostion" is missing an "i". My fault for not reading more carefully though...

closed 2018-04-23 by jeffmann