luni64/TeensyStep #22: Program stalls when creating more than two StepControl objects

nate266, 2018-08-06:
I ran into an error when I tried to run three stepper motors using three different StepControl objects; the whole program would freeze when I tried to call any move commands. However, it worked perfectly fine with two motors in the same configuration. Do you know why this occurred?

luni64, 2018-08-07:
I'm currently traveling but try to have a closer look into that in a couple of days. Can you post a minimal example showing the behaviour?

nate266, 2018-08-07:
Sure, here's a very simplified version of my code that demonstrates the bug I've encountered. It works fine until the third stepper motor is commanded to move, at which point the program stalls.

```
// Libraries
#include <StepControl.h>     // Make sure to set the USE_TIMER to TIMER_FTM3 in the library's config.h file - otherwise the PWM of certain pins won't work

// Pin Definitions
#define PUMP_STEP_ENABLE_PIN              24
#define PUMP_STEP_PIN                     35
#define PUMP_DIR_PIN                      38
#define PLUNGER_STEP_ENABLE_PIN           24      // Pin number for the step enable signal to stepper motor controller
#define PLUNGER_STEP_PIN                  36      // Pin number for the initiate step signal to stepper motor controller
#define PLUNGER_DIR_PIN                   39      // Pin number for the direction signal to stepper motor controller
#define CAROUSEL_STEP_ENABLE_PIN          24      // Port number of step enable port for carousel stepper motor
#define CAROUSEL_STEP_PIN                 37      // Port number of step port for carousel stepper motor
#define CAROUSEL_DIR_PIN                  28      // Port used for direction control

// Create stepper motor control objects
Stepper motor0(PUMP_STEP_PIN, PUMP_DIR_PIN);
Stepper motor1(PLUNGER_STEP_PIN, PLUNGER_DIR_PIN);
Stepper motor2(CAROUSEL_STEP_PIN, CAROUSEL_DIR_PIN);
StepControl<> controller0;
StepControl<> controller1;
StepControl<> controller2;

void setup() {
  Serial.begin(9600);
  while (!Serial) {;} // wait for serial port to connect. Needed for native USB port only
  Serial.println("Initialized");
}

void loop() {
  
  // Move motor 0
  Serial.print("Starting motor 0...");
  controller0.move(motor0);
  Serial.println("move complete");

  // Move motor 1
  Serial.print("Starting motor 1...");
  controller1.move(motor1);
  Serial.println("move complete");

  // Move motor 2
  Serial.print("Starting motor 2...");
  controller2.move(motor2);
  Serial.println("move complete");

  delay(1000);

}
```
Here's the printout from the Serial monitor. Note that it stalls after trying to move motor 2. 
```
Initialized
Starting motor 0...move complete
Starting motor 1...move complete
Starting motor 2...
```

nate266, 2018-08-08:
Yeah, I tried your code, and I am still getting the exact same error.

luni64, 2018-08-08:
Strange, which Teensy do you use? Which Version of Teensy Duomo?

nate266, 2018-08-08:
Here's what I'm using:
    Teensy 3.5
    Arduino 1.8.5
    Teensyduino 1.41

luni64, 2018-08-09:
Just ordered a T3.5 to test this.

luni64, 2018-08-12:
Meanwhile got the T3.5 and compiled with the following:

- Teensyduino 1.4.1, 
- Arduino 1.8.5 
- 120MHz
- Optimize: faster

It just works here. Any other idea how to reproduce your problem?

AmitParmar2005, 2018-08-23:
Hi,

I am having the same problem. When I create 3 controller objects and try to move 3 stepper motor asynchronously with moveAsync function, I see nothing in my serial monitor. It works perfectly fine with 2 controllers. 
I am using: Tennsy 3.6, Arduino 1.8.5, Teensyduino 1.41
Here is my code:

#include <StepControl.h>

Stepper motor_1(2, 3);
Stepper motor_2(4, 5);
Stepper motor_3(6, 7);

StepControl<> controller1;
StepControl<> controller2;
StepControl<> controller3;
  
void setup()
{
  Serial.begin(9600);
  while (!Serial) {;}
  Serial.println("SetUP");
  
  motor_1
    .setMaxSpeed(1600)
    .setAcceleration(95000);
  
  motor_2
    .setMaxSpeed(3200)
    .setAcceleration(95000);

  motor_3
    .setMaxSpeed(6400)
    .setAcceleration(95000);
}

void loop() 
{   
     Serial.println("Started");
     motor_1.setTargetRel(5000);
     motor_2.setTargetRel(6000);
     motor_3.setTargetRel(7000);

     controller1.moveAsync(motor_1);
     Serial.println("M1 Move");
     controller2.moveAsync(motor_2);
     Serial.println("M2 Move");
     controller3.moveAsync(motor_3);
     Serial.println("M3 Move");

     while(controller1.isRunning() || controller2.isRunning() || controller3.isRunning())
     {
       Serial.println("Motors Moving");
       delay(10);
     }
}

luni64, 2018-08-23:
Ok, I'll check again.

luni64, 2018-08-25:
Good, using your code I can reproduce the problem. I try to find out what is going wrong here during the weekend.

luni64, 2018-08-25:
I think I found identified the issue already and your sketch runs for me now. Can you give it a try before I merge the fix into the master branch? (Fixed library can be found in branch "bugfix")

AmitParmar2005, 2018-08-25:
Sure, Give me sometime. I will check and let you know.

AmitParmar2005, 2018-08-25:
Perfect. Working fine. Thank you so much for quick fix.

luni64, 2018-08-25:
Fix merged

closed 2018-08-25 by luni64