luni64/TeensyStep #26: Weird behavior with 3 and 4 controller objects

luni64, 2018-09-18:
@conkerts 
Does this work by now for you?
----
Hi, I seem to have weird behavior with 3 and 4 controller objects as well.
I just recently cloned the repo (when I wrote my first question here), and I checked I have the version=0.98 .

Basically I've got 4 Motor objects and 4 controllers (and each controller moveAsync a motor, so pretty straight forward). It first works for some commands, then I get weird motor movements, even I send commands to a completely different controller & motor. After reset of the Teensy it works again for some commands ...
(Teensy 3.5 btw , with all current Arduino versions)

Should I post my details here, or create a new, separate issue ?
And how can I debug this issue most effectively ? .

conkerts, 2018-09-18:
Thank you for asking.
I worked yesterday on a minimal sketch like you suggested and as this makes the most sense.

I tested a truly minimal example and it worked. 
So I think the problem seems to be in connection with serial IO & the GeekFactory Shell Library and is not the fault of the TeensyStep library.
But it would be cool, if you have a look at it, as well and maybe provide some tips, how to debug it. more effectively.

[ts_4motor_4controller_test2.txt](https://github.com/luni64/TeensyStep/files/2392014/ts_4motor_4controller_test2.txt)

So, in the terminal, I start with motor 1 and let it run 100 steps.
mrun 1 100
Then I let motor 2 run 100 steps
mrun 2 100
Then I let motor 1 run again
mrun 1 100

Usually then both motors move, but this should not happen, as all 4 motors and controllers are independent in the code.

The weird thing is, the file contains also commented out code from the first minimal example, where I let the motors run hardcoded (you just need to comment out the shell task in the loop), and it works as expected). Totally weird.

luni64, 2018-09-18:
I had a look at your code. The reason for the problem with two running motors is your switch case in  sh_run_motor()

```c++
shell_printf("settings motor %s : move %d \r\n", argv[1], number);
  switch (argv[1][0] ) {
    case '1':
      M1.setTargetRel(number);
      break;
    case '2':
      M2.setTargetRel(number);
      break;
    case '3':
      M3.setTargetRel(number);
      break;
    case '4':
      M4.setTargetRel(number);
      break;
    default:
      shell_println("wrong mot number");
  }
  //->>>>> BUG, the following always moves all motors....
  controller1.moveAsync(M1);    // Do the move
  controller2.moveAsync(M2);    // Do the move
  controller3.moveAsync(M3);    // Do the move
  controller4.moveAsync(M4);    // Do the move

  return SHELL_RET_SUCCESS;
}
```
In your four cases you set the target of the selected motor but then you start **all** motors. Which is what you observe. You need to include the moveAsync in the case blocks to make it work: 

``` c++
 switch (argv[1][0])
  {
  case '1':
    M1.setTargetRel(number);
    controller1.moveAsync(M1); // Do the move
    break;
  case '2':
    M2.setTargetRel(number);
    controller2.moveAsync(M2); // Do the move
    break;
  case '3':
    M3.setTargetRel(number);
    controller3.moveAsync(M3); // Do the move
    break;
  case '4':
    M4.setTargetRel(number);
    controller4.moveAsync(M4); // Do the move
    break;

  default:
    shell_println("wrong mot number");
  }
  return SHELL_RET_SUCCESS;
}
```
**Remark: **
Make sure to never send a move command while the controller still moves a motor. This would mess up internal variables. If you use async commands you should aways check if the motor stands still before sending other commands. 

Here one way how to achieve that: 
```c++
...
switch (argv[1][0])
  {
  case '1':
    if (controller1.isRunning())
    {
      shell_println("ignored, motor running");
    }
    else
    {
      M1.setTargetRel(number);
      controller1.moveAsync(M1); // Do the move
    }
    break;

  case '2':
...
```

Hope that helps

conkerts, 2018-09-19:
Thank you, that was really helpful and it would've taken me some time to figure that out. This explains all the behaviour I saw. Also funny, that my minimal hard coded sketch did run without problems, as the bug was really in the callback function of the shell.

Part of my problem was probably, that I expected to use this library similar to the Accelstepper lib - 
which even required to have the call to the move command "as often as possible" in the main loop. I found this really weird. 
The move commands in this library and especially the Async variants are well-conceived compared to that.

One thing, I find slightly unintuitive, is that the setTargetRel(number) isn't reset to zero internally after the move is finished. 
Wouldn't that be more logical? It would at least prevent this (in my case) unexpected behaviour ;). On the other hand, yes, one has to be sure every time when to call the move commands and there wont be any problems.

Your remark is really interesting as well. This explains also some "hiccups" of the motors I saw, when I send new commands while they were moving.
So one way would be basically to use stopAsync()  and then issue the new command (while always checking, that the motor isn't running) ?

My problem is, that I actually planned to use an outer control loop with external sensors, and then update the TargetRel in every controller iteration. So I probably have to switch to speed commands.

Then another question would be if this remark also applies to the rotate commands ? Can I change the speed while the motors are running ?

closed 2018-11-16 by luni64