luni64/TeensyStep #27: TeensyStep and TMC2130 SST

ecrou, 2018-11-13:
Hi,
I don't know if i'm in the good place to post this issue, that is my first message on github. I ll move/delete the message if the place is not approprieted.
I 'm trying to move nema17 stepper with tmc2130 and TeensyStep, with teensy 3.2.
Recently, i had no problem with the classic AccelStepper lib, but it's not powerful enough for my application.
This is my sketch:
```C++
#define EN_PIN    4  
#define DIR_PIN   3  
#define STEP_PIN  2
#define CS_PIN    10  

int pos;

#include <TMC2130Stepper.h>
#include <StepControl.h>
TMC2130Stepper driver = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN);
Stepper motor(2, 3);
StepControl<>controller;

void setup() {
  motor.setMaxSpeed(30000);
  motor.setAcceleration(100000);
  
  SPI.begin();
  Serial.begin(250000);
  while (!Serial);
  Serial.println("Start...");
  pinMode(CS_PIN, OUTPUT);
  pinMode(EN_PIN,OUTPUT);
  digitalWrite(CS_PIN, HIGH);
  driver.begin();             // Initiate pins and registeries
  //driver.rms_current(2000);    // Set stepper current to 600mA. The command is the same as command TMC2130.setCurrent(600, 0.11, 0.5);
  driver.run_current(20);
  driver.hold_current(5);
  driver.hold_delay(2);
  driver.stealthChop(1);      // Enable extremely quiet stepping
  driver.stealth_autoscale(1);
  driver.microsteps(32);

  //digitalWrite(EN_PIN,HIGH);
  
}

void loop() {
    pos = motor.getPosition();
    pos=pos+1000;
    Serial.println(pos);
    motor.setPosition(pos);
    controller.move(motor);    // Do the move
    delay(2000);
}
```
Right now, motor is not moving. 
I was looking on problem concerning EN_PIN, but even by commenting/uncommenting
```C++
digitalWrite(EN_PIN,HIGH);
//or
digitalWrite(EN_PIN,LOW);
```
Nothing is appening.
I double check code/wiring, but i may make a mistake.
Can someone have an idea about this issue?
Thanks in advance.
Regards.

luni64, 2018-11-16:
I don't know your TMC2130 library but the code related to TeensyStep seems to be OK. Might be something related to SPI? Any chance to use a simple DRV8825 or a A4988 board for testing to exclude SPI problems?

ecrou, 2018-11-16:
thank's Luni for this answer. SPI for the tmc2130 is just used once in the setup, but indeed, that's the only difference between tmc drivers and classic drivers. i ll take a look at the code of libs to see where the problem. Thanks. Sincerly.

closed 2018-12-29 by luni64