luni64/TeensyStep #28: Compiler error in StepControl.h

ramboerik, 2018-11-15:
When compiling the code it produces the following error:
```
TeensyStep/src/StepControl.h:267:52: error: no matching function for call to 'min(double, float&)'
```

It is related to the constants 0.0/1.0 being translated by default to double while relSpeed is of type float. The template for min/max cannot resolve a function with different data types as parameters and fails.

By explicitly specify that the constants should be of type float everything works: 
```
change:
relSpeed = std::max(0.0, std::min(1.0, relSpeed)); // limit relative speed to [0..1]

into:
relSpeed = std::max(0.0F, std::min(1.0F, relSpeed)); // limit relative speed to [0..1]

```

gcc version: 5.4.1
Compiled with c++14 support.

luni64, 2018-11-15:
Looks like you are not compiling with Teensyduino / Arduino?  Teensyduino sually defines  
`-fsingle-precision-constant` so that all numeric constants are defined as float instead of double. Anyway, it won't hurt being more explicit here. -> I'll fix it.

ramboerik, 2018-11-16:
ah, suspected something was different with my environment, thanks!

yes that's right. I'm using the Makefile found in teensy core (https://github.com/PaulStoffregen/cores/blob/master/teensy3/Makefile) and it doesn't use that compiler flag.

closed 2018-11-16 by ramboerik