luni64/TeensyStep #29: Check for NULL in doRotate?

frdfsnlght, 2018-11-19:
I'm trying to use this library with an array of 16 steppers. I need to have some subset (up to 16) of the motors rotate continuously but I don't know which motors until run-time. The motors that need to rotate are specified via a command to the Teensy. I've already increased MaxMotors.

So, the method I've written doesn't know beforehand how many motors will be rotating, so it declares an array of sixteen Stepper pointers initialized to NULL, then fills in as many as necessary for the given the command. It then passes this array to rotateAsync.

If I'm reading the library code correctly, this entire array (16 elements, some of which are NULL) is copied to motorList and doRotate is called with N=16. In the loop in that method there is no check against NULL before calling the setTargetRel method on each motor. That seems bad for my use case.

Should there be a NULL check before making that call? Or can you suggest a different way of doing what I need?

Also, sort of related to this, the templated rotateAsync method has a line: motorList[N] = nullptr; But the motorList is only initialized to have MaxMotors elements, which in my case is 16. Doesn't that line overwrite the memory beyond the end of the array for N=16? Should motorList actually have MaxMotors +1 elements? Or should line be removed?

frdfsnlght, 2018-11-19:
Regarding the last paragraph: setting MaxMotors to 17 fixes the problem and satisfies the assert statement in rotateAsync. I guess the name MaxMotors is a slight misnomer?

luni64, 2018-11-19:
You are right, MaxMotors is misnamed somehow... But the assert statement correctly prevents overwriting the array. 

Regarding your first question: 
the interrupt functions need the nullptr as marker for the end of the array.  So your idea won't work. 

Why not simply adding another interface? Something like: 
```c++
void StepControl<p, u>::rotateAsync(Stepper** steppers, int N, float relSpeed)
{
    // copy N stepper pointers  into the motorlist, check that N < MaxMotors-1, set motorlist[N] to null
}
```
you then can use a dynamic array instead of the compile time array currently implemented.

frdfsnlght, 2018-11-19:
A new interface where N can be specified certainly works.

But it seems like a NULL check would still be good. I don't want to pass in an array of 16 elements with random elements set to NULL, a NULL would indicate the end of the array and in fact all elements beyond that would also be NULL, so wouldn't the interrupt routine still function correctly? So the NULL check in doRotate could simply break out of the loop when it first detects a NULL element.

luni64, 2018-11-19:
Just had a closer look, there are a few calls to library functions which also need a correctly set N. Might be a larger change to get it going...

What about changing: 
```c++
inline void StepControl<p, u>::rotateAsync(Stepper *(&motors)[N], float relSpeed)
{
    static_assert((N + 1) <= sizeof(motorList) / sizeof(motorList[0]), "Too many motors used, please increase MaxMotors");

    for (unsigned i = 0; i < N; i++)
    {
        motorList[i] = motors[i];
    }
    motorList[N] = nullptr;
    doRotate(N, relSpeed);
}
```

to this (untested!): 

```c++
inline void StepControl<p, u>::rotateAsync(Stepper *(&motors)[N], float relSpeed)
{
    static_assert((N + 1) <= sizeof(motorList) / sizeof(motorList[0]), "Too many motors used, please increase MaxMotors");

    for (unsigned i = 0; i < N; i++)
    {
        if(motorList[i] == nullptr) break;
        motorList[i] = motors[i];
    }
    motorList[i] = nullptr;
    doRotate(i, relSpeed);
}
```

luni64, 2018-11-19:
Yes, you are right.  I try to implement this and the dynamic array interface tomorrow.  

> BTW, I don't know if you remember me or my project, but I'm the guy building a Barbot. I needed a change like this because of a late addition to functionality I just added. I'm nearing completion of the project, perhaps a couple of weeks. The whole project is here: https://github.com/frdfsnlght/Barbot It's lacking a ton of documentation right now. I'll get to that eventually, and some pictures!

Yes, remember you, still waiting for those pictures / videos :-)

frdfsnlght, 2018-11-19:
Another thought... I don't think the proposed change solves the problem. The doRotate method will still try to setTargetRel on a NULL motor. And I just realized doRotate then calls doMove with the value N (16 in my case), which will cause further problems since there isn't really 16 motors in the array. Perhaps something like this:

`void StepControl<p, u>::doRotate(int N, float relSpeed)
{
    uint32_t maxSpeed = (*std::max_element(motorList, motorList + N, Stepper::cmpV))->vMax;
    float fac = (float)std::numeric_limits<int32_t>::max() / 2.0f / maxSpeed;

    for (int i = 0; i < N; i++)
    {
        if(motorList[i] == nullptr) break;
        motorList[i]->setTargetRel(motorList[i]->vMax * fac * motorList[i]->direction);
    }

    doMove(i, relSpeed, false);
}`

Note the extra line in the beginning of the loop, and the change to the doMove method using "i" instead of "N".

frdfsnlght, 2018-11-26:
I had a chance to test this over the weekend and the final code I'm using is this:

```c++
void StepControl<p, u>::doRotate(int N, float relSpeed)
{
    uint32_t maxSpeed = (*std::max_element(motorList, motorList + N, Stepper::cmpV))->vMax;
    float fac = (float)std::numeric_limits<int32_t>::max() / 2.0f / maxSpeed;

    for (int i = 0; i < N; i++)
    {
        if(motorList[i] == nullptr) {
            N = i;
            break;
         }
        motorList[i]->setTargetRel(motorList[i]->vMax * fac * motorList[i]->direction);
    }

    doMove(N, relSpeed, false);
}
```

I'm going for fork your project and make the changes for myself. I can submit a pull request if you'd like, but perhaps you're working on a more complete solution.

luni64, 2018-11-26:
Sorry, I'm currentyl completely absorbed in another project. A pull request would be great...

closed 2019-03-22 by luni64