luni64/TeensyStep #30: movement planner.. small stepping (turn off acceleration or deacelleration)

rvsh2, 2018-11-30:
Hi,

I would like to use teensy to move 5 dof robotic arm. Actually I'm already using it with some success.
The problem is moving the arm in straight line in inverse kinematics... It is moving point to point so the end position is reached properly but the movement on track is "bending".

I would like to do many small steps to move in straight line in real world kinematics. So I would calculate inverse kinematics for each step but the problem is that every small step has accelleration and deacceleration so it can't run smoothly. 
It would be helpfull to have a posibility to moveAsync but with some simple motion planning..

What do you think?

BR,
Krzysztof

luni64, 2018-11-30:
Implementing a motion planner would be quite a big effort I assume. But if you just want to disable acceleration as you wrote in the heading you could set it to some very large value, e.g., 500000. This would effectively turn of acceleration. Would that help?

rvsh2, 2018-12-03:
Ok. Thanks. I will try this.

rvsh2, 2018-12-03:
I would describe what I would like to do:

Let's say that I want to move from point A1 = 0 to Point A1000.
And I would like to make 802 moves (small steps) to do this.
Step 1 would be 1 move from point A1 =0 to point A100 with constant acceleration and without deacceleration.
Step 2 would be 800 moves from A100 to A901 by 1 without acceleration and deacceleration.
Step 3 would be 1 move from position A901 to A1000 without accelearation and constant deacceleration. 

I need this because I want to recalculate the position in space(inverse kinematics) with every move.

Is this possible to do?

P.S. I think I can use the newest callback function to achieve some of this but how to turn of accel **OR** **deaccel.**

luni64, 2018-12-03:
I don't know how good your programming skills are, but on Saturday I uploaded a new version of the library where I extracted all the acceleration / deceleration code so that it can be easily changed by users. You could then implement your own acceleration algorithm. The points from above should be possible then. 

You find more info here: https://forum.pjrc.com/threads/43491-New-Stepper-Motor-Library?p=192145&viewfull=1#post192145. 

Hope that helps

closed 2018-12-29 by luni64