luni64/TeensyStep #31: Start motion after stopasync

rvsh2, 2018-12-03:
Hi,

Is there a functionality to continue motion after using stop or stopasync?
How to do this?

Thanks in advance.
Krzysztof

rvsh2, 2018-12-03:
Thanks. I undestand this. The problem is if I don't know exact target position (because I don't store it anywhere..) and I use interrupt to stop the motor. 
For example. There is a 5 axis robot in the cell that do some movement using moveAsync  and if you open safegate then interrupt is called to stop movement but in the interrupt I don't know what was the target position so I can't continue.. Can I get the target position somehow from some variables so I can start the motors again? Also after stop I don't know exactly which motor was running before but If I could know target positions of motors and actual positions of motors then I could start motors again to run to destination after safegate is closed and reset/start buttons pressed.

Many thanks for help!

rvsh2, 2018-12-03:
Maybe this is trivial question but I'm stuck.

luni64, 2018-12-03:
There is no simple solution to access the target positions from the library. But I really don't understand why you don't simply store the positions after you set them? I can send you a quick example later today to show you how I would do it.

closed 2018-12-29 by luni64