luni64/TeensyStep #33: Beta Version of new StepControl

luni64, 2018-12-23:
I updated a beta version of an improved controller. (Branch _develop_newRotate_)

**Changes:** 

1. Move and moveAsync commands now take up to MaxMotors (i.e. 10) motors instead of 3
2. Improved acceleration efficiency. For a Teensy 3.1 the acceleration update is done about 30% faster than the old one. 
3. Removed the rotation commands. They will go in a separate controller for efficiency reasons
4. API for acceleration plugin slightly changed.

frmdstryr, 2018-12-24:
Can you do a pull request to make it easier to add comments on specific code changes (using github's review process)?

frmdstryr, 2018-12-24:
Regarding the PIT timer... Do you think it would be better to request that [Teensy cores](https://github.com/PaulStoffregen/cores) simply make the IntervalTimer's channel protected to allow access to the channel directly via subclasses (or have request the IntervalTimer provide a function to set channel cycles directly) ?

closed 2018-12-29 by luni64