luni64/TeensyStep #35: STM32duino port

cdrose, 2018-12-28:
Hi, I have been working for a little while on a stepper project based on one of the STM32 arduino cores. I have had a look at porting your library to STM32 though with the current structure it is very tightly coupled to the Teensy platform so I have had to hack it up quite a bit.

Would you be interested in working with me to support both platforms in the same library? Or at least decoupling the stepper algorithm from the timer and gpio stuff so I could more easily keep an STM32 version in sync with yours?

luni64, 2018-12-28:
Hi, 
Yes, I'm definitely interested. I had the same idea and already got a STM32 board for testing yesterday (Maple mini). I already started accessing the timers through a more generic interface in the rotational branch. Need to clean that up a bit and add the delay timer to the interface. This will also allow for a simple "tick-timer" (call tick() in loop), so that the library would be accessible from all arduino boards...

Here my current idea of an interface (to be discussed) 
```c++
class ITimerField
{
    void begin(); 

    void stepTimerStart();          
    void stepTimerStop();
    void stepTimerSetFrequency(unsigned f); //Hz, f=0 should stop the timer, using period instead of freqency??

    void accTimerStart();   
    void accTimerStop();
    void accTimerSetFrequency(); //Hz, f=0 should stop the timer, using period instead of freqency??

    void delayTimerTrigger(unsigned delay); //µs does not need to be very accurate   
}
```

Stepper.cpp also uses some optimized code for step/dir signals which also needs to be abstracted. 

I was thinking of implementing the plugins via a template mechanism to keep performance high and allow for simple combination of variable parts (different accelerators, different timers etc). 

I need a couple of days finalizing some other projects then I plan to work on the abstraction. Would be great if you could work on the STM32 part.

cdrose, 2018-12-28:
Sounds great!

I currently use two basic timers, one for the accel and delay times, and then the second extended to 32bit in software for the step timer. With a bit of rework I think all 3 could run off a single timer, 1 compare channel is used to count overflows and make it behave like a 32bit timer and then the other 3 can be used for step, dir and acc timing.

The STM32 arduino environment is a bit fragmented at the moment, there are several cores available each supporting different boards. I am most familiar with Roger Clarks core https://github.com/rogerclarkmelbourne/Arduino_STM32 which is built off the old Maple code, however I think the official STM core https://github.com/stm32duino is probably the best target in the end. Unfortunately the F103 boards do not have a USB bootloader in the STM core which is a pain for me.

luni64, 2018-12-29:
Uploaded branch plugin-system. It currently only contains plugins for different accelerators (currently only for the RotationController & SinRotAccelerator is just a copy of the linear accelerator), I'll work on the timer plugins later. 
Can you have a look if this makes sense to you?

luni64, 2018-12-30:
@cdrose: I finally removed all hardware specific code from the controllers (PR #36). To port TeensyStep to another architecture, e.g. STM32, a class with the following interface needs to be defined for the target system. This class will be plugged into the controllers in TeensyStep.h. 

- Location: src/timer/ARCHXXX/
- Example src/timer/teensy3/TimerField.h  (very crude but working wrapper around IntervalTimer and TeensyDelay)

Let me know if you are still interested in porting

```c++
class TimerField 
{
public:
  inline TimerField(TF_Handler *);

  inline bool begin();   // do whatever needs to be done to setup the timer field
  inline void end();

  inline void stepTimerStart();   // stepTimer generates the step pulses, needs to be a 32bit periodic timer up to 300kHz required. 
  inline void stepTimerStop();
  inline bool stepTimerIsRunning() const;
  inline void setStepFrequency(unsigned f); 

  inline void accTimerStart();  // accTimer is a periodic timer, typical period ~500µs - 10ms 
  inline void accTimerStop();
  inline void setAccUpdatePeriod(unsigned period);

  inline void setPulseWidth(unsigned delay);   // pulseWidth timer is a  one shot timer, typical period 5-15µs
  inline void triggerDelay();
};
```

cdrose, 2019-01-01:
That looks great. Ill see if I can update my current code to work with the new interface over the next few days.

luni64, 2019-01-01:
Good, just uploaded the latest Version to PR #36. The timer interface works nicely so far. I think it is stable enough now to program against it. 
Looking forward to your code.

cdrose, 2019-01-03:
Hey just a quick update. I haven't brought my code into the new plugin system yet, but I have refactored my stm32 timer to use a single hardware timer and roughly follow the new timer interface. Unfortunately I have a bug somewhere in my timer code that causes it it to occasionally miss an interrupt and then it hangs until the timer overflows and it catches it the next time round. As soon as I get this sorted ill rearrange it all so it can be used as a plugin.

luni64, 2019-01-03:
Great, I'll try to set up a STM32 toolchain to be able to test.  'll useI the official STM32duino from GitHub

cdrose, 2019-01-03:
Mine is built on Roger Clarks core, I haven't looked into the official core yet, hopefully the timer setup is at least somewhat similar.

AmitParmar2005, 2019-10-03:
Hey Guys,

I am also interested in using this library for my STM32 boards. Have you made any progress? Did you merge changes into Master Branch? I tried to built Master Branch with my STM32F103C8 Blue Pill and It gave me an error.

cdrose, 2019-10-03:
I had some code which mostly worked on the Roger Clark Arduino core but I stopped working on it when development stopped on that core. In the STM Official core the hardware timer library was (is?) still very basic and there is quite a bit of work to do to make it compatible with Teensy Step. I did some work on this but ran out of time before I got anything working.

AmitParmar2005, 2020-01-23:
Hi Again,

Sorry I was very busy so, could not get back to you. Do you mind sharing your mostly working code? I will try to give it a shot. If I make any progress, I will let you know.

luni64, 2020-10-05:
See v2.2.0

closed 2020-10-05 by luni64