luni64/TeensyStep #38: relSpeed in doMove

djamu, 2019-02-18:
Hi, first of all thanks for this great library, great coding here.

I have some questions / suggestions.
I noticed an undocumented option when looking at the library > relSpeed.
I assume an intended use is for speed compensation when using 2 steppers in a Cartesian  X/Y configuration, more precisely to achieve a constant speed on diagonal moves.
> for a diagonal 45deg move, motors need to run at 0.707106...  times the speed of a straight move.
This can also be done by just using setMaxSpeed on the 2 motors.
Which makes relSpeed kind of redundant.

Since you properly implemented the Bresenham algorithm, in which either 1 or 2 motors move simultaneously, the integral sum of both moves(steps) as C is exactly the same as the hypotenuse from a (X)  b(Y).  ( both as steps)  > c = √(a² + b²)
( as opposed to the implementation where either motor moves separately  and the hypotenuse becomes a + b ... in theory  :-) > funny math but true ... ) 

My point is that in this instance (cartesian 2D kinematics) relSpeed should always either be 1.0 ( when 1 motor steps ) or 0.707106  when both step ... ( on every doMove ).  
> directly computed onto the StepTimer reload values.
This would free up quite a lot of resources since one doesn't need to calculate either relSpeed. or setMaxSpeed. to get a constant absolute speed like one has to do now.. 

A Stepcontrol.setMaxSpeed function for the controller, that overrides the individual  Stepper.setMaxSpeed function can then be used to control the actual absolute speed. 

Yet another suggestion would be to implement a callback function on  (!Stepcontrol.isRunning())
Now you have to constantly poll this function which makes your elegant code in practice a bit crude.
Even better would be an option to define a shared ring buffer, which can be handy for next suggestion.
> I'm assuming you'll want people to actually use this in for example their own gcode interpreter ...

Last suggestion.
Acceleration / deceleration always goes from 0 > max > 0.  
An option to define a min acceleration would be great.
I know one can effectively disable acceleration by  setting setPullInSpeed to vMaxMax, but my suggestion is not the same...

One could then use said ring buffer to predict how much acceleration / deceleration is actually needed for the next move, depending on the difference in angle.  
For example a 90 deg angle difference would result in 0.0 > 1.0 for X / 1.0 > 0.0 for Y.  (acceleration/deceleration for X/Y)
A 10 deg  > angle  would then need  0.984... > 1.0 for X /  1.0 > 0.984... for Y.  
Easy math ... 
 
I might have a go at it myself later on.. I'm just curious what you think of these ideas.
Your programming skills are better then mine.... 

Lastly.
I'm not sure why people haven't picked this up...
A very easy way to get high speed pulses.

Up to now I always used shift registers connected to SPI to create high speed pulses.
For example 2 * 74hct595 daisy chained ) connected to a 16MHz SPI  gives you  500Khz on/off  on 15 pins ),  use the msb bit to auto latch the chips, you'll lose 1 bit but don't need an SS pin then ( and it's latency)
DMA transfers  makes this very lightweight ( you do need a reset pin at startup ).
Do a begin transaction once in setup,  and just transfer....

use HCT if you need 3.3 > 5V voltage translation ( HIGH starts at max 2V  typ 1.6V  input pins) 
use HC for 3.3V  ( works from 2V - 7V  )  both up to 100MHz ..

luni64, 2019-02-18:
> I assume an intended use is for speed compensation when using 2 steppers in a Cartesian X/Y configuration, more precisely to achieve a constant speed on diagonal moves.

No, the original idea was to have all the physical properties like the maximal speed of the motor in the stepper class and use the relative speed to set the speed you want the motors run. But this quickly turned out to be too complicated. You can regard this feature as depreciated. The new version (Branch develop) doesn't have this anymore. 

> My point is that in this instance (cartesian 2D kinematics) relSpeed should always either be 1.0 ( when 1 motor steps ) or 0.707106 when both step ... ( on every doMove ).

This would only be true for 45° movements. So a very limited use case.  Doing the math for arbitrary angles is easy but I don't think this should be in a general purpose library. You can always calculate that outside the library. 

> Yet another suggestion would be to implement a callback function on (!Stepcontrol.isRunning())
> Now you have to constantly poll this function which makes your elegant code in practice a bit crude.

This is already implemented in the next version (see the development branch)

> I'm assuming you'll want people to actually use this in for example their own gcode interpreter ...

Actually I don't. There are much better solutions for cnc controlling applications out there. Look for grbl, Mach3, linuxCNC ....

> Last suggestion.
> Acceleration / deceleration always goes from 0 > max > 0.
> An option to define a min acceleration would be great.

The acceleration is constant. You probably mean velocity? 

> I know one can effectively disable acceleration by setting setPullInSpeed to vMaxMax, but my suggestion is not the same...

Better to set the acceleration to the highest value (500'000) this will effectively disable acceleration. 

> One could then use said ring buffer to predict how much acceleration / deceleration is actually needed for the next move, depending on the difference in angle.
For example a 90 deg angle difference would result in 0.0 > 1.0 for X / 1.0 > 0.0 for Y. (acceleration/deceleration for X/Y)
A 10 deg > angle would then need 0.984... > 1.0 for X / 1.0 > 0.984... for Y.
Easy math ...

Yes, but as already mentioned, there are very good solutions for that problem available. Look for cnc controllers, or some 3d printer firmware. 

>I might have a go at it myself later on..
Great, feel free to use TeensyStep as a starting point.  Looking forward to your implementation. 

Have fun

djamu, 2019-02-19:
My use case is a bit academic. I know there are very mature complete controllers available, but I needed something that could easily generate high step rates without the need to attach end stops etc... . Didn't feel like hacking marlin / teacup / G2 .. yet ... just to disable every and all feature aside from stepping... 
and I still doubt the stock versions handle the very high pulse rates I need.
The solution with SPI is solid, but not finished. I do plan to program a module for marlin.

I'm finishing up a very cheap very accurate closed loop stepper controller, as drop in addition / replacement for existing 3D printers and motors ( currently only polulu ). In it current incarnation it does 819200 steps / revolution, and has a 40kHz control loop ( it verifies / adjusts position at that rate ). 
Which is quite challenging to get running at any decent speed ( not from the motor controller standpoint ). Even bringing it down to 32768 steps / revolution is  quite challenging for anything that gives out pulses. 
To put that into perspective, 32768 steps / revolution, using a 16 teeth gt2 pulley means a little under 0.001 mm accuracy, max deviation I got at 200mm/s (375RPM) was 0.004mm ( but nobody prints at hose speeds ) the 0.001 mm precision is maintained upto 80mm/s.  
The aim was not to get closed loop as such ( don't need 40kHz control loop for that ), but to get rid of "ringing" ( as you no doubt have a 3D printer yourself, the artifacts you get behind sharp corners ). Contrary to popular belief, the primary reason for that is not the flexing of the frame, but the springiness of the steppermotor itself, ( loosing steps is when it "springs" too far, and can happen both on acc / dec) a motor can safely "spring" < 2 * 1.8 deg  in each direction, the advice of some to not tighten your belts too much as to damp this phenomenon is sound, the advance in microstepping drivers won't change that. This is also the reason why some people are adamant about the thesis that microstepping doesn't improve accuracy. Both parties are right, it does and it doesn't but for very different reasons. 

I've never came across a microstepping driver that creates a linear movement, where each input step creates an evenly amount of movement. More important even is following.
Testing the closed loop sensor, I found out that about any stepper motor has a cyclic and symmetric error. The cyclic being repeated every 4 (full motor ) steps . Going into detail about this is very much out of scope, but I found that on most tested motors the sum of both errors at some point exceeded 1/4 (motor) step, in short the only advantage  128/256 ustep drivers will ever have over 32ustep drivers is their lack of noise, and possible smoother operation, but certainly not their precision, that is a hoax.

back to the topic:

>> My point is that in this instance (cartesian 2D kinematics) relSpeed should always either be 1.0 ( >>when 1 motor steps ) or 0.707106 when both step ... ( on every doMove ).

>This would only be true for 45° movements. So a very limited use case. 

From a theoretical perspective.
The Bresenham algorithm essentially only makes 0 or 45 degree moves . Either 1 or both motors step. 
The thesis on the integral sum of steps I posed was incomplete, but is valid, stepper motors don't step slow. they just step to the next position with the highest possible acceleration and velocity they can momentarily achieve. ( they don't do 1/2 speed steps ), therefore in an X/Y configuration they either move 0 / 45 deg .
The actual speed at which they rotate is an apparent one,  an average of steps / sec :  
So while c = √(a² + b²) seems valid ( and is the real world ).  C = is still the apparent distance covered, from a theoretical perspective, not the real one, not taking into account friction/ inertia etc....
 
The actual distance covered by 2 steppermotors (in a X/Y configuration) driven by Bresenham is actually:  
where either or both step : b<a   c = (a - b) +(  b / 0.7071 )  
where only 1 of them steps at any given time    c = a + b 
I know this is weird math...  somewhere someone might explain this better then me ... as it is very contradictory... 

Just think of this example, imagine a bike with an infinite long wheel base. front wheel steered by Bresenham, for each finite element, front wheel can either be 0 or 45 deg.  driving across a diagonal. The total distance covered by front wheel = C   b<a   c = (a - b) +(  b / 0.7071 )
Back wheel will still be c = √(a² + b²).
A gantry will because of damping / friction / inertia act like the back wheel in the example,
But that doesn't mean one should try to drive the front wheel like the back wheel. And the example proves that you don't need to do that. 

It actually simplifies things , as one can git rid of all sin / sqr operations. and just use constants, not even lookup tables are needed.
As I said I'll give it a go...

>The acceleration is constant. You probably mean velocity?
Yes of course sorry for that.

>Better to set the acceleration to the highest value (500'000) this will effectively disable acceleration.

No it doesn't, I tried, it still accelerates 500.000 steps/s^2  I double checked this with a scope.
With normal motor drivers it's probably hard to tell what is caused by inertia and what by software.

Scratch that, creating a .setPullInSpeedAcc(n) and  .setPullInSpeedDec(n)  might do exactly what I want for successive lines now. 

Sorry for the lengthy post

 cheers 
Jan

luni64, 2019-02-20:
Interesting project indeed.  Just a few questions/remarks:

>...To put that into perspective, 32768 steps / revolution, using a 16 teeth gt2 pulley means a little under 0.001 mm accuracy, max deviation I got at 200mm/s (375RPM) was 0.004mm ( but nobody prints at hose speeds ) the 0.001 mm precision is maintained upto 80mm/s.

I assume this 1µ precision is defined as the deviation of the sensor value from the target position? Just for my understanding: what sensor do yo use and where is it mounted (motor axis or final movement axis?) 

> ... springiness of the steppermotor itself, ( loosing steps is when it "springs" too far, and can happen both on acc / dec) a motor can safely "spring" < 2 * 1.8 deg in each direction, the advice of [...]

Fully agreed, the concept of step losses seems to be completely misunderstood out there. (There is no steploss smaller then 4 fullsteps...) As you certainly know, the spring constant of that "motor spring" is larger for higher motor current, thats the reason why lower motor current leads to smoother movment, but of course also to a larger overshoot if you decelerate too quickly,

>...I've never came across a microstepping driver that creates a linear movement, where each input step creates an evenly amount of movement. 

There are drivers where you can adjust the current for each and every microstep to compensate that nonlinearity. 

>More important even is following.
Testing the closed loop sensor, I found out that about any stepper motor has a cyclic and symmetric error. The cyclic being repeated every 4 (full motor ) steps . Going into detail about this is very much out of scope, but I found that on most tested motors the sum of both errors at some point exceeded 1/4 (motor) step,

Yes, I observed that quite frequently when developing high precision drives for spectrometers. The imprecision is greatly dependent on the quality (price) of the motor. The 2pi error probably comes from the axis not really running true. The 4 step error is probably  related to the mechanical precision of the poles. Try some good motors (Oriental, Nanotec) to see the difference. Using 0.9° motors instead of 1.8° also improves precision a lot. 
BTW: same is true for the sensors. I recently tested the performance of those cheap rotary encoders you get for <10EUR by comparing to a professional (Heidenhain) encoder. The result was quite bad. Opening it clearly showed that the encoder disk was mounted very crudely.  

>Scratch that, creating a .setPullInSpeedAcc(n) and .setPullInSpeedDec(n) might do exactly what I want for successive lines now.

Yes, I gave the whole idea of a simple motion planner some thought yesterday and will try to implement something like that in TeensyStep. However, I have to finish some other projects first, so this may take some weeks. If you need a quicker solution feel free to use TeensyStep as a starting point.

closed 2019-03-05 by luni64