luni64/TeensyStep #39: Closed Loop Steppers controlled by teensy

dimisxls, 2019-02-21:
hi luni64!
Congratulations on the amazing work you have done. I have been using the teensy step library for some time to drive a boring machine with multiple drills (4). The selected drill lowers down with pneumatic valve and moves in three dimensions (XYZ) with nema 34 stepper motors (2 for X and 1 for each Y and Z). All these are guided by a teensy 3.5 plugged on a pcb board of my own design (5vTTL and optical isolated switches and relays) .

To have high precision in movements I only use 1.8 degree steps and I have set the drivers at 800 steps per round for smooth movement. So I use steps multiple of 4 to drive the motors (the same for home position). The problem is that I have a very low resolution on X and Y axes. With 31 teeth module1 pinion the resolution is 31 * 3,1415 / 200 = 0.48 mm for each step. An analysis of 0.1mm per step would be completely satisfactory and for this I'm thinking of using microstepping and closed loop motors controlled by teensy.

Do you think that tennsy can drive 3 motors simultaneously and read the 3 encoders at the same time? A second thought is to use 2 teensy, one for driving and one for reading, which will communicate once the drill reaches its target so that it can check if the coordinates (X and Y) are correct before z movement.

luni64, 2019-02-21:
Hi, 
glad that you like TeensyStep. Moving 3 motors at moderate speed and reading out 3 encoders will definitely not generate a significant load on a Teensy 3.2 

Remarks: 
I don't think that you really need a closed loop system for that kind of application. I'd do a simple experiment  and increase the microstepping to 16 (3200 steps / rev) and measure the precision. 

If I understand correctly, you are using a rack/pinion drive? Couldn't you switch to a simple lead screw drive? At a pitch of 6mm you'd get a resolution of  6mm/800steps = 0.0075mm/step. 
Best

dimisxls, 2019-02-22:
hi!
I use rack/pinion drive because X and Y axis are quite long for lead screw. The table is 3000mm on X, 1300mm on Y and also I want it the machine to be fast. I ve noticed that with microstepping the machine looses stepps after a while becouse it is heavy and it moves fast. Probably drives can't hold the position betwin the 1.8 degrees angle of stepper motor poles.

That's why i am thinking of closed loop steppers with full control by microcontroler. Maybe that solution wouldn't prevent the deflection on axis position betwin the poles instantaneously (if that loss comes from the pressure exerted the moment the drill hits the surface of the panel and not from the rapid accelaration/decelaration) but that deflection would not be cummulative.

luni64, 2019-02-23:
Sorry for the late answer.

dimisxls, 2019-02-24:
Hi luni64,

Thank you for your answer. This conversation was enlightening to me, particularly after reading your topic to pjrc forum which I was not aware of it. I think you are right about closed loop stepper motors, they are complicated and not worth applying (code for encoders uses interrupt pins that maybe affect the program) . I will try to lower the acceleration in compination with microstepping and maybe I will implement a time belt/pulley to pinion drive in a way to increase mechanicaly the resolution.

There is something else also. When I noticed step losses teensy board was not completely optical isolated from relays and when spindle motors were toggling on or off there was an unusual behavor to the microcontroler, probably from inductive current. After optical isolation microcontroller works perfectly. So I have to test accuracy again

By the way, typical step frequencies are :
PullInSpeed:  40
MaxSpeed:    3000
Accelaration: 6000

microstepping 1/4
Pinion resolution: 97.39mm per round  (module:1, Z:31)
Steppers: Nema34 1841oz/in, 5A

luni64, 2019-02-24:
Thanks for the info. Your speed and acceleration settings are very slow, On a T3.2, TeensyStep should not have any issues up to about 100kHz. 
Trying it with the optical isolators is a very good idea. 
Do you have some videos/photos of your machine? I'm always interested in what people are building with the library. 
Best

dimisxls, 2019-02-25:
Good news!

I tested the machine without multiplying steps by 4 and there was no step loss so far. 
With 800 steps per revolution I have 0.12mm resolution witch is more than enough for my application.

Here is a video of the machine:
https://youtu.be/B-9xlkEFI9I

and a photo of teensy:
https://photos.app.goo.gl/2jwP19MeYxTDUE238

luni64, 2019-02-25:
Perfect, the machine looks really good. I like pragmatic ideas like the use of normal hand drill instead of expensive spindles. 
May I post a link to the video on the github page?

dimisxls, 2019-02-25:
Yes sure you can post a link!

xoxota99, 2019-03-04:
The "overrideSpeed" method doesn't seem to actually exist in the released version. Can you release it?

luni64, 2019-03-22:
> The "overrideSpeed" method doesn't seem to actually exist in the released version. Can you release it?
> EDIT: For that matter, the RotateControl class is also absent. Where is the code for this?

Sorry, didn't see the last message. The code is still in the development branch. Just switch to develop here 

[image]

to download or clone the library

luni64, 2019-04-27:
> Yes sure you can post a link!

I finally found some time to setup a TeensyStep web page and added a link to your nice video: [http://127.0.0.1:4000/TeensyStep/applications/cnc_drill/](http://127.0.0.1:4000/TeensyStep/applications/cnc_drill/)

Feel free to send me some more content if you like.

dimisxls, 2019-05-08:
hello luni64,

I installed V2 and everything seems ok. There was an unusal behavor after the installation and the modifications according, but it may not mean anything.

Here is a part of code where the z axis pulls back (V1):
....
motor_4.setTargetAbs(xyzz[5][i]); 
if (motor_4.getPosition() > xyzz[5][i]) {       // axis Z speed
      motor_4.setMaxSpeed(8000);
} else {
      motor_4.setMaxSpeed(12000);
}
 controller1.moveAsync(motor_4);
while (controller1.isRunning()) {
.....

After the installation, z axis instead of going backwards, it was going forwards.
I moved the .setTargetAbs after the .setMaxSpeed and it worked properly.

After:
....
if (motor_4.getPosition() > xyzz[5][i]) {       // axis Z speed
      motor_4.setMaxSpeed(8000);
} else {
      motor_4.setMaxSpeed(12000);
}
motor_4.setTargetAbs(xyzz[5][i]); 
controller1.moveAsync(motor_4);
while (controller1.isRunning()) {
.....

closed 2019-05-08 by dimisxls

luni64, 2019-05-08:
Thanks for testing.  
Possible reason for your issue: Both, setTargetAbs and setMaxSpeed will set the direction pin according to their sign. If you have a negative target and call a "positive" setMaxSpeed later the direction will be overridden. 

Did you have a look at the web page? [https://luni64.github.io/TeensyStep/applications2/015_cnc_drill](https://luni64.github.io/TeensyStep/applications2/015_cnc_drill) Posted a video of your machine. Feel free to add more info if you like. 
Best