luni64/TeensyStep #40: 4 controllers

elfren, 2019-02-28:
Just adding two controllers to the RoseFunctionFollower example makes the steppers run very slowly.  I'm working on merging the RoseFunctionFollower code into existing rose engine code which uses two StepControl controllers, so I do need four controllers.

luni64, 2019-02-28:
Yes, each controller requires one interval timer and the example uses one for calling the tick function. This makes 5 in total and the Teensies only have 4, so that won't work. 

Options:
- Do you really need two additional step controllers? One controller can move up to 10 motors in parallel.
- You do not need to have the controllers in global space. If for example you have a function moving the motors using the StepControllers you can declare them locally in the function. They will be destroyed when the function is left. 
- You can also use a different scheme for calling the tick function. E.g. something like

```c++
//...
elapsedMillis tickStopwatch = 0; 

void loop()
{
  if(tickStopwatch > recalcPeriod)
  {
    tickStopwatch = 0; 
    tick();
  }
  // other code 
}
```
Let me know if you need a worked out example

elfren, 2019-02-28:
Moving StepControllers to functions allowed both RotateControllers to work.  However, when the function exits, Teensy 3.5 crashes.

luni64, 2019-02-28:
Yes, just found that out myself. Already fixing it stay tuned.

luni64, 2019-02-28:
Ok, should work now, some destructors were implemented wrongly

elfren, 2019-02-28:
StepControl is still crashing when it's in the function. It's not crashing when it's global.

luni64, 2019-02-28:
Can you post an example showing this?

elfren, 2019-02-28:
Here's the function.  It creates a saw tooth pattern.  Currently the StepControl is enabled globally.

`/// <summary>
/// DoRec1_Spindle
/// </summary>
/// <comment>
/// </comment>
/// <param name="wavDir">Counterclockwise: -1, Clockwise: 1</param>
/// <returns></returns>
void DoRec1_Spindle(int wavDir)
{
	//StepControl stepController;

	// Set speed and acceleration
	teensyStep_Spindle
		.setMaxSpeed(configRec1_S.maxSpd_Spindle * configMain.speedPercentSpindle_Rec1S * .01)
		.setAcceleration(configRec1_S.accel_Spindle);

	float spindleDegrees = configMain.degrees_Rec1_S / (configMain.waves_Rec1_S * 2);
	int spindleSteps = round((configMain.microsteps_Spindle * configMain.steps360_Spindle * configMain.gearRatio_Spindle) * (spindleDegrees / 360) * wavDir);
	returnSteps_Rec1_S = spindleSteps*configMain.waves_Rec1_S * 2;

	teensyStep_Axis_Z
		.setMaxSpeed(configRec1_S.maxSpd_Axis_Z * configMain.speedPercentAxis_Rec1S * .01)
		.setAcceleration(configRec1_S.accel_Axis_Z);

	float axisDistance = configMain.amplitude_Rec1_S;
	int axisSteps = round((axisDistance / configMain.distancePerRev_AxisZ) * configMain.microsteps_Axis_Z * configMain.steps360_Axis_Z * wavDir);

	digitalWrite(PIN_SPINDLE_ENABLE, LOW); // Enable
	digitalWrite(PIN_AXIS_Z_ENABLE, LOW);  // Enable
	for (int i = 0; i < (configMain.waves_Rec1_S * 2); i++)
	{
		axisSteps *= -1;

		teensyStep_Spindle.setTargetRel(spindleSteps);
		teensyStep_Axis_Z.setTargetRel(axisSteps);

		stepController.moveAsync(teensyStep_Spindle, teensyStep_Axis_Z);

		while (stepController.isRunning())
		{
			// Check for Cancel code  
			if (SerialAvailable() > 0)
			{
				incomingByte = SerialRead(serialId);
				switch (incomingByte)
				{
					case 93: // ] - Stop Sync
					{
						stepController.stop();
						return;
						break;
					}
					default:
					{
						break;
					}
				}
			}

			delay(10);
		}
	}

	// Update Nextion
	Serial1.print("pageRec1_Sp.bt3.pco=0");// The first one may be ignored by Nextion, so resend.
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.print("pageRec1_Sp.va0.val=0");
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.print("pageRec1_Sp.bt3.val=0");
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.print("pageRec1_Sp.bt2.val=0");
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.print("pageRec1_Sp.bt1.val=0");
	Serial1.write(0xff);
	Serial1.write(0xff);
	Serial1.write(0xff);
}`

The full code is here: https://github.com/elfren/RoseEngine_SpindleAndAxis/tree/develop
It uses a Nextion touchscreen to send commands to the Teensy (3.2 or 3.5).  To run the above function send Hex CF 1 or CF-1 (Ascii 207)
https://github.com/elfren/RoseEngine_SpindleAndAxis/blob/develop/Nextion%20Multiple%20Stepper%20Control%20Screen%20Descriptions%20with%20microSD.pdf gives an overview of the Nextion screens.  Page 4 is the screen used for the above function (The Accel and Speed settings have since been moved to another screen).

luni64, 2019-03-01:
To test your code I took your function and set the config to some (probably not reasonable) values, commented out the display stuff and tested it. Here the code (should compile out of the box) 
```c++
#include "Arduino.h"
#include "TeensyStep.h"
#include "Rose4Defines.h"

void DoRec1_Spindle(int);

Stepper teensyStep_Axis_Z(0, 1);
Stepper teensyStep_Spindle(2, 3);

void setup()
{
  while(!Serial);
  Serial.println("start");

  pinMode(LED_BUILTIN, OUTPUT);

  configRec1_S.maxSpd_Spindle = 10000;
  configMain.speedPercentSpindle_Rec1S = 100;
  configRec1_S.accel_Spindle = 10000;
  configMain.degrees_Rec1_S=1000;
  configMain.waves_Rec1_S = 2;
  configMain.microsteps_Spindle = 16;
  configMain.steps360_Spindle = 200;
  configMain.gearRatio_Spindle = 5;

  configRec1_S.maxSpd_Axis_Z = 10000;
  configMain.speedPercentAxis_Rec1S = 100;
  configRec1_S.accel_Axis_Z = 10000;
  configMain.amplitude_Rec1_S = 25;
  configMain.distancePerRev_AxisZ = 3;
  configMain.microsteps_Axis_Z = 8;
  configMain.steps360_Axis_Z = 200;  

  DoRec1_Spindle(1);
  Serial.println("\nI'm still alive\n");

  delay(500);

  DoRec1_Spindle(-1);
  Serial.println("\nI'm still alive\n");
}

void loop()
{
  digitalWriteFast(LED_BUILTIN, !digitalReadFast(LED_BUILTIN));
  delay(500);
}

///------------------------------------------------------------------------
/// DoRec1_Spindle
///
///
/// Counterclockwise: -1, Clockwise: 1
///
void DoRec1_Spindle(int wavDir)
{
  StepControl stepController;
  Serial.printf("spindle Speed: %d\n", (int) (configRec1_S.maxSpd_Spindle * configMain.speedPercentSpindle_Rec1S * .01));
  Serial.printf("spindle Acc: %d\n", configRec1_S.accel_Spindle);

  // Set speed and acceleration
  teensyStep_Spindle
      .setMaxSpeed(configRec1_S.maxSpd_Spindle * configMain.speedPercentSpindle_Rec1S * .01)
      .setAcceleration(configRec1_S.accel_Spindle);

  float spindleDegrees = configMain.degrees_Rec1_S / (configMain.waves_Rec1_S * 2);
  int spindleSteps = round((configMain.microsteps_Spindle * configMain.steps360_Spindle * configMain.gearRatio_Spindle) * (spindleDegrees / 360) * wavDir);  
  returnSteps_Rec1_S = spindleSteps * configMain.waves_Rec1_S * 2;

 Serial.printf("z Speed: %d\n", (int) (configRec1_S.maxSpd_Axis_Z * configMain.speedPercentAxis_Rec1S * .01));
 Serial.printf("z Acc: %d\n", configRec1_S.accel_Axis_Z);
 
  teensyStep_Axis_Z
      .setMaxSpeed(configRec1_S.maxSpd_Axis_Z * configMain.speedPercentAxis_Rec1S * .01)
      .setAcceleration(configRec1_S.accel_Axis_Z);

  float axisDistance = configMain.amplitude_Rec1_S;
  int axisSteps = round((axisDistance / configMain.distancePerRev_AxisZ) * configMain.microsteps_Axis_Z * configMain.steps360_Axis_Z * wavDir);

  digitalWrite(PIN_SPINDLE_ENABLE, LOW); // Enable
  digitalWrite(PIN_AXIS_Z_ENABLE, LOW);  // Enable
  for (int i = 0; i < (configMain.waves_Rec1_S * 2); i++)
  {
    Serial.printf("Wave: %d\n",i);
    axisSteps *= -1;

    Serial.printf("  spindleSTeps: %d\n  axisSteps: %d\n", spindleSteps,axisSteps);
    

    teensyStep_Spindle.setTargetRel(spindleSteps);
    teensyStep_Axis_Z.setTargetRel(axisSteps);

    stepController.moveAsync(teensyStep_Spindle, teensyStep_Axis_Z);

    while (stepController.isRunning())
    {

      if (Serial.available())
      {
        while (Serial.available() > 0)
        {
          Serial.read();
        }
        stepController.stop();
        return;
      }

      // // Check for Cancel code
      // if (Serial.Available() > 0)
      // {
      //   incomingByte = SerialRead(serialId);
      //   switch (incomingByte)
      //   {
      //   case 93: // ] - Stop Sync
      //   {
      //     stepController.stop();
      //     return;
      //     break;
      //   }
      //   default:
      //   {
      //     break;
      //   }
      //   }
    }

    delay(10);
  }
}

// // Update Nextion
// Serial1.print("pageRec1_Sp.bt3.pco=0"); // The first one may be ignored by Nextion, so resend.
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.print("pageRec1_Sp.va0.val=0");
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.print("pageRec1_Sp.bt3.val=0");
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.print("pageRec1_Sp.bt2.val=0");
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.print("pageRec1_Sp.bt1.val=0");
// Serial1.write(0xff);
// Serial1.write(0xff);
// Serial1.write(0xff);
//}
```
Here the output: 
[image]

And here LA traces of the step signals
[image]

[image]

Everything looks pretty normal here. 

Maybe it is a timing issue. Can you run my test code using realistic config settings?

elfren, 2019-03-01:
Your changes worked properly on my machine.  I'm not sure what changed, but the original code is working now as well.  I'll let you know if the problem repros again.

luni64, 2019-03-01:
Perfect, maybe some old *.obj files were compiled in after exchanging the lib. 
Let me know if you find something else. Any feedback helps improving the library.

closed 2019-03-03 by luni64