luni64/TeensyStep #42: Homing motors using limit switches?

xoxota99, 2019-04-14:
This is basically the same question as in https://github.com/luni64/TeensyStep/issues/6 , where it didn't really receive an answer.

At the startup of my system (6DOF robot arm, with 6 steppers), I Would like to "home" the system by moving motors slowly until limit switches are hit for each joint, then moving forward until the limit switch is no longer engaged, at which point I would set the zero position for the motor.

When the limit switch is hit, I want to stop the motor *immediately* (ie: no deceleration, no waiting, etc). But I want to do this independently for each of the six motors, and there doesn't seem to be a way to do that with the StepControl class (emergencyStop(), stop() and stopAsync() affect all motors).

What is the best way to e-Stop (or even just regular stop) a single motor? Do I need to instantiate multiple controllers (one for each motor)?

xoxota99, 2019-04-14:
Can I set a motor's maxspeed to zero, while I wait for the other motors to finish hitting their limits?

luni64, 2019-04-15:
> and there doesn't seem to be a way to do that with the StepControl class (emergencyStop(), stop() and stopAsync() affect all motors).

Yes, there is no possibility to do that. The whole idea of moving multiple motors with one controller is to move them in sync. If you stop one of the synced motors they obviously won't be in sync anymore. 

Of course I do not know your application but usually you'd home the motors one by one.  If for some reason you need to home more than one motor at a time, you can use up to 4 controllers so at least 4 motors can be homed in parallel. If you really NEED to home all 6 motors you can play tricks with pinMode(stepPin, INPUT). Make sure to stop the controller before you enable the step output again.

luni64, 2019-04-15:
> Can I set a motor's maxspeed to zero, while I wait for the other motors to finish hitting their limits?
No, you can not change this during a movement.

xoxota99, 2019-04-15:
Thanks for your response!

"If, for some reason you really need to home all 6 motors in parallel, you can play tricks with pinMode(stepPin, INPUT)"

Do you mean that if I set one of the STEP pins to Input mode, it will stop pulsing?

I think I understand your point about "moving in sync", but should that still apply for calls to rotateAsync(), since there is no target specified?

luni64, 2019-04-15:
> Do you mean that if I set one of the STEP pins to Input mode, it will stop pulsing?

The controller wont stop pulsing but the step pulses don't get output on the pin. This is why I recommended to make sure that the controller is stopped before you re-enable the pin. 

> I think I understand your point about "moving in sync", but should that still apply for calls to rotateAsync(), since there is no target specified?

rotateAsync() is also moving all passed in motors in sync. For rotateAsync it is the motor speeds which stay in sync, even during acceleration/deceleration.

xoxota99, 2019-04-15:
Thanks! Resolving...

closed 2019-04-15 by xoxota99