luni64/TeensyStep #46: moveAsync no longer supports overriding speed. What's the equivalent?

xoxota99, 2019-05-12:
Some recent change has removed the signature for StepControl.moveAsync() with the relative speed parameter (I used to be able to provide a float to move at a speed somewhere between zero and the max motor speed).

What would be the equivalent way of doing this now?

luni64, 2019-05-12:
I assumed that nobody uses this feature so I didn't carry it to V2 to clean up the code. Anyway, if you need it I can easily re-implement it.

luni64, 2019-05-12:
Re-implemented the speed override in the DevTimer branch. Please not that for technical reasons the optional speedOverride parameter is now the first parameter to the move commands. i.e. 

```c++
StepControl sctr;               // construct a controller

  m1                            // setup motor parameters
   .setAcceleration(150000) 
   .setMaxSpeed(25000)
   .setPullInSpeed(200);

  m1.setTargetAbs(15000);       // set the movement target
  sctr.move(0.5f, m1);          // move to the target with 50% speed

  m1.setTargetAbs(0);           // new target
  sctr.move(m1);                // move motor with max speed
```
The code (speed override and pullInSpeed) is not yet tested extensively. Let me know if you find anomalies.

xoxota99, 2019-05-15:
Thanks! Resolving...

closed 2019-05-15 by xoxota99