luni64/TeensyStep #47: Slow Speed direction changes

luni64, 2019-05-14:
The following code does not work in the current devTimer branch

see [https://forum.pjrc.com/threads/54529-TeensyStep-How-to-switch-from-1-speed-to-another?p=205792#post205792](https://forum.pjrc.com/threads/54529-TeensyStep-How-to-switch-from-1-speed-to-another?p=205792#post205792)

```
#include "TeensyStep.h"

Stepper motor(48, 47);
RotateControl controller(5, 1000);

void setup() {
  motor
  .setMaxSpeed(124)
  .setAcceleration(198);

  controller.rotateAsync(motor);
  delay(2000);
  controller.stop();

  motor
  .setMaxSpeed(-124)
  .setAcceleration(198);
  
  controller.rotateAsync(motor);
  delay(2000);
  controller.stop();
}

void loop() {}
```

luni64, 2019-05-14:
1) I can reproduce that and will have a look
2) Sorry, forgot to mention that the setPullInSpeed currently only works for the stepControllers, I'm currently working on implementing it for the rotate controllers

luni64, 2019-05-14:
The first issue shoudl be fixed  (0edee2d5d44093144c5ddbf233a769f35fbf89e1)
Can you give that a try?

jpk73, 2019-05-14:
Yes, I tried and can confirm: it works!

jpk73, 2019-05-14:
But it stops working after a couple of repeats, please try this code:

````#include "TeensyStep.h"

Stepper motor(46, 44);
RotateControl controller(5, 1000);

void setup() {
  for (int i; i <= 10; i++) {         // moves only 4 times instead of 10 times
    motor
    .setMaxSpeed(124)
    .setAcceleration(198);

    controller.rotateAsync(motor);
    delay(1000);
    controller.stopAsync();

    delay(100);
  }
}

void loop() {}

luni64, 2019-05-14:
Can't test it right now bot it looks like you are starting a movement while the old one still runs. Can you do a while(controller.isRunning()) after the stopAsync to make sure it doesn't run anymore when  you start the new movement

jpk73, 2019-05-14:
Yes, exactly: i start movements in the middle of existing moves all the time, in many cases together with changing direction. This worked until now very well, and with the new dev-timer branch also, but only for a couple of times, and after that something crashes. That's why I thought it might be a bug... Otherwise I have to change everything to speedOverride which might be tricky in cases where e-stops are involved especially if I start a motor, e-stop it and automatically move back with a step controller (for a travel target as shown in my previous code examples about e-stop on the teensy forum)...

luni64, 2019-05-15:
> Yes, exactly: i start movements in the middle of existing moves all the time

I see. Looks like you were really lucky if it worked in your application because this always led to undefined behavior. Generally, it will generate unpredictable jumps in the  pulse frequency which will most probably lead to steploss errors. 

Problems related to that and wishes to be able to change the rotation speed
on the fly came up a couple of times. Consequently I added the [overrideSpeed](https://luni64.github.io/TeensyStep/documentation/rotateControl/) function for exactly this purpose. I strongly recommend to use this instead.
**Technical:**
Starting a movement will reserve the required timers (1 PIT + 2FTM channels). They are released when the controller is stopped, or a target move is finished. Since you intercept the stopping process the timers are never released. After doing that for 4 times you run out of Teensy timers and the code crashes. 

I will add some safety code to the move/rotate commands to prevent starting a new movement while the controller is running.

jpk73, 2019-05-15:
I see! For me it would suffice to have the safety code just to re-use reserved timers as long as they are not yet released. Would that be possible?

luni64, 2019-05-15:
>  For me it would suffice to have the safety code just to re-use reserved timers as long as they are not yet released. 

I know, but I don't like it :-) 

l'll **try** the following: 
1) I provide a general and optional error callback. This callback will be invoked whenever an error occurs (start movement while another runs, reserved too much timers, wrong settings, etc..). It is then is up to the user to handle this error. E.g. ignore, try to fix, inform the user halt execution etc.
2) In case the user started a movement while another one runs I can **try** to reuse the current timer and not grab a new one. Since that would only fix the timer allocation problem but not the other problems which will occur in such a case I'll still issue an error via the callback. It's up to you then to ignore this error condition and just continue. 

Would that work for you?

jpk73, 2019-05-15:
> Would that work for you?

Yes, I think this would allow me to use my current code. But please don't change your concept just for my convenience if that is not what you consider it should work like.

I am also looking into your recommendation: in case I go for the speedOverride route I would only need to be sure that in case of an estop I can move the motors right away in the opposite direction but with a step controller without worry not to having enough timers to move back. So maybe a switch for the controllers to ask for a permanent or a temporarily reservation would be another way to give the users flexibility? Or even a release() command...?

luni64, 2019-05-15:
> Even more elegant would be to start the new movement from the actual speed at that moment which would practically result in a speedOverride command

This only looks elegant on a first glance. If you restart the movement how would you handle the situation if more than one motor changed speeds / targets? Handling that will either loose motor synchronization and/or generate dangerous speed jumps for some of the motors.  OverrideSpeed avoids all that complications by doing a relative speed change for **all** attached motors simultaneously which will guarantee a smooth and synchronized acceleration of all the motors to their new speeds. 

> in case I go for the speedOverride route I would only need to be sure that in case of an estop I can move the motors right away in the opposite direction 

I thought I fixed that a couple of days ago? Should work with the code in DevTimers...

> but with a step controller without worry not to having enough timers to move back.

estop will immediately release the timers so the new controller will always find a free timer after the estop.  

> So maybe a switch for the controllers to ask for a permanent or a temporarily reservation would be another way to give the users flexibility? Or even a release() command...?

Now we are back where we started some days ago. This will generate the same mess than constructing and deleting the controllers dynamically. The advantage of the current solution is that the user does not need to take care of those highly technical internals. She can generate as much controllers as is convenient and  -as long as there are enough free timers- it will just work (at least if you don't misuse them :-) ).

jpk73, 2019-05-15:
OK, got it :-) Sorry that I don't get things faster...

jpk73, 2019-05-17:
Hi again... I still struggle to understand how I can do things like shown in [this video](https://www.youtube.com/watch?v=a7u3MZOv1X4) with speedOverride: I arrange manually the winding wires or the position of a motor by toggling a flip switch. The switch has 3 positions, and I handle it so that the motor starts when I toggle the switch to the right or to the left, and if I let it it goes back to the middle and stops the motor. If I to this with speedOverride I would have to permanently reserve a timer. Currently I issue stopAsync with the middle position of the switch. With the master-branch version of TeensyStep this works well because I can interrupt the deceleration and start a new movement right away, but with dev-timer I use up the 4 timers and get stuck... Step loss is no problem for me because my servos seem to keep up well with such situations.

luni64, 2019-05-17:
I understand your problem. Is this some special manual adjustment mode or can the user operate the switch at any unpredictable time? (In the first case you could simply release the timers by stopping or deleting the controller before you leave this mode)

The fix I proposed [above](https://github.com/luni64/TeensyStep/issues/47#issuecomment-492617228) should solve that in principle, but such things should really be possible without smelly tricks.   
Might be best to simply release the timers whenever the motor reaches a target speed of 0 and grab them whenever the motor requests a speed != 0. I don't oversee at the moment if that would have bad side effects but I try to do a few experiments on the weekend.

jpk73, 2019-05-17:
> Is this some special manual adjustment mode or can the user operate the switch at any unpredictable time?

At any unpredictable time except when winding or other automatic stuff happens. But even manually fiddling with the wires includes short winding moves back and forth until I find a good point to start the real winding process.

> Might be best to simply release the timers whenever the motor reaches a target speed of 0

Yes, that was the reason why I asked for a release command: so that I can issue the release when speed goes zero...

luni64, 2019-05-17:
> At any unpredictable time except when winding or other automatic stuff happens. 

Sounds like you have a "manual mode" and a "automatic mode" What I would do then is: 

- Wenn entering the manual mode start a rotation controller and override the speed to 0. 
- Whenever the switch is operated in manual mode simply do a overideSpeed -1 / 0 / 1 depending on the switch position (one of those small thumb joysticks might be even better since you could override the speed according to the pot value). You can do that as often as you like. 
- When you finally leave the manual mode simply do an eStop on the controller, this will immediately release the timers. 

Am I missing something? 
Anyway, I'll try to improve the timer handling and add the error function as described above in any case.

jpk73, 2019-05-17:
> Sounds like you have a "manual mode" and a "automatic mode"

If looking at only one motor I could distinguish between manual and auto and use step controllers for automatic and rotate controllers for manual stuff. But I also have to move other motors prior to winding. I made [a new video](https://www.youtube.com/watch?v=4FwyxwV7yNY) where you can see what's going on before I start the winding process at the end of the video. During filming I can only show the moves sequential, but in real world usage I also have foot switches and start moves with my feet and by hand, and also much faster and/or simultaneously...

closed 2019-10-23 by luni64