luni64/TeensyStep #48: HOWTO: Coordinate non-linear movement with waypoints in a multi-motor system?

xoxota99, 2019-05-15:
I am controlling an X/Y axis machine (similar to 2DOF CNC) with two stepper motors, and would like to be able to support a circular or arc-like movement (think GCode G02 or G03), by setting a series of "Segment" X/Y coordinates, and having the stepper motors move through them in a coordinated fashion.

However, I don't want the controller to accelerate / decelerate between waypoints. I only want that acceleration / deceleration at the beginning of the series, and the end. What would be the best way to go about doing something like this using TeensyStep?

luni64, 2019-05-15:
I have this on my todo list for the next large feature. 
When I re-implemented the pull-in-speed, I also implemented a pull out speed for that purpose. The idea is to somehow precalculate the required pull-in and pull-out speeds in a small motion planner so that the steppers pass the target points with the optimal speed. 

However, this are just ideas at the moment and not tested at all. If you like to do some experiments you need to use the DevTimer branch. Here the (more or less untested) interface:

[image]

xoxota99, 2019-05-16:
Ah, sweet! So If I understand correctly, the pullin speed is the speed above which the controller won't bother decelerating / accelerating into the next movement, correct? So would I need to issue a bunch of setTarget / moveAsync calls, one after the other? Do I need to time these in any way (such as looking for an appropriate current motor position before setting the new target)?

Any plans / ETA for release of the DevTimer branch?

xoxota99, 2019-05-16:
Thanks for all your hard work on this library, by the way. Super easy and powerful to use!

luni64, 2019-05-17:
> So If I understand correctly, the pullin speed is the speed above which the controller won't bother decelerating / accelerating into the next movement, correct?

The pull in speed is the speed the acceleration starts with. I.e. if you request a speed smaller than the pullIn speed the motor would just start with that speed. PullOut Speed is the speed at which the motor stops after decelerating.  Usually you set these to get the motor quicker up to speed. 

> So would I need to issue a bunch of setTarget / moveAsync calls, one after the other? 

Yes and you need to determine the start and stop speed required by both motors to achieve a smooth movement. 

> Any plans / ETA for release of the DevTimer branch?

Not really but it is an interesting problem so I'll definitely will play with it. But I doubt that it will be ready before late summer.  

> Thanks for all your hard work on this library, by the way. Super easy and powerful to use!

Good to hear.  BTW: I'm still looking for applications for the [TeensyStep ](https://luni64.github.io/TeensyStep/) web site. If you have something interesting you want to share I'll be happy to put in the application section.

closed 2020-06-29 by luni64