luni64/TeensyStep #49: Current Speed?

MastaRob, 2019-06-06:
Hi,
Thanks for providing the library!
Looking to use it for continuous rotation, with a pot controlling speed (lots of overrides...). I would like to be have the current speed display on an LCD screen eventually, but for now it would be useful just to test my code. I saw from an update you had posted somewhere that this is possible, but cannot find how?
Thanks for your time!

luni64, 2019-06-07:
Hi, 

sorry that was missing in the class reference.  You can do that with the `int32_t getCurrentSpeed()` member of the controllers. If you attached more than one steppers to the controller it returns the speed of the leading motor. The others always move in the fixed ratio given by the target speed of the motors. 

Here the link to the updated class reference https://luni64.github.io/TeensyStep/documentation/rotateControl/#rotating

MastaRob, 2019-06-07:
Great, thanks Luni! Just one issue, I am seeing 100 when output from the pot is on 0? It's definitely receiving an override (0) at this point, and stepper is not moving. 
Re only update every 50ms or so, good idea I'll implement. I was just running it now with continuous override whenever the pot value changes, but I guess this is pretty pointless.... And nice vid by the way, very impressive!

By the way I think you dropped a ( in the description:

Serial.printf("pos: %d, speed: %d\n", s1.getPosition(), controller.getCurrentSpeed))

luni64, 2019-06-07:
> Great, thanks Luni! Just one issue, I am seeing 100 when output from the pot is on 0? It's definitely receiving an override (0) at this point, and stepper is not moving.

That might be a bug. The function returns the frequency by reading out the reload value from the timer. But an override(0) disables the timer (can't set a reload of infinity) So the returned value is probably the last frequency before stopping the timer. I'll fix that but I'll be traveling the next days. 
Meanwhile you can workaround by something like

``` c++
int32_t speed = controller.isRunning() ? controller.getCurrentFrequency() : 0; 
``` 

> By the way I think you dropped a ( in the description:

That's true,  I'll fix it later.

MastaRob, 2019-06-07:
Nice, makes sense and the workaround does the job - only with controller.getCurrentSpeed() rather than frequency :)

luni64, 2020-06-29:
Added this to the to do list

closed 2020-06-29 by luni64