luni64/TeensyStep #50: Converting from old library to new library

thatrobotguy, 2019-07-11:
Hey
I downloaded your library about 7 months ago and I used it for a project.
About 2 weeks ago, my Ubuntu OS got wiped (attempting to install Arch was a bad idea).
I recloned your repository, and I have now realized you have made some **huge** changes to the library.  I figured this out because the library importing
```
#include <StepControl.h>
```
in my project totally crashes.
I have looked through your tags and found that you still have the original code, which is _great_. Now I am trying to figure out, why did you create a new version of the library? What are the differences between versions? What is so special about the update?
I do not want to say that this library is bad... It has been super useful. I just don't see a migration chart explaining the differences between libraries.

luni64, 2019-07-11:
Hi, 
sorry for the inconvenience. 

> What is so special about the update?

Let me explain: 
A lot of users asked for improvements like the ability to change rotation speed on the fly, support for other architectures, support for different acceleration schemes (s-curve) etc.

Trying to implement some of those requirements quickly showed that the original structure of the library was not fit for those improvements. Especially the combination of the rotational mode and the move to a target mode in one controller was quite limiting. 

After a long time of beta testing and trying out if the new structure is more suitable for future improvements I decided to introduce those breaking API changes in version 2 and keep the old version available and maintained (bug fixes only). 

So, currently from the user point of view there is not much advantage besides the implementation of the fly changes of the rotation speed. But from a technical point of view the change opened up a lot of possibilities for future improvements.

thatrobotguy, 2019-07-12:
Okay, sure: on-the-fly speed regulation required a software overhaul. So, if I am to port to the newest version of the library, these are the changes to make:
1. The import `#incldue <StepControl.h>` is now `#include "TeensyStep.h"`.
2. The object `StepControl<> turntable_controller;` is now ` StepControl turntable_controller;`.

Am I missing anything else? Maybe I was freaking out too much... but seeing a privately managed library all of a sudden not compile right triggers all the alarms in my head. I feel like there is something missing... Anyways, thanks for the quick reply. It appears that I can quickly update my software with the newest version. I should mention that I use the `StepControl` functionality with `Stepper` objects. Anyways, keep up the good work (and correct me if anything I said is wrong). Thanks man

luni64, 2019-07-12:
Yes, that should be all you need to change. But if this is for an existing app which works why don't you simply use the old lib? (Branch V1) 

>  Maybe I was freaking out too much... but seeing a privately managed library all of a sudden not compile right triggers all the alarms in my head.

I fully understand that, had some bad experiences as well. Since then I always try to have self-contained projects with all libraries makefiles etc included. Works out nicely so far.

closed 2019-07-12 by thatrobotguy