luni64/TeensyStep #51: On teensy, pin A9 cannot be usable when I load teensyStepV1

AlphaBlackBee, 2019-07-18:
If I create this main on my teensy, I cannot use A9 pin. Can you say me why?

```
int main(void) {
                Stepper motor(2, 3);
                    StepControl<> controller;
                    //motor.setTargetRel(1000);
                    //controller.move(motor);

                    int pos = 0;
                    uint8_t testAttach = myservo.attach(A9); // Pin 23
                    Serial.print("testAttach : ");
                    Serial.println(testAttach);
                    uint8_t testAttached = myservo.attached();
                    Serial.print("testAttached : ");
                    Serial.println(testAttached);
                    for(pos = 0; pos < 180; pos += 1) { // goes from 0 degrees to 180 degrees, 1 degree steps
                        myservo.write(pos);              // tell servo to go to position in variable 'pos'
                        delay(15);                       // waits 15ms for the servo to reach the position
                    }
                    for(pos = 180; pos>=1; pos-=1) {   // goes from 180 degrees to 0 degrees
                        myservo.write(pos);              // tell servo to go to position in variable 'pos'
                        delay(15);                       // waits 15ms for the servo to reach the position
                    }
                    delay(2000);
}
``` 

But if I delete `StepControl<> controller;` I can.

Thanks for your help !

luni64, 2019-07-18:
Might be a conflict of the used FTM module with analog write. I can have a look if you tell me which Teensy you use.

AlphaBlackBee, 2019-07-18:
Teensy 3.2

luni64, 2019-07-18:
Ok, there is a table on the PJRC page showing which PWM pin uses which timer module https://www.pjrc.com/teensy/td_pulse.html

[image]

Pin 23 requires FTM0 which is also used by TeensyStep. However you can change the Timer TeensyStep uses by editing config.h (https://github.com/luni64/TeensyStep/blob/Version-1/src/TeensyDelay/config.h).  For a T3.2 you can change it to FTM1 or FTM2 (instructions in the file) but since those modules only have 2 channels you will be limited to one controller at a time. Alternatively, if possible, you could use pin 3,4,25,32 for your PWM.

closed 2019-07-22 by luni64