luni64/TeensyStep #52: Support for teensy 4.0

Krys88, 2019-08-07:
high im not sure where to post  this but i would like to know if your library supports teensy 4.0 or will in the future?

luni64, 2019-08-07:
Sure, it should already work. Tested it with the beta version early this year. Need to wait to get one of the series version for final tests.

Krys88, 2019-08-07:
ok how many controllers can we assign for this one?

luni64, 2019-08-07:
Sorry, I was answering from the phone and mixed up repositories (VisualTeensy / TeensyStep). 

TeensyStep is currently _not_ working with the T4. During the tests with the beta unit I found that although the processor is very fast, the peripherals (timer, access to ports etc) were slower than in the T3.6. However, the now released board has an upgraded processor which should fix this but I still need to test it. Might as well be that I need a completely different approach (e.g. polling instead of interrupts) to utilize the speed of the processor. In any case, the migration of the library to the T4 will take quite some time.

closed 2019-08-12 by luni64