luni64/TeensyStep #53: Limit on number of controllers

geraldft, 2019-09-04:
Hi.  I've been writing an app to control 4 or more steppers using random independent graph curves.  This is a bit like a cnc application but quite the same.  I have struck a problem with the apparent limit on 4 controllers.  I need to use a separate rotatecontroller for each motor because of the independent but coordinated movement of these axes.  I also need to run atleast one steppercontroller which is used for positioning of the individual axes when not running them via the rotatecontroller.  This effectively limits me to 3 axes.  Without going into too much detail and alternatives, my question is if it is possible to increase the number of timers so that more than 4 controllers can be used simultaneously?  Thanks

luni64, 2019-09-04:
The number of controllers running at the same time is limited by the number of hardware timers, i.e, to 4. There are however ways to work around this limitation: 

1) You don't need to declare the controllers globally. You can have them as local variable in functions. In your case you could do something like this:   
```c++
void goHome(Stepper& motor_A, Stepper& motor_B)
{
  StepControl localSC; // create the controller locally. Will be released when leaving the function

  motor_A.setTargetAbs(0);
  motor_B.setTargetAbs(0);
  localSC.move(motor_A, motor_B);
}
```
And a similar function for the rotation part of your sketch. 

2) I did an improvement a couple of month ago which only allocates the timer when it really uses it. So you can have an unlimited number of controllers but you can only operate 4 of them in parallel.    
Here a more complete example showing both methods:
```c++
#include "TeensyStep.h"

//-----------------------------------------------------------------------
void goHome(Stepper &motor_A, Stepper &motor_B)
{
  StepControl localSC;

  motor_A
      .setAcceleration(25000)
      .setMaxSpeed(10000)
      .setTargetAbs(0);

  motor_B
      .setAcceleration(25000)
      .setMaxSpeed(10000)
      .setTargetAbs(0);

  localSC.move(motor_A, motor_B);
}

void startMotor(Stepper &m, RotateControl &rc)
{
  m.setMaxSpeed(5000);
  rc.rotateAsync(m);
  delay(500);
  rc.stop();
}

void stopMotor(Stepper &m, RotateControl &rc)
{
  rc.stop();
}

Stepper m1(0, 2), m2(4, 6);

RotateControl rc1, rc2, rc3, rc4, rc5;
StepControl sc1, sc2, sc3, sc4, sc5;

void setup()
{
  m1
      .setMaxSpeed(2000)
      .setAcceleration(2000);

  m2
      .setMaxSpeed(2000)
      .setAcceleration(2000);

  m1.setTargetRel(2000);
  sc1.moveAsync(m1);

  m2.setTargetRel(100);
  sc2.move(m2);

  m2.setTargetRel(500);
  sc3.moveAsync(m2);
  delay(300);
  sc3.stop();

  m2.setTargetRel(100);
  sc4.move(m2);

  m2.setTargetRel(100);
  sc5.move(m2);

  while (sc1.isRunning()) ;

  goHome(m1, m2);

  //------------
  delay(100);
  startMotor(m1, rc5);
  startMotor(m2, rc1);

  stopMotor(m1, rc5);
  stopMotor(m2, rc1);
}

void loop() {}
```
**Please note:**  You need to use the library from the devTimer branch to use it. (I didn't find time yet to merge it into the main branch)

geraldft, 2019-09-04:
Thanks for the detailed explanation.  So theoretically I can do what I need with a limit of 4 motors. 

Initially I have some basic questions.  

It seems you are saying I can have more than 4 controllers declared globally or within a function, so long as only use 4 at a time?  In your sample code setup() I see 5 controllers declared with the fifth sc5 only called after sc3 has been stopped.  I presume that using sc3.stop() ensured that controller was released?  Is there another way of checking that there is a free timer?

You also mention this is only possible if I use the "devTimer" branch?)  I hope you don't mind the innocent question : How do I do that?  (I have the V2 Teensystepper code installed.  )

Finally - any chance in future to add one or two more timers?

luni64, 2019-09-04:
> It seems you are saying I can have more than 4 controllers declared globally or within a function, so long as only use 4 at a time? 

Yes, by "use" I mean the controller is actually driving a motor. I.e. if isRunning returns true.

> In your sample code setup() I see 5 controllers declared with the fifth sc5 only called after sc3 has been stopped. I presume that using sc3.stop() ensured that controller was released? 

This code is a couple of month old, I don't recall why I placed that stop() here. Probably without any reason. 

> Also the ".isrunning" query can confirm if it is free?

Yes. 

Personally I prefer declaring the controllers locally in the functions. (Encapsulation is always a good idea) 

> You also mention this is only possible if I use the "devTimer" branch?) I hope you don't mind the innocent question : How do I do that? (I have the V2 Teensystepper code installed. )

Just switch to the other branch using the shown dropdown and download as you are used to. (Consider to use GitHub Desktop which makes those things much easier, you can switch on the fly, get notice of new versions etc)

[image]

> Finally - any chance in future to add one or two more timers - or because they are "hardware timers" that is limited by the Teensy hardware?

No, the timers are built into the processor chip. So unless I redesign the chip I can't change that :-)

geraldft, 2019-09-04:
Fabulous thanks.  I'm a bit of a virgin with Github...

Re the encapsulation, I agree that seems ideal.  I may use a combination, such as the rotation controllers global and others local - then I can check if one of the rotation controllers is still busy from a separate function.  Does that sound fair?  Or do you think another way of doing that is preferable?

To put it in context, the steppercontrollers will only be used occasionally for isolated events such as moving axes to a position, which can't be done with the rotation controller.

geraldft, 2019-09-04:
Ok.  

I had a quick experiment but the devTimer library is not working properly for me.  Maybe I am overlooking something, but I couldn't get it respond to a locally declared controller.  Then I noticed some other problems, even when not using a local controller, just running my code which was working previously, and which uses only global controllers.  Basically what the program does is allow me to  jog an axis back and forth..  I am sending via serial to the Teensy, and when I press a jog button on my application it sends a code to identify the axis and a positive or negative value  :

speedFactor = receivedint0 / 20000.0;
  Rctl_0.rotateAsync(Mtr_0);
    Rctl_0.overrideSpeed(speedFactor);

Then when I release the button it sends a stop signal like this:

Rctl_0.overrideSpeed(0); 

It also sends serial data back to my application to show the motor position.  I can jog like this back and forth all day with the previous library, but the new devTimer on it ceases to respond after a half dozen or so times pressing the button.  The Teensy is still running at this stage and I can zero the position of an axis, but I can no longer jog it.

luni64, 2019-09-04:
Since this branch is still under development it definitely can be that there are bugs in it. Would it be possible to generate / post a minimal sketch showing the behaviour so that I can try to fix  it?

geraldft, 2019-09-05:
[2 chnl Teensy Serial debug.zip](https://github.com/luni64/TeensyStep/files/3578045/2.chnl.Teensy.Serial.debug.zip)

Simplified Windows app and ino file which shows a bug using devTimer version.    You can jog ok initially and zero axes.  However as soon as you make a goto position move, the jog fails to respond.  Goto position continues to work however.

This is not exactly the same bug as I saw earlier but will have to do, and it is fairly clear when changing between devKit and previous version where it worked fine.  I intend to play more with creating multiple controllers locally, but will move on to that once I have more time.

luni64, 2019-09-05:
I downloaded your files and everything seems to work (Teensy 3.2). How long do I need to play with the jog until the problem usually appears?

luni64, 2019-09-05:
Tried now for about 5min. I'm not able to crash it. 

Looking at your code (nicely written by the way)  only see a few possible problems: 

1) You should definitely avoid calling an move / rotate command while the motor is moving. If you send a G0 command while the motor is still moving you might do exactly that. 
```c++
if (letter0 == "G0")
   {
      //Rctl_0.stop();
      moveto = receivedint0;
      Mtr_0.setTargetAbs(moveto);
      Sctl_0.moveAsync(Mtr_0);
      counting = true;
      letter0 = "qq";
   }
```
The new code is more sensitive to that because it will allocate a new timer when you start the movement. If the old controller is still running and you might run out of timers. Best would be to make sure that the motor stands still right in front of the setTargetAbs command. 

2) You are using Strings for your command parser. This may work (probably) but could generate problems due to heap fragmentation. (But I don't really believe that this will happen here)

3) Remark: (not related to your problem) A lot of people are not aware that Serial doesn't need a baudrate setting. It will alwas transmit with USB speed. In fact you don't need Serial.begin() at all. 
Here the interesting defintion of Serial.begin()
```c++
class usb_serial_class : public Stream
{
public:
	constexpr usb_serial_class() {}
        void begin(long) {
		//uint32_t millis_begin = systick_millis_count;
		//disabled for now - causes more trouble than it solves?
		//while (!(*this)) {
			// wait up to 2.5 seconds for Arduino Serial Monitor
			// Yes, this is a long time, but some Windows systems open
			// the port very slowly.  This wait allows programs for
			// Arduino Uno to "just work" (without forcing a reboot when
			// the port is opened), and when no PC is connected the user's
			// sketch still gets to run normally after this wait time.
			//if ((uint32_t)(systick_millis_count - millis_begin) > 2500) break;
		//}
	}
```

geraldft, 2019-09-06:
Hi thanks

I think you might still be thinking of the bug I mentioned earlier.  The code sample has a slightly different bug which I mentioned in the comment.   I said the bug happens after issuing a goto command.  ie. with cursor in the goto position input press the Enter key.  Motor will move to the commanded position, but after that the jogging will no longer respond. :(

The existing code is really just for R&D so it is a bit quick and dirty variety.  I will likely not use strings, and code structure will be developed to prevent cross purpose problems.  

But also if I use local declarations it is going to have to be quite different.

I already found for example that using local declaration inside for the goto if() statement has a problem.  ie. when the statement is exited, the motor will stop moving even if it has not completed its move to position...  :(  So I will have to make sure it remains inside the statement - which currently means the counting (feedback) is blocked and no other goto or jogging can be performed during this time.

So I can see it getting a lot more complicated...

luni64, 2019-09-06:
> I think you might still be thinking of the bug I mentioned earlier. The code sample has a slightly different bug which I mentioned in the comment. I said the bug happens after issuing a goto [...]

When I wrote the answer I did think of the other bug indeed, but after rereading I tested the Go command as well. Works without problems. I can even jog while it moves and I can have  both motors moving. Nothing crashes. 

BTW: I didn't bother to connect motors but looked at the display only hope that shows the error as well?

Which board are you using? I used a T3.2 for the tests.

Are you sure that you link in the correct branch? In case you use the Arduino IDE it often keeps cached object code around. To check: the devTimer branch  implements a setPullInOutSpeed function which is not there in the master branch. If a call to this function compiles you should be good.

[image]

>I already found for example that using local declaration inside for the goto if() statement has a problem. ie. when the statement is exited, the motor will stop moving even if it has not completed its move to position... 

That's clear. If you leave a function all local variables will be deconstructed so the local controller does not exist anymore and can't move your motors. In your application where you don't finalize the movements within one function a local controller is not the right tool.

geraldft, 2019-09-06:
I must not have the devTimer installed correctly as that function returns "no matching function" error.  
I removed the previous TeensyLibrary folder from my Arduino libraries folder and replaced it with the devTimer library folder as downloaded.  Is there something else special I need to do to install it?

luni64, 2019-09-06:
Since Arduino is caching quite agressively I'd delete all the object files. They are somewhere at %temp% if IIRC google for arduino build folder.
 Need to leave now, will be available in the evening.

geraldft, 2019-09-06:
I messed around using the Arduino library manager as well as deleting what looked like possible cache files in User\LoginName\AppData\Local\Arduino15\ *.json    

Searching for "Arduino Build Folder" lead me to the folder where the sketch's compiled hex file goes.  Didn't seem to be what I'm looking for...   so still no joy...  no compile of setPullInSpeed...

BTW  I am using Teensy V3.6.  Maybe that's the problem?

Thanks

luni64, 2019-09-06:
They are in this path:
C:\Users\YOUR_USER_NAME\AppData\Local\Temp
You can delete all the arduino_cache  and arduino_build folders. You might need to restart Arduino after that. It really is a shame that there is no "clean" function in the IDE.

Please check also that you have the right sources installed in the lib folder. 
..\Arduino\libraries\TeensyStep-DevTimer\src\stepper.h
[image]

Regarding the T3.6. Very unlikely but I'll try in the evening

geraldft, 2019-09-06:
Arduino is being very belligerent I'm afraid.  I deleted those files but made no difference.  Under Arduino library manager it lets you back track to older versions of installed libraries but there is no option to remove or replace - which seems what is really needed in this case...  Short of completely uninstalling and re-installing Arduino I'm a bit stuck.  Then maybe even that won't work!  I am getting ready to travel now - so this will have go on the backburner...  :(

luni64, 2019-09-06:
:-) This IDE is perfect for giving newbies a quick start but it definitely is not suited for serious work. Hate to advertise myself but since you are under Windows (?) you could give [VisualTeensy ](https://github.com/luni64/VisualTeensy) a try. Removes a lot of headaches 

Anyway: Did you check the sources as I wrote above? Maybe you simply copied the wrong ones?

geraldft, 2019-09-06:
Thanks, I'll check out Visual Teensy when I get a chance.

Yes the source was in the folder.  I used this line to call it.

Mtr_0.setPullInOutSpeed(1000);

Ah, I just realised a mistake - that line needs two values.  So I tried this :

  Mtr_0.setPullInSpeed(1000);

And now it compiles.  But does it still have the bug?  

Yes.  Although it is only the first axis that is affected (will not jog).

luni64, 2019-09-06:
> Ah, I just realised a mistake - that line needs two values. So I tried this :+1:  
Mtr_0.setPullInSpeed(1000);

Sure, but this function is there in both branches so it doesn't help to check if you have the right branch installed. 
I'll add a version command to make distinguishing easier.

luni64, 2019-09-06:
I added version information to TeensyStep. You can find out which version/branch you currently use by

```c++
#include "TeensyStep.h"

void setup()
{
    while(!Serial);

    Serial.printf("Version: %s\n", TeensyStep::version);
    Serial.printf("Branch:  %s\n", TeensyStep::branch);
}

void loop(){}
```

Should print out:

[image]

luni64, 2019-09-06:
Tested everything with a T3.6. Works perfectly. Something seems to be wrong at your side...

geraldft, 2019-09-07:
get_version_sep07a:8: error: 'TeensyStep' has not been declared

Am I supposed to download and install a new version of devTimer as well?

luni64, 2019-09-07:
> So now I get the response Ver2.01 devTimer.
Good, at least one unknown gone...

Tested the T3.6 already (see above). 

Good idea with the IDE. Here my Setup:
- Arduino 1.8.9
- Teensyduino:  1.4.7 
- CPU Speed: 180MHz
- Optimize: Faster

Testing: 
1) Enter a value (5000) into the GoTo box
2) Wait until the display reaches the target (I have no motors connected)
3) Try to jog afterwards

This works perfectly. Repeated with different targets, first or second motor etc.

luni64, 2019-09-07:
Just to make sure i recorded the generated signals (stp / dir)

Performed Sequence: 
Goto 20000
Jog left
Jog right
Goto 0

[image]

Measured # of pulses: 
+20000 -6654 +5553 -18899 = 0
So, that also seems to work...

geraldft, 2019-09-07:
I have the same.    
    Arduino 1.8.9
    Teensyduino: 1.4.7
    CPU Speed: 180MHz
    Optimize: Faster
So I have to say I am just giving up on the devTimer version.  It does not work for me and I can't see why.  I will continue with the previous version for now.  Can you maybe explain again exactly what the deVTimer offers in addition?  Particularly with reference to declaring controllers on the run.  Thanks G

luni64, 2019-09-07:
> So I have to say I am just giving up on the devTimer version

I'd really like to nail this down but I do understand that you don't want to spend more time on it. 

> Can you maybe explain again exactly what the deVTimer offers in addition?

The main new feature in the devTimer version is the on the fly assignment of the timers. This allows you to have as much controllers constructed as you want as long as you don't have more than 4 controllers actually moving motors at the same time. This would be a perfect fit for your application since you could have the 4 rotational controllers as well as 4 step controllers living in global space. 

Current plan is to merge the devTimer branch into the master branch when I did the adaptions to the T4.

luni64, 2019-09-07:
Just in case I can talk you into a last test: 

I thought maybe the command interface plays tricks (EMC noise etc) Can you try the version below? I added code to toggle the LED on received commands and to go into panic mode (fast blinking LED) if you start a movement while one is running. This should not happen as long as you don't Jog while the motor is in a Goto movement. 

```c++
#include "TeensyStep.h"

int currentPos;

float period = 100; //interval between counts in milliseconds
long counter = 0; //increment counter
unsigned long startMillis;
unsigned long currentMillis;
boolean counting = true;

float  speedFactor;
float  accelFactor;
float  moveto;

// stuff needed for SERIAL to work
String sendstr, pref;

String inData; //everything from Serial comes to this variable to then be processed.
String receptacle0, receptacle1;// this variable is used to store data that has been stripped from any letters, keeping numbers (in String form).
String letter0, letter1;//this variable holds the first letter of the incoming data.This first letter determines where to send the data.
float receivedint0, receivedint1; // Set value
//-------------------------------

RotateControl Rctl_0;
RotateControl Rctl_1;

Stepper Mtr_0(0, 1);  // STEP pin: 0, DIR pin: 1
Stepper Mtr_1(2, 3);
Stepper Mtr_2(4, 5);
Stepper Mtr_3(6, 7);

StepControl Sctl_0;
StepControl Sctl_1;

bool isSomethingRunning()
{
  return Sctl_0.isRunning() || Sctl_1.isRunning() || Rctl_0.isRunning() || Rctl_1.isRunning();
}

void panic()
  {
    while(1)
    {
      digitalWriteFast(LED_BUILTIN, !digitalReadFast(LED_BUILTIN));
      delay(50);
    }
  }

void setup() {

  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(57600);
  Serial.println("");

  Mtr_0.setMaxSpeed(20000);
  Mtr_0.setAcceleration(20000);
  Mtr_1.setMaxSpeed(20000);
  Mtr_1.setAcceleration(20000);
  Mtr_2.setMaxSpeed(20000);
  Mtr_2.setAcceleration(20000);
  Mtr_3.setMaxSpeed(20000);
  Mtr_3.setAcceleration(20000);

  Rctl_0.rotateAsync(Mtr_0);
[... 120 lines cut ...]
    Rctl_1.overrideSpeed(speedFactor);
    counting = true;
    letter1 = "qq";
  }

  //-------------------------------------------------------------

  if (counting) {

    if (currentMillis - startMillis >= period) {
      counter++;

            int p0 = Mtr_0.getPosition();
            int p1 = Mtr_1.getPosition();
            int p2 = Mtr_2.getPosition();
            int p3 = Mtr_3.getPosition();

      Serial.printf("P0%i,\P1%i,\P2%i,\P3%i\n", p0, p1, p2, p3);

      startMillis = currentMillis;
    }//end current millis
  }//end counting

}//end loop
```

geraldft, 2019-09-08:
Thanks for your patience.  I am on vacation for the next few weeks, in Belgium of all places and later in the Balkans..  Despite that I hope I can still find some time to do a little test.  I don't have the same computer with me, but I have the Teensy, so we will see...

geraldft, 2019-09-12:
Hi Lutz.  Thank you for the extra effort.    I installed Arduino on my travel computer and retried the code.  Surprise it now works!  Very strange.  Both are Win7 but maybe the other computer has more "history" whereas this one is a fresh installation.  Also possible the process I went through of changing from the previous teensystepper library to the devTimer messed things up?  Anyway, now I can explore using local controllers...  in my spare time when not travelling...

closed 2019-10-23 by luni64