luni64/TeensyStep #54: Can go higher than 65 k steps/s

EtienneGameSeed, 2019-09-10:
Hi,
Thanks for this library, which seems very promising.
I'm doing tests to see if I can indeed reach the maximum speed of 300 000 steps/s on an T3.6 @240 MHz, with this simple code :

`#include "TeensyStep.h"

Stepper motor(10,9);       // STEP pin: 2, DIR pin: 3
StepControl controller(1);    // Use default settings 

void setup()
{
  motor.setMaxSpeed(300000);
  motor.setAcceleration(500000);
}

void loop() 
{
  motor.setTargetRel(10000);  // Set target position to 1000 steps from current position
  controller.move(motor);    // Do the move
  delay(500);
}`

However it seems the pulse frequency never go beyond 65 000 steps/s (I'm using a logic analyser which grabs signals at 500 Mhz).
Is it a bug or am I doing something wrong ?

luni64, 2019-09-10:
That's clear. It can't reach the target speed within that short movement range. Here the relevant formulas for an accelerated movement

v^2 = 2 as  
(where v = speed (step/s), a = acceleration (step/s^2) and a = distance (steps))

Since you need to  decelerate to zero after the acceleration you only have 5000 steps for the acceleration and get:

v = sqrt(2* 500000 steps/s^2 * 5000 steps) = 70710 steps/s  which corresponds to your observation. 

To reach the max speed you need an acceleration distance of at least: 

s = v^2 / 2a = 500000^2 / (2*300000) = 90000 steps

So you need to change the target to at least 180000

Hope that helps

EtienneGameSeed, 2019-09-10:
Thanks for the quick reply,
I just tried with a target of 180000, but it does the same thing.
Tried with a much bigger target  (400000), it reaches 300000 steps/s, but during the acceleration phase, the rates rises up to 65 Khz and then goes back to 8.22 Khz, several times, instead of rising from 0 to 300 Khz.
Same for the deceleration phase.
Is there an easy way to disable acceleration system ? (I'm planning to drive servo motors with position feedback, so I won't loose steps).

luni64, 2019-09-10:
> I just tried with a target of 180000, but it does the same thing. Tried with a much bigger target (400000), it reaches 300000 steps/s, but during the acceleration phase, the rates rises up to 65 Khz and then goes back to 8.22 Khz, several times, instead of rising from 0 to 300 Khz.

Both observations should't happen (and didn't happen when I tried high speed movements last time). I'll have a look at it later today. I assume you used the sketch from above for the experiment?

> Is there an easy way to disable acceleration system ? (I'm planning to drive servo motors with position feedback, so I won't loose steps).

You can give the code in the DevTimer branch a try. It has a SetPullInSpeed setting on the stepper class. This setting defines the starting speed of the stepper. If you set this equal to the target speed the motor should start with the target speed. Note: the branch and this feature is not yet tested very thoroughly but you can certainly try if it works for you.

EtienneGameSeed, 2019-09-10:
Yes I use the above sketch, with Arduino IDE 1.8.9 and TeensyDuino 1.47.

I could not find any reference to SetPullInSpeed, only a call in the MultipleStepper example, where it is commented (comment says it's currently deactivated...)

luni64, 2019-09-10:
Tested it with your sketch. Using a T3.6 @240MHz. The only thing I changed was the targetPosRel which I set to 180000. Here a few screenshots from my LA

1) Overview, showing that it makes 180000 steps indeed:
[image]

2) Same data, zoomed in the center to show the max frequency:
[image]

3) Start of the movement, don't see any irregularities
[image]

4) Quickly scrolled through the data, don't see anything you observed....

Seems to work as expected.

EtienneGameSeed, 2019-09-10:
Ok, thank you very much for checking.
Is it with the Master branch that you did these tests ?

luni64, 2019-09-11:
I did the test with the DevTimer branch. Tried it with the master branch now which does indeed show the strange behaviour you describe.  **Thanks for finding that!**

A quick check shows that this bug only occurs with very high accelerations/speeds which is probably the reason why nobody stumbled over it (or cared to report) so far. 

Besides other changes regarding the timer handling the DevTimer branch also includes some cleanup in the acceleration code which obviously fixed the issue by chance - usually it's the other way round :-)) 
I'll try to do some more tests on the weekend and pull the DevTimer changes branch into the main branch. 

Please let me know if you find something else

EtienneGameSeed, 2019-09-11:
Great news ! I'm glad I could help. 
The work you've done with the timers really amazing.
I'll let you know if I found something else.

closed 2019-10-03 by luni64