luni64/TeensyStep #56: Support for Teensy 4.0?

xoxota99, 2019-09-15:
Teensy 4.0 is out now! Any chance you'll consider supporting it?

luni64, 2019-09-16:
Sure, I'm currently working on getting the T4 timers tamed, as soon as this is finished I can use them for TeensyStep. Will take some time though.

shaikez, 2019-11-06:
Hey luni64!
I made a controller module using Teensy 3.2 for four stepper motor using your library.
The controller works perfectly, thanks for the library :).

Further I wanted to add a current control but I left with only 2 usable pwm pins (As I use FTM1 for CAN communication I was left with only FTM2) and Teensy 4.0 looks like a great solution.

When do you think the library will be ready for use with Teensy 4.0?

PMTaylor, 2019-12-30:
I'm stoked on seeing how much faster TeensyStep will be on T4 too.
Looking to have high-ish speed/resolution Step/Dir commands to industrial servo motor controllers. 
Working on a hardware solution with TMC4361a now, but very much curious if teensystep is a smarter move. I don't expect it to go as fast as dedicated hardware, but it'll be easier to control, and cheaper.

I don't think i can be of much help to you luni64, but if you want some help, lemme know...

luni64, 2019-12-30:
Unfortunately the implementation of the timer peripherals on the T4 processor is 'a bit' suboptimal. I.e. despite its fast ARM core the processor is not even able to use all 4 PIT timers to generate 100kHz signals at the same time. Reason is, that the peripherals run on a separate bus and synchronizing this with the ARM core is a very slow process... So, unfortunately don't expect any TeensyStep performance increase with the T4.

PMTaylor, 2019-12-30:
Thanks for the update. That's kind of a bummer, but not surprising considering the delay in this thread/issue.
Thanks for the hard work!

1DropDiag, 2020-01-15:
> Unfortunately the implementation of the timer peripherals on the T4 processor is 'a bit' suboptimal. I.e. despite its fast ARM core the processor is not even able to use all 4 PIT timers to generate [...]

Hello
I own a T4 and I'm using it with AccellStepper but I'd like to make the change to TeensyStep soon. Do you have an ETA for the support of T4 ?

You also mention that the timer implementation on the T4 is suboptimal. Is this the case on the T3.6 as well? Would I get better performance with a 3.6 at high speeds for 3 motors or will the T4 show similar performance once it can run TeensyStep?

Thanks!!

luni64, 2020-01-15:
I published [TeensyTimerTool](https://github.com/luni64/TeensyTimerTool)  yesterday which is supposed to be the new timer base for TeensyStep. I don't have a concrete ETA so far. 

>You also mention that the timer implementation on the T4 is suboptimal. Is this the case on the T3.6 as well? 

No, the T3.6 has much better timer support. I suspect that the T4 will perform not as good as a T3.6 in high speed  applications.  What is your target speed for the 3 motors?

1DropDiag, 2020-01-16:
Thanks a lot for the quick answer.

I will run the motors in microstepping with 51200 steps per revolution and I would like to have 3.5 rotations per second ideally, which leads us to ~150000-250000 steps/s for the speeds.

luni64, 2020-01-16:
Thats fast indeed. I'm afraid that the T4 is not the right board for that. In the table shown here https://forum.pjrc.com/threads/57959-Teensy-4-IntervalTimer-Max-Speed?p=218577&viewfull=1#post218577
you see the generated load for just flipping a pin @100kHz and 4 interval timers. If it even works at 250kHz 3 timers it will be borderline...

T3.6 should be fine, If you want I can give it a try later today.

1DropDiag, 2020-01-16:
Amazing. I willl try to get a T3.6 then.

Do you ming if I ask you 2 questions:
- I currently monitor limit switches during a movement using moveAsync followed by a 
`while(controller.isRunning()){
    if (digitalRead(15)){
        controller.stop();
    }
}`
Is there an other way?
Is it possible to stop only one motor of the array and keep the other finish the movement started by moveAsync?

Cheers

luni64, 2020-01-16:
I meanwhile did some tests with a T3.6@240MHz optimized to fastest.
```c++
#include "Arduino.h"
#include "TeensyStep.h"

Stepper       m1(1, 5), m2(2, 6), m3(3, 7);
RotateControl rctl1(1), rctl2(1), rctl3(1);

void setup()
{
   while (!Serial) ;   

   m1.setMaxSpeed(200'000).setAcceleration(250'000);
   m2.setMaxSpeed(190'000).setAcceleration(250'000);
   m3.setMaxSpeed(180'000).setAcceleration(250'000);

   rctl1.rotateAsync(m1);  
   rctl2.rotateAsync(m2);
   rctl3.rotateAsync(m3);
}
void loop()
{      
    Serial.printf("s1: %d, v1:%d\n", m1.getPosition(), rctl1.getCurrentSpeed());
    delay(100);
}
```
Here the result. Traces show m1 (@200khz) m2(190kHz) and m3 (180kHz). Motors are driven by 3 independent rotation controllers. You clearly see instabilities in the pulse witdh but the overall frequency looks good. 250MHz is quite borderline. 

[image]   

Here the result for all 3 motors at 150kHz
[image]

>I currently monitor limit switches during a movement using moveAsync followed by a
while(controller.isRunning()){ if (digitalRead(15)){ controller.stop(); } }
Is there an other way?

Depends on why you monitor them. 
- If this is for safety, i.e. you want to stop the motors if you touch a switch then you usually attach an interrupt to the limit switch and in the handler you'd call controller.emergencyStop(). You then don't need to monitor the limit switch like you did. 
- If you want to 'home' the motors the method you used is OK in principle. However, depending on the required accuracy you want to implement a more stable procedure. Something like 
  - Drive in direction of the switch and do a normal stop if you pass it
  - Slowly move out of the switch and record the counter when the switch is left. The counter value can then be used as offset. 

>Is it possible to stop only one motor of the array and keep the other finish the movement started by moveAsync?

No, the whole point of the multi motor movement is to keep them synchronized. If you want to stop them separately you can use one controller for each motor. (See example sketch above)

kusstech, 2020-03-01:
Hi luni, do you have any idea when its gonna be a version of teensystep for teensy 4.0 ? thanks a lott

xarple, 2020-03-07:
Hi, I just did a simple pin switch testing on a Teensy 4.0, in my situation it seems that T4.0 can be easily generate high frequency pulse, below is a result of generating 1MHz signals on eight channels, the shown graph looks very promising.

`#include <Arduino.h>

IntervalTimer timer;
#define TEST_PIN_1 2
#define TEST_PIN_2 3
#define TEST_PIN_3 4
#define TEST_PIN_4 5
#define TEST_PIN_5 6
#define TEST_PIN_6 7
#define TEST_PIN_7 8
#define TEST_PIN_8 9

void onTimer() {
  digitalWriteFast(TEST_PIN_1, HIGH);
  digitalWriteFast(TEST_PIN_2, HIGH);
  digitalWriteFast(TEST_PIN_3, HIGH);
  digitalWriteFast(TEST_PIN_4, HIGH);
  digitalWriteFast(TEST_PIN_5, HIGH);
  digitalWriteFast(TEST_PIN_6, HIGH);
  digitalWriteFast(TEST_PIN_7, HIGH);
  digitalWriteFast(TEST_PIN_8, HIGH);
  delayNanoseconds(250);
  digitalWriteFast(TEST_PIN_1, LOW);
  digitalWriteFast(TEST_PIN_2, LOW);
  digitalWriteFast(TEST_PIN_3, LOW);
  digitalWriteFast(TEST_PIN_4, LOW);
  digitalWriteFast(TEST_PIN_5, LOW);
  digitalWriteFast(TEST_PIN_6, LOW);
  digitalWriteFast(TEST_PIN_7, LOW);
  digitalWriteFast(TEST_PIN_8, LOW);
}

void setup() {
  pinMode(TEST_PIN_1, OUTPUT);
  pinMode(TEST_PIN_2, OUTPUT);
  pinMode(TEST_PIN_3, OUTPUT);
  pinMode(TEST_PIN_4, OUTPUT);
  pinMode(TEST_PIN_5, OUTPUT);
  pinMode(TEST_PIN_6, OUTPUT);
  pinMode(TEST_PIN_7, OUTPUT);
  pinMode(TEST_PIN_8, OUTPUT);

  timer.begin(onTimer, 1);
}

void loop() {

}`

[image]

luni64, 2020-03-07:
Teensy 4 is able to flip pins at about 150MHz in a loop.  Problem is, that the timers are not coupled to the core as tight as the GPIO is. That generates a huge amount of sync waiting. Try to use 4 Interval timers, each flipping one pin and increase the frequency. You'll see that you can not do much more than 500kHz before the T4 overloads. Changing the timer clock to 150MHz improves the situation but it is still not better than the T3.6. 

I'm working on adapting the lib to T4, but don't expect any better performance compared to T3.x Progress is slow due to other higher priority projects....

kusstech, 2020-03-08:
> Teensy 4 is able to flip pins at about 150MHz in a loop. Problem is, that the timers are not coupled to the core as tight as the GPIO is. That generates a huge amount of sync waiting. Try to use 4 [...]

Coool! i think , to be able to use the teensystep like the 3.6 on teensy 4.0 it will be amazing, now i have the T4 but i could not use it with teensy step because doesn't compile.

closed 2020-06-29 by luni64

coltonddlabs, 2022-07-16:
Hello I was wondering if you could provide an update on Teensystep library to support Teensy4.0 boards @luni64 .  Thanks!