luni64/TeensyStep #57: How to get the motor moves accordingly to an analog input?

christophepersoz, 2019-09-30:
Hello,

After looking at the available functions, I'm not sure that there is a way to change in real-time the speed of a motor, weather it is in Sync or Async mode. Am I correct? If yes, is there a way to achieve this?

Thanks

luni64, 2019-10-01:
You can use overrideSpeed for changing the speed on the fly. See here 
https://luni64.github.io/TeensyStep/documentation/rotateControl/ for a command description and here
https://luni64.github.io/TeensyStep/applications2/010_winder/winder for an example making heavy use of speed overriding

yuranikspb, 2019-10-08:
Hello Luni64!
setTargetAbs not recalculate accleration and speed ?
i want to set target to motor randomly time;
example:
first target 10000 steps ( time to travel 5s with accelration )
but after 3 seconds from the start of the movement, a new target is set 2000
the motor should brake and return to 3000 
how to do it ?
accelstepper library it supports

{
 stepper.moveTo(analog_in);
 stepper.run();
}

luni64, 2019-10-08:
Changing the target during a move is not possible at the moment. It would be simple for one motor but  TeensyStep is designed for moving up to 10 motors to a predefined target synchronously. If you change one of the targets what should happen to the other motors? Loosing sync? Staying synced and change the targets respectively?  

IIRC Accelstepper can only move one motor at a time? Does the same work for its companion class mulitstepper? 

Anyway, if you tell me what you finally want to achieve I might be able to make a suggestion how to solve your problem. 

> Link to "Winder class" not work .

I assume you refer to the link to the GitHub repository from the web site. This is broken indeed. 
Here the correct link; https://github.com/luni64/TeensyStep/tree/master/examples/Applications/Winder

christophepersoz, 2019-10-08:
Hi Luna64,

Indeed overrideSpeed seems to do the job, thanks!

christophepersoz, 2019-10-08:
The fact is, Teensystepper is build to move xx motors in Sync or Async movement; but to a known destination. No problem until that. But, prior to do a move, in some cases, you need to move the motor to specify manually the start and end point for each motors. Once this done, then you can use the lib as it is to makes the move. That's maybe why @yuranikspb ask this, and it's also a part of my question. Prior a move, unless you can program the move you want (ie Winder app), how can you move independently the motor freely? Using an another lib? maybe - but how to do the link in between...

What is your though about it?

luni64, 2019-10-08:
Hope I understood your problem correctly. I assume in that 'prior move' you want to position the motors manually to some starting point? Using some joystick or similar? 

If so, a simple approach would be to use a rotation controller for this prior move and override the speed according to that pot/encoder/joystick..., As shown in the video I linked above you can use this to freely move one or more motors around. After you finished that step you can use a step controller to move the motor(s) to the targets.

christophepersoz, 2019-10-08:
Exactly. Moving the motor to reach the starting and ending point with the use of a pot, joystick, encoder, whatsoever. I read about rotation controller and the method `RotateControl(unsigned stepPulseWidth, unsigned speedUpdatePeriod)` - I have to go a bit deeper it seems in the Winder app to see how you are managing it, and much more, determine the limits for pulseWidth and speedUpdatePeriod.

closed 2019-10-08 by christophepersoz

yuranikspb, 2019-10-08:
3 axis car motor (track and turrel)moved async ,target reached from serialport

luni64, 2019-10-08:
Ok, use 3 rotation controllers, pass target speed instead of target position via serial port and use override speed to adjust the speed to the values from serial port.

yuranikspb, 2019-10-08:
turrel have  resistive ruler, i can use pid () control override speed ,target will be adc , and async move and 3 controllers :)

luni64, 2019-10-08:
Yes, this is what I did a couple of times. Isually works great. For an example see here https://forum.pjrc.com/threads/43491-New-Stepper-Motor-Library?p=198994&viewfull=1#post198994

And here https://www.youtube.com/watch?v=RMiKSQuO4Zo I used a nice function to generate the target positions for two motors.

You probably do not need a full PID the P part might be sufficient.

yuranikspb, 2019-10-08:
Thanks Luni64

chihoonl, 2021-09-03:
Hi Luny, My name is Jihoon from Korea. I would like to thank you first for sharing this wonderful Library.

I've been looking and testing the Teensystep in this last few days for my camera slider. And finally, I completed the test coding to save and return the In and Out target values using serial communication with Teensy3.6 and two motors. The result is successful and works better than expected! It's really wonderful.

`#include "TeensyStep.h"
StepControl controller;     // Use default settings

Stepper motor1(3, 2);       // STEP pin: 3, DIR pin: 2
Stepper motor2(7, 6);       // STEP pin: 7, DIR pin: 6

const int ms1 = 11;
const int ms2 = 13;

char s; 

int InandOut = 0; 

int XInPoint = 0; 
int YInPoint = 0;

int XOutPoint = 0; 
int YOutPoint = 0;

void setup() {

  Serial.begin(9600); 

  pinMode (ms1, OUTPUT);
  pinMode (ms2, OUTPUT);
  digitalWrite(ms1, LOW);
  digitalWrite(ms2, HIGH);

  motor1
  .setMaxSpeed(6000)      
  .setAcceleration(6000)
  .setPullInSpeed(200) 
  .setInverseRotation(LOW) 
  .setStepPinPolarity(LOW); 

  motor2
  .setMaxSpeed(6000)
  .setAcceleration(6000)
  .setPullInSpeed(200)
  .setInverseRotation(LOW)
  .setStepPinPolarity(LOW);
}

bool repeat = false;

void loop() {

  if (Serial.available()) {
    s = Serial.read(); 

    if (s == '1') 
    {
      Serial.println("IN_Set!");

      XInPoint = motor1.getPosition(); 
      YInPoint = motor2.getPosition(); /
      delay(200);
      delay(200);
    }

    if (s == '2')
    {
      Serial.println("Out_Set!");

      XOutPoint = motor1.getPosition();
      YOutPoint = motor2.getPosition();
      delay(200);
    }

    if (s == '3')
    {
      Serial.println("Move to IN!");

      motor1.setTargetAbs(XInPoint); 
      motor2.setTargetAbs(YInPoint); 
      controller.move(motor1, motor2);
      delay(200);
    }

    if (s == '4')
    {
      Serial.println("Move to OUT!");
      motor1.setTargetAbs(XOutPoint);
      motor2.setTargetAbs(YOutPoint);

      controller.move(motor1, motor2);
      delay(200);
    }

    if (s == '0')
    {
      Serial.println("Move to Temp!");
      motor1.setTargetAbs(10000); 
      motor2.setTargetAbs(7000); 
      controller.move(motor1, motor2);
      delay(200);
    }

    if (s == '9')
    {
      Serial.println("Set Position!");
      motor1.setPosition(0); 
      motor2.setPosition(0); 

      delay(200);
    }

    if ( s == '7')
    {
      Serial.println("Auto Repeat!");
      repeat = true;

      while (repeat)
      {

        motor1.setTargetAbs(XInPoint);
        motor2.setTargetAbs(YInPoint);

        controller.move(motor1, motor2);

        delay(1000);

        motor1.setTargetAbs(XOutPoint);
        motor2.setTargetAbs(YOutPoint);

        controller.move(motor1, motor2);

        delay(1000);

      }

    }

    Serial.end();
  }
}`

My next goal is move the motor to a free target(prior move) using 'RotateControl Class' by two Potentiometer instead the target "Move to Temp!" for InandOut (pls, refer of my code next below.)

To achieve this goal, but I realized that it was impossible to control by the adjust value of speed of the rotate class finally.

So I think, plan to do mapping the analog input of Potentiometer to overrideSpeed in rotateControl Class.. I hope so this way is correct.

Could you please advise on how to solve this problem?

To summarize my questions:
1. How can I design a 'prior move' by potentiometer using overrideSpeed?

2. How can I combine the two codes(stepControl with Rotate class) without any problems?

`#include "TeensyStep.h"
RotateControl controller;

Stepper motor1(3, 2);     
Stepper motor2(7, 6);       

#define JoyX  A14
#define JoyY  A15

int JoyXPos;
int JoyYPos;

void setup() {

  motor1
  .setMaxSpeed(3000)      
  .setAcceleration(5000);

  motor2
  .setMaxSpeed(3000)
  .setAcceleration(5000);

}

void loop() {

  JoyXPos = analogRead(JoyX);
  JoyYPos = analogRead(JoyY);

  if (JoyXPos > 550) {
    JoyXPos = map(JoyXPos, 550, 1023, 0, 3000);
    motor1.setMaxSpeed(JoyXPos);
  }

  else if (JoyXPos < 420) {
    JoyXPos = map(JoyXPos, 420, 0, 0, 3000);
    motor1.setMaxSpeed(-JoyXPos);
  }

  else {
    motor1.setMaxSpeed(0);
  }
  controller.rotateAsync(motor1);

  if (JoyYPos > 550) {
    JoyYPos = map(JoyYPos, 550, 1023, 0, 3000);
    motor2.setMaxSpeed(JoyYPos);
  }

  else if (JoyYPos < 420) {
    JoyYPos = map(JoyYPos, 420, 1, 0, 3000);
    motor2.setMaxSpeed(-JoyYPos);
  }

  else {
    motor2.setMaxSpeed(0);
  }
  controller.rotateAsync(motor2);

}`