luni64/TeensyStep #58: Problem with Blocking of Serial Send

geraldft, 2019-10-20:
Hi Lutz.  I am back to my little project.  

For various reasons I am trying to use a combination of global and local controllers, because I need to switch between controller types for different functions.  I could have just stuck to 4 global controllers if there was a type that did everything I need.  But the problem in my case is that the rotatecontrol can't be used for moving to a specific position - so I need to use a local stepcontrol if I wish to perform that operation.  But - the problem I am finding is that the local control will not allow me to monitor the motor position and return it by serial to my external program, while it is carrying out it's business.

Here is a simplified sample of the way I am doing this...

if (letter0 == "G0")   {
    StepControl localSC0;  //declare a local control
    moveto = receivedint0;  
    Mtr_0.setTargetAbs(moveto);  //set the target position for the move
    localSC0.moveAsync(Mtr_0);  //start moving the motor
    while(Mtr_0.getPosition()!=moveto){
    Serial.printf("P0%i\n", Mtr_0.getPosition());  //send feedback on where it is
                                                           }
    }

To explain a little more.  I first tried to use "localSC0.move(Mtr_))" which ran the motor to the required position, but completely blocked any other activity while it was moving the motor.  Hence the above code using "moveAsync" with the while loop which keeps it "alive" - otherwise the function doesn't work at all.  Now inside that loop I want to send the serial data to update my external control program so it knows where the motor is...  But it doesn't work.  The serial command does not work in that loop.  

Any idea why - or if there is another way to achieve what I need?  Thanks G

geraldft, 2019-10-20:
PS - I'd also like to able to add other code in the loop that checks if a stop message is sent from the external program - ie. if the operator decides to abort the goto movement.  Is that possible?

luni64, 2019-10-20:
Hi Gerald
>  I am back to my little project.

good to hear. 
> ... But - the problem I am finding is that the local control will not allow me to monitor the motor position...

This is highly unlikely. The motor positions have nothing to do with the controllers at all, they are maintained by the steppers. Whenever the motor makes a step -driven by whatever controller- it necessarily updates its position variable. 

At a first glance the code snippet seems to be OK but for actually testing it I need some compile-able piece of code. I assume it uses the same win app you sent in #53?

Instead of 
```c++
while(Mtr_0.getPosition()!=moveto)
{
    Serial.printf("P0%i\n", Mtr_0.getPosition());  //send feedback on where it is 
}
```
I would do
```c++
while(localSCO.isRunning())
{
    Serial.printf("P0%i\n", Mtr_0.getPosition());  //send feedback on where it is 
}
```
but this is probably not the reason. 
Remark: Looking at the code from #53 I suspect (gut feeling only)  your command interpreting / executing logic might be the root cause of your problem. I suggest to rewrite this using a simple state machine, 

>which ran the motor to the required position, but completely blocked any other activity while it was moving the motor.

What do you mean by "blocking"? TeensyStep usually is very efficient, so it doesn't generate a lot of processor load in the background and all other background tasks  (Serial etc) should work as usual. If, by blocking you mean that the move function does not return until the motor reached its target, well that's what it is supposed to do? move is implemented very similar like you did in your snippet
```c++
void StepControlBase<a, t>::move(Steppers &... steppers)
{
    moveAsync(steppers...);
    while (this->timerField.stepTimerIsRunning())
    {
        delay(1);
    }
}
```
i.e. it simply starts an async move, waits until it is done and returns after that. 

BTW: did you look at this example?https://github.com/luni64/TeensyStep/tree/master/examples/Interfacing

It more or less shows how to move motors controlled from Serial and reports back the current position while the motor moves.

geraldft, 2019-10-20:
Oh thanks for the quick reply.  

Yes essentially the same as before with a .net application sending data.  

By blocking I meant it doesn't release until the function is complete, which as you say it is meant to... but that was more by way of pre-amble.

I did think there was a catch whereby running the while loop was somehow not allowing the serial write to work properly.   So I tried your suggestion to replace (Mtr_0.getPosition()!=moveto) with (localSC0.localSCO.isRunning()).  Still something was wrong until I added delay(50) to the loop and suddenly it started working...  :)

I then went back and tried while (Mtr_0.getPosition()!=moveto) again and that now also works.  

I assume though using .isRunning is safer?

I'll look at your other more detailed example, but at least I have gotten over this little hurdle...

Thanks G

luni64, 2019-10-20:
Good catch! 
Didn't notice that you don't wait in your loop. That will quickly fill up the serial buffers (which is not a problem per se, but it will slow down things unnecessarily). Additionally, it will generate some huge load on TeensyStep since you asking it if it is ready every couple of µs - pretty much like pulling out some sapling every hour to check if it got roots already :-)

> I assume though using .isRunning is safer?

No, but nicer :-) 

>I'll look at your other more detailed example, but at least I have gotten over this little hurdle...

If I find some time I'll do an example how to implement a clean real life interface to a PC using a more elaborated scheme.

geraldft, 2019-10-20:
Thanks - that will be useful.  At present I just need it to work so I can test in principle.  But later it will need to be more robust and flexible I think...  :)

geraldft, 2019-10-21:
PS - I read the interfacing example and basically it seems to use functions to simplify things, but equally could also be written in a more linear and verbose fashion, and the result would be the same.  For me it does not seem to answer how to deal with things when there are 4 local and 4 global controllers which need to be accessed.  As an immediate example, using my sample loop (as above) to send one motor a goto, I can't currently issue a similar command to a second motor since it would be outside that particular loop.  So I'm not sure if I should look at loops within loops or some equivalent?

Sorry about my not so sophisticated knowledge of C if I'm not seeing something obvious, but maybe you can comment?

geraldft, 2019-10-21:
Hi again.  Sorry but I've had more thoughts.  In fact none of this would be necessary if the rotate controller had a function to move a motor to position like the stepcontrol does.  Perhaps I would be better off just writing a function that uses the rotatecontrol to do the same?   Then I don't need to use the stepcontrol.  In a simple way it would not be too hard, except that getting it to stop exactly within one step pulse might require a little extra effort...  :)

luni64, 2019-10-21:
It is a bit difficult to combine the information bits from your various mails to identify your problem. Since the code from https://github.com/luni64/TeensyStep/issues/53#issuecomment-528245008 did work nicely with two motors, I assumed you have problems making it work with all 4 motors, right? 

The problem with your code is that you always have all rotation controllers running and set the speed to 0 instead of stopping them. While overriding the speed to zero will effectively stop the motor, it does not release the allocated timer -> your code allocates all available timers and never releases them ->you don't have any timers left for the step controllers. The solution to the problem is quite simple: 

- Don't start the rotation controllers in setup()

- The stop command should call stopAsync() (which releases the timer) instead of overrideSpeed(0) which keeps the timer allocated. 

- In the jog commands check if the controller is already running, if not start it. If it already runs override its speed. 

### Additional Changes:
- To keep the housekeeping simple I globally defined one step- and one rotation controller for each motor. (note: you can define as much controllers as you like. Just make sure that you don't have more than 4 running at the same time)

- Instead of declaring the motors and controllers as dedicated variables I packed them in arrays which makes the handling much easier. 

Here working '4 motor code' based on your example. Unfortunately most of the functions are blocked in your controlling app so I could only test the jog and the goto functionality. Using this code, it should be straightforward to  implement the M0 command you mentioned above. Let me know if not. 

```c++
#include "Arduino.h"
#include "TeensyStep.h"

int currentPos;

float period = 100; //interval between counts in milliseconds
long counter = 0;   //increment counter
boolean counting = true;

// stuff needed for SERIAL to work
String sendstr, pref;

String inData;                   //everything from Serial comes to this variable to then be processed.
String receptacle0, receptacle1; // this variable is used to store data that has been stripped from any letters, keeping numbers (in String form).
//String letter0, letter1;          //this variable holds the first letter of the incoming data.This first letter determines where to send the data.
String letter[4];
float receivedint0, receivedint1; // Set value
float receivedint2, receivedint3; // dummy, to make parser work, you need to implement those
//-------------------------------

// forward delcaring functions (this block is not needed if you use the Arduino builder - yuck)
bool newCommand();
void GoTo(int motorId, float position);
void Jog(int motorId, float speed);
void Zero(int motorId);
void SetAcc(int motorId, float acc);
void Stop(int motorId);
//------------------------------------

// Arrays of pointers to motors, rotate- and step controllers to make commands simpler
RotateControl* Rctl[4];
StepControl* Sctl[4];
Stepper* Mtr[4];

void setup()
{
    Serial.println("");

    // setup 4 motors, 4 step and 4 rotate controllers you can of course do that without the loop if you need special settings for each motor/controller
    for (unsigned i = 0; i < 4; i++)
    {
        Sctl[i] = new StepControl();
        Rctl[i] = new RotateControl();

        Mtr[i] = new Stepper(i * 2, i * 2 + 1); // (0,1), (2,3) ...
        Mtr[i]->setMaxSpeed(20000);
        Mtr[i]->setAcceleration(20000);
    }
}

elapsedMillis displayStopwatch = 0; // easier to use than naked millis()

[... 125 lines cut ...]
}

void Zero(int mId)
{
    Mtr[mId]->setPosition(0);
    letter[mId] = "qq";
}

void Stop(int mId)
{
    if (Rctl[mId]->isRunning()) Rctl[mId]->stopAsync(); // overrideSpeed(0) would not release the timer
    if (Sctl[mId]->isRunning()) Sctl[mId]->stopAsync();
    letter[mId] = "qq";
}

void SetAcc(int mId, float acc)
{
    float accelFactor = acc / 20000.0f;
    Rctl[mId]->overrideAcceleration(accelFactor);
    letter[mId] = "qq";
}
```

Hope that helps
Lutz

geraldft, 2019-10-21:
Ok - that was more detail than I expected..!  :)  Thanks so much for what looks like a very nice model for processing the data.  Yes - I am working with 4 axes, that was always the intention.  I only used two axes earlier to keep it a little simpler in the early stages.  I will need to add more functions but with this template it will be much easier..

The main answer to my quandry is obvious now - ie. that I can declare 4 rotate and 4 step controls globally, so long as I issue a stop command before changing controls.  Being global I don't have to worry about any blocking...  :)

I wonder if at some stage is it technically possible to add a goto feature to the rotate control?  Seems it would all be much easier to work with if only one control was needed...

luni64, 2019-10-22:
>I wonder if at some stage is it technically possible to add a goto feature to the rotate control? Seems it would all be much easier to work with if only one control was needed...

I originally had it that way but decided to divide the controller into dedicated ones later, which made a lot of things much easier. E.g. what would you do if you have a moveToAsync running and somebody overrides the speed with a negative value. The motors would never reach their targets. Quite confusing. What, if you override the acceleration to some small value while the motor is already near to its target. It would overshoot and needs to go back. And so on... I think it is much cleaner to have a controller for specific tasks. 
I'm thinking of having yet another one which only controls one motor but is able to change targets during movements. Again something which would be difficult for the stepController (what to do with the other motors if you change the target of one?) 

Anyway, the underlying problem you face is not the fact that there are dedicated controllers but the limited number of available hardware timers. This will always force you to do some house keeping and make sure that you don't use more than available. If you look at the code I posted you will see that there is not much which would change if you'd have that combined controller. You'd still need to make sure that you are not in a rotating operation when you want the motor to goto a specific position. Or, what if you need to control 5 instead of 4 motors...

>The main answer to my quandry is obvious now - ie. that I can declare 4 rotate and 4 step controls globally, so long as I issue a stop command before changing controls.

Just to make that very clear: Technically, you do not need the stop command to change controllers you only need to make sure, if you start some movement, that the controller finds at least one free hardware timer. The controllers allocate one at the beginning of each movement and release it at the end. So, for example make sure that you don't block a timer by overrideSpeed(0) when in fact you want to stop the motor which would release the timer. 

Looks like I definitely need to add some example to the repository. :-)

geraldft, 2019-10-22:
I would still prefer a combined controller. It's easy enough to make sure the control program doesn't break any rules...  for instance, in any case for real world reasons, while running a programmed move, none of the other functions are allowed other than stop.  And when a programmed move finishes or is stopped, it always stops all the motor movement before subsequent commands are allowed.     🙂  In the case of swapping controls it's just as easy to make a mistake and crash it - as I have found ..  🙁

Perhaps if you can confirm one thing - my programmed moves using the rotatecontrol always start and finish with speedOverride value zero.  So I must then issue a stop command before using another control?  At present I still seem to have problems releasing the rotatecontrol...

Also to explain.  When making a programmed move pretty much always 4 motors are being run, so I would rarely have the case of having one spare controller.  It's mostly going to be all or none...

Thanks Gerald

luni64, 2019-10-22:
> Thanks.  I would still prefer a combined controller. 

Sure, the code is open source, Feel free to adapt it to anything you like. 

> Perhaps if you can confirm one thing - my programmed moves using the rotatecontrol always start and finish with speedOverride value zero.  So I must then issue a stop command before using another control?  

The rule is quite simple. speedOveride(0) is nothing bad, it stops the motor but does not release the timer. If you want to stop the motor and release the timer for other controllers, use one of the stop commands.

> At present I still seem to have problems releasing the rotatecontrol...  at least that's what I suspect is happening..

Difficult to judge without more information. Do you observe this with the code I posted? 

>Also to explain.  When making a programmed move pretty much always 4 motors are being run, so I would rarely have the case of having one spare controller.  It's mostly going to be all or none...

Since you have 4 timers and only 4 motors, you should never run into a problem if you don't block the timers.

geraldft, 2019-10-22:
Thanks.  I'd love to unify the controller, it would also make things less confusing generally...  but my skills will permit this in the short term..  🙂

Meantime - maybe if you can write a more complete "rule book" to explain the relationship between all the different functions and what they effect.

Re the bug I had - it was my control program sending a jog value of zero when I pressed the stop button.  Really I just need to send a stop but I had added jog zero first - which unfortunately set maxVelocity to zero and then things wouldn't move..  🙁

Debugging arduino is not too easy - I wish there was a way to monitor variables...

BTW I also tried to install Arduino IDE with Teensy on another computer but the devTimer version didn't work again.  I can't even remember how I got it to work last time.
Is it possible to make it a full release and maybe that would help?   Otherwise a step by step description for dummies showing how to update a library...

Interesting maybe, the Arduino version I have devTimer working with is 1.8.9    On the other PC Arduino is 1.8.10   But that may be irrelevant I guess...

Thanks Gerald

luni64, 2019-10-22:
> BTW I also tried to install Arduino IDE with Teensy on another computer but the devTimer version didn't work again.  I can't even remember how I got it to work last time.
Is it possible to make it a full release and maybe that would help?   Otherwise a step by step description for dummies showing how to update a library...

I usually don't use the Arduino IDE so I'm not too familiar with that. Are you sure that it is a library problem? I.e. did you try to upload a hex file compiled on the 'good PC' on the 'bad system'? If this still shows the problems it is more likely a hardware issue? Just guessing here. 

Did you try to install the 1.8.9 on the "bad system" does it work then? 

**Debugging**
> Debugging arduino is not too easy - I wish there was a way to monitor variables...

That can be done with some effort. I recently added the functionality to VisualTeensy. Here some information about the required hardware modifications on a Teensy 3.6 and a video showing a simple debug session https://github.com/luni64/VisualTeensy/releases/tag/v0.9.7.0

**Bug in Release of Timers**
While playing around with your code I found a bug preventing the timer release in some cases. Might be that some of your problems are related to this. I didn't analyse it completely but it looks like fixing this requires some effort. I'll work on that but this will take some time :-(

geraldft, 2019-10-23:
I have found a couple of bugs in my code.  One of the "Letter" variables was not reset.  But more critically I was often exceeding the value of 1.0 allowed by overrideSpeed which led to crashes and misbehaviour, which I was confusing with other causes..  🙁

So I've written in a check for that now, and hopefully can move on a bit...

Thanks for the debug link.  I'm used to be being able to send info to console during operation but when using serial this isn't available...

BTW - one apparent anomally I'm noticing.  As you may be aware I have a list of speeds that I send in a 50hz stream and which change the overrideSpeed value to create a repeatable profile move.  I noticed that when I run it in reverse the distance it travels is often a little further, when I would have expected it to be the same, since it is the same data stream, and the example I'm using is symetrical.   Is there anything different about using negative speedOverride vs positive?

luni64, 2019-10-23:
>I have found a couple of bugs in my code.  

Good to hear that it is your bugs and not mine :-)

>Thanks for the debug link.  I'm used to be being able to send info to console during operation but when using serial this isn't available.

In this cases I connect one of those cheap USB/Serial cables to (GND) and pin1 (TxD) and print my debug messages on Serial1 which works nicely. TyCommander can display the serial data from the USB/Serial cable which is quite convenient. Putty also works great for that. 

**Bug fix** 
I (hopefully) fixed the bug regarding the not released timers. This should work now. Update uploaded to GitHub. 

**Error Function**
I also included a possibility to attach an error function which will be called in case of errors. E.g. using too much controllers at the same time, starting a movement when the controller is running etc... This is quite useful in case of strange behaviour. Here a quick usage example:
```c++
#include "TeensyStep.h"

RotateControl rc1, rc2, rc3, rc4, rc5;
StepControl   sc1, sc2, sc3, sc4, sc5;
Stepper m1(2, 3), m2(4, 5), m3(6, 7), m4(8, 9), m5(10, 11);

void onError(int module, int code)
{
    Serial.printf("ERR: M:%i E:%i\n", module, code);
    while(true)
    {
        yield();  // keep the board responsive
    }
}

void setup()
{
    TeensyStep::ErrorHandler::attachCallback(onError);

    while (!Serial);        
    Serial.println("start debug-----------------");

    rc1.rotateAsync(m1);
    rc2.rotateAsync(m2);
    rc3.rotateAsync(m3);
    rc4.rotateAsync(m4);
    rc1.stop();           // <- commenting this out will generate an error    
    rc5.rotateAsync(m5);
}

void loop()
{
}
```

Alternatively you can use a built-in error function which prints out the errors on a settable stream (Serial in this example) and blinks the LED. 

```c++
#include "TeensyStep.h"

RotateControl rc1, rc2, rc3, rc4, rc5;
StepControl   sc1, sc2, sc3, sc4, sc5;
Stepper m1(2, 3), m2(4, 5), m3(6, 7), m4(8, 9), m5(10, 11);

void setup()
{
    TeensyStep::ErrorHandler::attachCallback(TeensyStep::verboseHandler(Serial));
    pinMode(LED_BUILTIN, OUTPUT);

    while (!Serial);        
    Serial.println("start debug-----------------");

    rc1.rotateAsync(m1);
    rc2.rotateAsync(m2);
    rc3.rotateAsync(m3);
    rc4.rotateAsync(m4);
    //rc1.stop();           // <- commenting this out will generate an error    
    rc5.rotateAsync(m5);
}

void loop()
{}
```

Output:
```
start debug-----------------
ERR M1C3: No timer available (PIT)
```
> Is there anything different about using negative speedOverride vs positive?
No, at least nothing on purpose.

geraldft, 2019-10-26:
Hi Lutz

Thanks for all that.  I haven't had time to try the debugging suggestions, in fact I just used a serial printline and sent it to my VB serial application.

Re errors I also haven't tried your error checking procedure yet.  Instead i just put some known error trapping code like limiting the input value of speedOverride to +-1.0.  (essential)  Also a simple trap to prevent using the steppercontrol while rotatecontrol is running.  Perhaps not ideal but seems to work...

    while(Rctl[mId]->isRunning()){
      Rctl[mId]->stopAsync();
    }

In general it's all working much better now though I can still crash it if my app does something really wrong.  So eventually I will need to make it more resistant to those possibilities...

I don't know if your most recent update made a difference since most of my errors were self inflicted I believe.  The reverse running anomaly turned out to be the stepper motor.  It just runs rougher in one direction, which I confirmed by reversing one pair of motor wires so forward was backwards...

The main challenge now is how to deal with the motors not reaching the exact end points when being driven by a serial stream.  (50hz of speed information for each axis)  Even though the data frequency is very well controlled there is still some random variation in each run.  I so long as the error is small I can issue a goto at the end of the move to re-position it.  In practice this is hardly noticeable but not the ideal solution.  I am also experimenting with corrections on the run, though initial tests haven't yet been convincing...  it's hard to work out the best way to time these things without interfering with what is already working...

Anyway - I'll forward a more tidy version of my code once I get it cleaned up and sorted a bit more.

BTW - I did finally work out how to update the devTimer library so it would work.   My procedure was :

  1.   Delete the old library folder from where it was installed originally.
  2.   Try and compile an ino which calls teensystep.  (This should fail and hence confirm the old one is gone.)
  3.   Download and save the devTimer update (as zip file) to temporary place.  (Do not extract the folder inside)
  4.   Open the menu  "Sketch" / "include library" and select "add zip library".
  5.  Load the zip file.

Previously i tried just replacing the old folder with the new, however I could never get that to work no matter what hoops I jumped through, whereas this seems to do the trick...

I should note to that a friend did update his Mac version of Arduino IDE teensyStep library without any issue, so this is just a PC problem.

BTW you earlier described some code to check the library version but I think that was in the other thread which you have since closed.  Is that documented elsewhere?

Thanks Gerald

luni64, 2019-10-26:
>  Perhaps not ideal but seems to work...
>     while(Rctl[mId]->isRunning()){
>       Rctl[mId]->stopAsync();
>     }

If this is supposed to be real code -> don't do that! It will torture your controller by continuously checking the status. I assume you did it  this way because originally the `isRunning ()`responded with false too early so that the timer was not yet released and you got a problem when using another controller. I fixed that problem in the last update. The usual 
```c++
 Rctl[mId]->stopAsync();
 while(Rctl[mId]->isRunning()){
>     delay(1)
>}
```
should work now without problem.  Looking at your snipped again I don't understand the idea of it. Does anything speak against the equivalent and much cleaner
```c++
if(Rctl[mId]->isRunning()) Rctl[mId]->stop();
``` 
>I can still crash it if my app does something really wrong. 

The new version should improve that a lot I hope. 

> The main challenge now is how to deal with the motors not reaching the exact end points when being driven by a serial stream.  (50hz of speed information for each axis)  Even though the data [...]

I would really like to support you here because it is very interesting for me to see the real life problems users of the library stumble over. But it is still difficult to understand what you are trying to achieve. Looks like you drive the motors in rotational mode since you want to adjust their  speed during the movement or generate some  speed profile. On the other hand you want to get to a fixed target? Or ist this two different operational modes? Or is the final goal to reach the target and control the speed profile during the move at the same time? 

> BTW - I did finally work out how to update the devTimer library so it would work.   My procedure was : ....

wow, that sounds super complicated. The Arduino IDE is perfect for beginners, everything runs out of the box. But using it for anything above blinking some LEDs is a real pain. I can only suggest again to switch to some modern IDE. I don't like to advertise myself but you could have a look at https://github.com/luni64/VisualTeensy.  It generates Teensy projects for vsCode, which is probably the most loved code editor these days. You can install it in about 10min. And you will never go back to the Arduino IDE voluntarily.

>BTW you earlier described some code to check the library version but I think that was in the other thread which you have since closed.  Is that documented elsewhere?

Here the issue: #53

BTW: You can have a look at all issues by simply deleting the default filter in the issue list
[image]

geraldft, 2019-10-26:
Thanks for quick reply..

Actually I think I now recall using the delay in that loop earlier but forgot about it this time..  🙁

The reason for the loop is to block and make sure it really is closed before the next command arrives.
Even if I tell it to stop it might take some time to react.  In my vb app I actually tried working around
it by sending a stop command then a small delay, then the new move command.  That did also help but I felt
it was not the best thing...

BTW I have now added your error generating code.  I still have the occasional error which is always Error3 -
I don't know what that means but I assume it is a problem with changing the controller, and it always locks up
the program... 🙁

Hopefully now with the delay it will be better...  but tell me if there is an error, is there a way to recover
without rebooting the Teensy?

block

See above - does error 3 mean anything?

It's to create a very specific profile which programmed by using keyframes to shape a curve.  The curve is in the form of a timeline vs position.  It has to be repeatable and has to reach the correct position at the end of the move.  Moving along the path of the move it should follow the curve values as accurately as possible.  Think of CNC but more free-form.  It will be used for running moving ,camera shots which have to be overlaid later.

If the speed generation was perfect it should always reach the same end point.  I realise it may not follow precisely the same path each time, but the object is to make it as perfect as possible...

I have used a Dynomotion Kflop controller up to now, which works very well, but it is relatively expensive, hence my interest to exploit the Teensy in a similar way...

I suppose another way of doing it might be to send a series of goto commands with infinite acceleration and varying velocities?

Here's an example looking at one axis with velocity shown as a dashed line.  The curve is drawn and then the data output to run the motors...

[cid:86bfa042-f9c3-41d3-b05a-ab802fe2aad1]

[image]<https://user-images.githubusercontent.com/12611497/67615364-ac246980-f7cb-11e9-9614-653a94a56a76.png>

luni64, 2019-10-26:
>The reason for the loop is to block and make sure it really is closed before the next command arrives. Even if I tell it to stop it might take some time to react. 

As mentioned, that was a bug which I fixed already, `isRunning()` only returns false now when the timer is definitely released. A simple 
```c++
if(Rctl[mId]->isRunning()) 
      Rctl[mId]->stop();
```
should do the job reliably. 

> does error 3 mean anything?

I need the reported module number and the reported code number to interpret this. 

> It's to create a very specific profile which programmed by using keyframes to shape a curve.  The curve is in the form of a timeline vs position.  It has to be repeatable and has to reach the [...]

Got it. This is something which I would solve differently. I did something similar some time ago I'll generate an example and put it on GitHub and the TeensyStep web page. 

> Here's an example looking at one axis with velocity shown as a dashed line.  The curve is drawn and then the data output to run the motors...
[cid:86bfa042-f9c3-41d3-b05a-ab802fe2aad1]

Something went wrong with your attachment/image whatever it was... can post that again?  Would be good to use real life data for the example (There is a preview tab in the edit field where you can check how your post will look like. Replying directly by email somehow badly messes up formatting and links. )

You can also reach me at lutz.niggl@lunoptics.com if that is easier for you.

luni64, 2019-10-26:
Was interested how that would work out and did a few experiments. First I drew an arbitrary function and digitized it to get this time/position graph:

[image]

For the sake of a quick experiment I simulated your serial connection by copying the data into a constant array and read out new targets from this array every 200ms.
 
The motor is moved by a very simple pid regulator which calculates the difference of the current position to the current target and determines a new motor speed from the difference (the larger the difference the larger the speed). The difference is multplied by some constant P which determines how aggressive the correction will be. You need to adjust P to your conditions. If you choose it too large you might get overshoot and / or oscillations. (goolge for regulator oscillations if you are interested). If you choose P too small the motor will lag significantly behind the changing targets. Just do some experiments....

Due to the regulation algorithm the motor will always reach the final target. 

The code periodically prints out  current time, current target and current motor position. 
I used this data to generate a plot showing both target and position over time. Looks quite good I'd say. 

[image]

And here the feasibility code

```c++
#include "Arduino.h"
#include "TeensyStep.h"

constexpr unsigned PID_Interval = 10; // ms
constexpr float P = 0.005;            // (P)roportional constant of the regulator needs to be adjusted
                                      // (depends on speed and accelerator setting)
Stepper m1(0, 1);
RotateControl rc;

int32_t target;

extern void pid();
extern const int targets[];
extern const uint32_t maxTargets;

void setup()
{
    while (!Serial && millis() < 1000);
    m1.setMaxSpeed(20000);
    m1.setAcceleration(50000);

    rc.rotateAsync(m1);
    rc.overrideSpeed(0);
}

elapsedMillis targetTimer = 0;
elapsedMillis outputTimer = 0;
uint32_t curPos = 0;

void loop()
{
    pid();

    if (targetTimer > 200 && curPos < maxTargets) // get new target every 200ms
    {
        targetTimer = 0;
        target = targets[curPos++];
    }

    if (outputTimer > 50) // print target and current position for display
    {
        Serial.printf("%d\t%d\t%d\n", millis(), target, m1.getPosition());
        outputTimer = 0;
    }
}

void pid()
{
    static unsigned lastTick = 0;

    if (millis() - lastTick > PID_Interval)
    {
        float delta = (target - m1.getPosition()) * (P / PID_Interval); // This implements a simple P regulator (can be extended to a PID if necessary)
        float factor = std::max(-1.0f, std::min(1.0f, delta));          // limit to -1.0..1.0

        rc.overrideSpeed(factor); // set new speed
        lastTick = 0;
    }
}

const int targets[] = {
    0,     179,   811,   1444,  2076,  2709,  3364,  3997,  4629,  5262,  5894,  6527,  7182,  7837,  8470,  9102,  9735,  10367, 11000, 11655, 12288, 12920, 13553, 14185, 14840,
    15473, 16105, 16761, 17393, 18026, 18658, 19291, 19923, 20556, 21166, 21798, 22453, 23041, 23651, 24261, 24848, 25413, 25955, 26474, 26994, 27491, 27942, 28394, 28755, 29161,
    29500, 29793, 30087, 30312, 30560, 30786, 30966, 31146, 31304, 31439, 31574, 31664, 31799, 31866, 31956, 32000, 32045, 32089, 32134, 32133, 32133, 32177, 32176, 32153, 32130,
    32129, 32129, 32060, 32037, 31946, 31945, 31854, 31808, 31739, 31671, 31580, 31489, 31398, 31306, 31215, 31102, 31010, 30874, 30760, 30647, 30510, 30396, 30215, 30079, 29920,
    29761, 29579, 29397, 29216, 29034, 28853, 28648, 28399, 28195, 27991, 27719, 27469, 27197, 26948, 26653, 26336, 26019, 25679, 25316, 24931, 24546, 24138, 23685, 23232, 22802,
    22326, 21851, 21398, 20877, 20424, 19881, 19382, 18884, 18363, 17888, 17412, 16959, 16506, 16076, 15668, 15283, 14898, 14536, 14173, 13833, 13516, 13244, 12972, 12745, 12496,
    12291, 12087, 11906, 11724, 11542, 11406, 11247, 11111, 11020, 10906, 10815, 10724, 10632, 10564, 10518, 10450, 10381, 10358, 10334, 10311, 10265, 10265, 10264, 10286, 10308,
    10307, 10352, 10419, 10441, 10508, 10575, 10620, 10710, 10799, 10912, 11024, 11137, 11272, 11384, 11519, 11677, 11834, 12015, 12217, 12420, 12601, 12803, 13006, 13232, 13457,
    13683, 13908, 14156, 14404, 14675, 14923, 15194, 15419, 15690, 15961, 16232, 16457, 16728, 16976, 17247, 17517, 17743, 17991, 18239, 18487, 18690, 18961, 19141, 19412, 19592,
    19817, 19998, 20178, 20358, 20538, 20651, 20808, 20943, 21078, 21213, 21348, 21461, 21596, 21731, 21821, 21956, 22068, 22181, 22316, 22428, 22540, 22675, 22788, 22923, 23058,
    23193, 23328, 23486, 23598, 23778, 23958, 24161, 24409, 24590, 24815, 25041, 25289, 25559, 25830, 26123, 26485, 26846, 27207, 27636, 28088, 28562, 29059, 29601, 30165, 30730,
    31363, 31995, 32628, 33283, 33938, 34661, 35384, 36107, 36830, 37553, 38276, 38954, 39609, 40242, 40942, 41552, 42185, 42772, 43291, 43811, 44285, 44714, 45143, 45549, 45911,
    46227, 46497, 46791, 47039, 47309, 47535, 47760, 47941, 48121, 48278, 48436, 48594, 48706, 48841, 48954, 49043, 49156, 49246, 49335, 49403, 49447, 49514, 49581, 49626, 49693,
    49715, 49760, 49782, 49849, 49871, 49870, 49915, 49959, 49959, 49981, 50002, 50002, 50001, 50000, 50000,
};

const uint32_t maxTargets = sizeof(targets) / sizeof(targets[0]);

```

geraldft, 2019-10-26:
Well that is throwing the cat amongst the pigeons!  You have the theory correct re the move profiles.  They might be anything you can draw that the motors can physically achieve.   It looks like a much better solution right from the start so long as I can manage it with 4 axes, ramp it down midstream when needed, and ability to do gotos...

closed 2020-05-02 by luni64

EeeLo, 2023-09-25:
> Was interested how that would work out and did a few experiments. First I drew an arbitrary function and digitized it to get this time/position graph:
> 
> [image]
> 
> For the sake of a quick experiment I simulated your serial connection by copying the data into a constant array and read out new targets from this array every 200ms.
> 
> The motor is moved by a very simple pid regulator which calculates the difference of the current position to the current target and determines a new motor speed from the difference (the larger the difference the larger the speed). The difference is multplied by some constant P which determines how aggressive the correction will be. You need to adjust P to your conditions. If you choose it too large you might get overshoot and / or oscillations. (goolge for PID oscillations if you are interested). If you choose P too small the motor will lag significantly behind the changing targets. Just do some experiments....
> 
> Due to the regulation algorithm the motor will always reach the final target.
> 
> The code periodically prints out current time, current target and current motor position. I used this data to generate a plot showing both target and position over time. Looks quite good I'd say.
> 
> [image]
> 
> And here the feasibility code (you can test this on a bare Teensy, no need to connect motors)
> 
> ```c++
> #include "Arduino.h"
> #include "TeensyStep.h"
> 
> constexpr unsigned PID_Interval = 10; // ms
> constexpr float P = 0.005;            // (P)roportional constant of the regulator needs to be adjusted
>                                       // (depends on speed and accelerator setting)
> Stepper m1(0, 1);
> RotateControl rc;
> 
> int32_t target;
> 
> extern void pid();
> extern const int targets[];
> extern const uint32_t maxTargets;
> 
> void setup()
> {
>     while (!Serial && millis() < 1000);
>     m1.setMaxSpeed(20000);
>     m1.setAcceleration(50000);
> 
>     rc.rotateAsync(m1);
>     rc.overrideSpeed(0);
> }
> 
> elapsedMillis targetTimer = 0;
> elapsedMillis outputTimer = 0;
> uint32_t curPos = 0;
> 
> void loop()
> {
>     pid();
> 
>     if (targetTimer > 200 && curPos < maxTargets) // get new target every 200ms
>     {
>         targetTimer = 0;
>         target = targets[curPos++];
>     }
> 
>     if (outputTimer > 50) // print target and current position for display
>     {
>         Serial.printf("%d\t%d\t%d\n", millis(), target, m1.getPosition());
>         outputTimer = 0;
>     }
> }
> 
> void pid()
> {
>     static unsigned lastTick = 0;
> 
>     if (millis() - lastTick > PID_Interval)
>     {
>         float delta = (target - m1.getPosition()) * (P / PID_Interval); // This implements a simple P regulator (can be extended to a PID if necessary)
>         float factor = std::max(-1.0f, std::min(1.0f, delta));          // limit to -1.0..1.0
> 
>         rc.overrideSpeed(factor); // set new speed
>         lastTick = 0;
>     }
> }
> 
> const int targets[] = {
>     0,     179,   811,   1444,  2076,  2709,  3364,  3997,  4629,  5262,  5894,  6527,  7182,  7837,  8470,  9102,  9735,  10367, 11000, 11655, 12288, 12920, 13553, 14185, 14840,
>     15473, 16105, 16761, 17393, 18026, 18658, 19291, 19923, 20556, 21166, 21798, 22453, 23041, 23651, 24261, 24848, 25413, 25955, 26474, 26994, 27491, 27942, 28394, 28755, 29161,
>     29500, 29793, 30087, 30312, 30560, 30786, 30966, 31146, 31304, 31439, 31574, 31664, 31799, 31866, 31956, 32000, 32045, 32089, 32134, 32133, 32133, 32177, 32176, 32153, 32130,
>     32129, 32129, 32060, 32037, 31946, 31945, 31854, 31808, 31739, 31671, 31580, 31489, 31398, 31306, 31215, 31102, 31010, 30874, 30760, 30647, 30510, 30396, 30215, 30079, 29920,
>     29761, 29579, 29397, 29216, 29034, 28853, 28648, 28399, 28195, 27991, 27719, 27469, 27197, 26948, 26653, 26336, 26019, 25679, 25316, 24931, 24546, 24138, 23685, 23232, 22802,
>     22326, 21851, 21398, 20877, 20424, 19881, 19382, 18884, 18363, 17888, 17412, 16959, 16506, 16076, 15668, 15283, 14898, 14536, 14173, 13833, 13516, 13244, 12972, 12745, 12496,
>     12291, 12087, 11906, 11724, 11542, 11406, 11247, 11111, 11020, 10906, 10815, 10724, 10632, 10564, 10518, 10450, 10381, 10358, 10334, 10311, 10265, 10265, 10264, 10286, 10308,
>     10307, 10352, 10419, 10441, 10508, 10575, 10620, 10710, 10799, 10912, 11024, 11137, 11272, 11384, 11519, 11677, 11834, 12015, 12217, 12420, 12601, 12803, 13006, 13232, 13457,
>     13683, 13908, 14156, 14404, 14675, 14923, 15194, 15419, 15690, 15961, 16232, 16457, 16728, 16976, 17247, 17517, 17743, 17991, 18239, 18487, 18690, 18961, 19141, 19412, 19592,
>     19817, 19998, 20178, 20358, 20538, 20651, 20808, 20943, 21078, 21213, 21348, 21461, 21596, 21731, 21821, 21956, 22068, 22181, 22316, 22428, 22540, 22675, 22788, 22923, 23058,
>     23193, 23328, 23486, 23598, 23778, 23958, 24161, 24409, 24590, 24815, 25041, 25289, 25559, 25830, 26123, 26485, 26846, 27207, 27636, 28088, 28562, 29059, 29601, 30165, 30730,
>     31363, 31995, 32628, 33283, 33938, 34661, 35384, 36107, 36830, 37553, 38276, 38954, 39609, 40242, 40942, 41552, 42185, 42772, 43291, 43811, 44285, 44714, 45143, 45549, 45911,
>     46227, 46497, 46791, 47039, 47309, 47535, 47760, 47941, 48121, 48278, 48436, 48594, 48706, 48841, 48954, 49043, 49156, 49246, 49335, 49403, 49447, 49514, 49581, 49626, 49693,
>     49715, 49760, 49782, 49849, 49871, 49870, 49915, 49959, 49959, 49981, 50002, 50002, 50001, 50000, 50000,
> };
> 
> const uint32_t maxTargets = sizeof(targets) / sizeof(targets[0]);
> ```
> 
> Here the corresponding *.xls [gerald_test1.xlsx](https://github.com/luni64/TeensyStep/files/3774791/gerald_test1.xlsx)

How to extend to PID about this code?thanks.