luni64/TeensyStep #59: dynamically change setTargetAbs() while motor is moving

popsure, 2019-11-12:
I want to use the motor in a way that it constantly receives new target data and that it simply tries to follow the target position. When using the StepControl class I have an easy way to move to an absolute position, but it doesn't allow for a graceful response when giving a new position while it is in the process of going somewhere. RotateControl does allow for dynamic shifts in direction with overrideSpeed(), though then I have to rebuilt the libraries ability to keep track of where it is on my side. Is there any way I could still use StepControl but allow for dynamically changing the absolute target?

luni64, 2019-11-12:
You currently can not change the target while the motor is moving. but there is a simple way to do this. Here some more information and a test sketch:  https://github.com/luni64/TeensyStep/issues/58#issuecomment-546596422

(You need to take the code fromthe DevTimer branch for this to work).

popsure, 2019-11-13:
Luni thank you so much!

Your example code from the other thread was a perfect solution :-)

closed 2019-11-13 by popsure