luni64/TeensyStep #60: setPullInSpeed() not recognised in library

thierryhubert333, 2019-11-15:
I'm trying to use the function motor.setPullInspeed(800); as on the site example. but it doesn't recognize it when i compile. I'm using a teensy 3.5. I tried looking in the library and can't find it anywhere.

luni64, 2019-11-15:
Sorry, this is not part of the currently released version.  However, you can use the devTimer branch which includes it. I'll release this branch together with the Teensy 4 adaptions later this year.

thierryhubert333, 2019-11-17:
Later this year, do you still mean 2019? I don't mean to put pressure or anyhting, just to be clear. Thank you for your fast response.

luni64, 2019-11-17:
Does anything speak against using the DevTimer branch?  The more people testing it the better :-)

thierryhubert333, 2019-11-17:
Oops, im sorry. I'm not experienced with github. I didn't know i could just be part of that and download it. Thank you, I will test it!

closed 2019-11-17 by thierryhubert333