luni64/TeensyStep #62: ESP32 arduino port

ramboerik, 2019-12-19:
Hi,

I'm currently working on a project that is about to migrate from the teensy platform to ESP32 and would like to continue using TeensyStep. Do you know if anybody else has attempted to port TeensyStep to ESP32?

There seems to be a timer implementation for steppers in the ESP32 port of [grbl CNC](https://github.com/bdring/Grbl_Esp32)  in https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/stepper.cpp. It is however deeply integrated with the grbl code but can maybe be used as a starting point.

If I'm about to give it a try, do you have any general advises?

luni64, 2019-12-19:
Current plan is to adapt TeensyStep to T4.0 over the Xmas holidays. To do this I plan to completely exchange the timer code by something more generic. So, if you build on the current lib you probably need to change that soon again. 
I did some preparation work already by writing TeensyTimerTool, a general purpose timer library (work in progress) which is intended to provide a generic interface to the various timers. I will use this library in TeensyStep to be less dependent on the actual hardware properties. Porting this timer library to ESP32 might be a good start for your project. If you are interested I can give you access to the currently private repo. 

> Do you know if anybody else has attempted to port TeensyStep to ESP32?

Not that I'm aware of.  There where some porting attempts for STM32 but I never heard if this was finally working.

ramboerik, 2019-12-19:
> To do this I plan to completely exchange the timer code by something more generic

Okay, is the `timer/generic` folder a trace of that work? I was a bit confused at first as it seemed like a generic interface existed but wasn't in use 😃 I think it's a good idea to move actual hardware dependency out of the stepper code and it would probably make my work much easier.   
> If you are interested I can give you access to the currently private repo.

Yes, please invite me and I can have a look at it.

luni64, 2019-12-19:
The timer/generic folder is old stuff. The complete timer folder will be replaced. Let me clean up the code and generate some examples on the weekend before I invite you...

luni64, 2019-12-19:
Added you as collaborator.

ramboerik, 2019-12-22:
Hi,

Short update from  my  side:

The ESP32 branch of my project is now dead as the Xtensa compiler doesn't handle c++11 well and my project depends on it. I get a lot of internal errors from the compiler. It's easier for me to switch platform instead of modify my project. Instead of the ESP32 I changed to use a STM32 board and I will continue to port the timer library to that instead. 

I have read a bit about the STM32 and it does already have a timer implementation for arduino called [HardwareTimer](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/HardwareTimer.h). I guess I could just write a wrapper around that as it utilize the STM32's general purpose timers. What do you think about that?

ramboerik, 2019-12-22:
Pushed what I have done so far to branch `feature-stm32f4`. I'm using platformio as build system so I had to restructure the project in order for it to compile. I have used platformio for only a short period of time but I'm starting to like it more and more and it's very easy when dealing with multiple platforms/boards.

RockerM4NHUN, 2019-12-29:
Hi Erik,
I've just created a timer library for STM32 to generate precise (interrupt) callbacks using a single hardware timer. It is intended for PlatformIO and STM32 HAL environment, not Arduino, but you might be able to use it somehow, I thought you might be interested.

The lib is STM32TimerArray, if you can't use it directly, porting to Arduino is probably very easy (especially when V0.4.0 is finished).

I'm also working on an STM32 HAL port of TeensyStep (forked with name STM32Step), however I've made some heavy modifications to remove lambdas and templates and adapt it to use my timer library. It is not even nearly ready yet, if you are interested, check in a week or two. I also have some abitious plans to improve the stepper lib by using STM32 hardware features, this might not be done at all.

ramboerik, 2019-12-30:
Hi Rockerm4n,

Nice to know, there seems to  be quite a lot of people searching for a teensystep port for stm32 :) I will  definitely keep an eye on your work.

ramboerik, 2020-01-26:
Hi @luni64,

 I noticed that you working hard on the TeensyTimerTool project, what the status? Are there still plans to adjust TeensyStep to use the new timer library?

luni64, 2020-01-26:
Yes, definitely. Unfortunately TeensyTimerTool is a bit fragile. I'm currently working on a new, more stable base for it which should also make it more easy to implement other hardware.  Might take a few days. As soon as this is working stably I try to adapt TeensyStep.  
I have a Maple mine clone (https://github.com/OliviliK/STM32F103/wiki/F103CB) lying around but never used it. Does this make sense for compiling experiments or do you recommend something else?

ramboerik, 2020-01-26:
Perfect! I have played around a bit with stm32's arduino timer implementation and written a wrapper to teensyTimerTool. I'm ready to start testning with teensystep as soon as it has been updated.
I'm using a board from the stm32f4 series, https://www.st.com/en/microcontrollers-microprocessors/stm32f429zi.html. I think the timer hardware differs from your board but as I rely mostly on their timer implementation it might work.

ramboerik, 2020-03-06:
Hi @luni64,

I saw that TeensyTimerTool was recently tagged and it seems to have reached a more stable state. What's the timeline for having teensystep modifed and use the new timer implementation? Can't wait to start testing it for the stm32f4.

RCP1, 2020-03-11:
I have ported the library to ESP32 in my fork. I would also like to use a more general timer interface. Btw. thanks for the library, it is great.

ramboerik, 2020-04-15:
FYI: I continued on @RCP1 's fork and added support for stm32f4 here: https://github.com/ramboerik/stm32f4Step/tree/feature-stm32f4

alekkh, 2020-06-05:
Hi ramboerik ,   gives error  "no matching function for call to 'HardwareTimer::attachInterrupt(TimerField::TimerField(TF_Handler*)::<lambda()>)'"

ramboerik, 2020-06-05:
Hi @alekkh! That's right, my implementation depends on the latest release(1.9.0) of the Arduino core for [stm32](https://github.com/stm32duino/Arduino_Core_STM32/releases) that has updated HardwareTimer interface. You will need to install it to get it to work.

closed 2020-10-05 by luni64