luni64/TeensyStep #63: Constant movement without a target

acastles91, 2020-01-10:
Is there a way of moving indefinitely without a target? Right now I used setTargetRel(LONG_MAX) from the limits.h library but I don't know if it's going to stop eventually. Great work on the library, thank you!

acastles91, 2020-01-10:
Ok, a step further in my tests. I'm stuck with the move function and cannot make the motor stop once is moving with target LONG_MAX

RockerM4NHUN, 2020-01-10:
To rotate the stepper with a given speed (indefinitely), use the RotateControl class. I think the doRotate and stop functions should work for you.

luni64, 2020-01-10:
Here a working example
```c++
#include "Arduino.h"
#include "TeensyStep.h"

RotateControl controller;
Stepper stepper(1, 2);

void setup()
{
  while (!Serial && millis() < 500);

  stepper.setAcceleration(50000);
  stepper.setMaxSpeed(50000);

  Serial.printf("Pos: %d \n", stepper.getPosition());

  controller.rotateAsync(stepper); // start constant movement
  delay(2000);                     // stop after 2s
  controller.stop();

  Serial.printf("Pos: %d \n", stepper.getPosition());
}

void loop()
{
}
```

acastles91, 2020-01-10:
Thank you both for your answers. I am trying to control the motor with a pair of switch state functions connected to my keyboard. What I need to do is to set the motor in motion with a key and let it run in the background until I stop it with another key, so that it doesn't stop the rest of my code from executing. I tried the RotateControl class but is protected, I'll try your solution now @luni64 Thanks!

luni64, 2020-01-10:
Here [https://github.com/luni64/TeensyStep/blob/master/examples/Interfacing/Interfacing.ino]( https://github.com/luni64/TeensyStep/blob/master/examples/Interfacing/Interfacing.ino) an example showing how to handle such things in principle.

acastles91, 2020-01-10:
I just tried your example, changed the values of my pins to my current setup, and when I press 'm' in the serial monitor it basically counts until 20000, as if those were the steps, but the motor itself doesn't move. I don't know what I'm doing wrong, the programs compiles fine.

luni64, 2020-01-10:
Sounds like some cabling/hardware issue. If it counts it generates step pulses. Do you have a logic analyser or scope to check?

acastles91, 2020-01-10:
I do, and the driver is sending the steps correctly. I know is not a hardware problem because I can move the motor with

`controller.moveAsync(stepper);
  delay(2000); `

Is the switch / case example the one that is not working... Also I still don't understand the moveAsync function. So basically the controller will move the stepper for the duration of the delay, then stop, and then repeat? How would you include that in a switch / case so that it could perform other tasks not related to the motor while it runs on the background? Because as far as I understand, the example that you shared calls functions that are all related to the motor, like start or stop. What if, as an example, I would like to turn on and off an LED while the motor runs?

luni64, 2020-01-10:
> I do, and the driver is sending the steps correctly.

So, if the pulses are generated correctly and the motor doesn't move it has to be some hardware problem.  StepFrequency set correctly (start with slow movements, like 1000 stp/s.) ? Enable input to the driver correct? Does your stepper driver accept 3.3V signals? 

More information about your setup would be helpful. 

> Is the switch / case example the one that is not working... 

Can you please post the complete sketch witch is not working? 

> l also I still don't understand the moveAsync function.
- Move starts the motor and returns when the motor reached the target. This is convenient when you want to position the motor and don't need to do anything else. 
- MoveAsync just starts the movement and returns immediately. The motor does its movement in the background. 

> How would you include that in a switch / case so that it could perform other tasks not related to the motor while it runs on the background? Because as far as I understand, the example that you [...]

You'd do it in the same fashion as shown in the example. But I suggest to solve the basic issue of not moving motors first. So please send the complete sketch so that I can have a closer look

acastles91, 2020-01-10:
Ok, I will prepare a simplified sketch, thanks!

luni64, 2020-01-10:
And some info about the used hardware please. Which Teensy, which stepper driver, cabling etc.

acastles91, 2020-01-10:
So this is more or less a simplified version of my code:

`#include <Arduino.h>
#include <limits.h>
#include <TeensyStep.h>

const int step_Pin = 3;
const int dir_Pin = 4;
const int enabPin = 7;
const int led = 9;

enum class State {
  stop,
  move,
  turn_on_led,
  auto_end
};

void stateSwitch(State &stateArg){
  if (Serial.available()) {
        uint8_t code = Serial.read();
        switch (code) {

          case 'a':
            stateArg = State::stop;
            break;
          
          case 'b':
            stateArg = State::move;
            break;

          case 'c':
            stateArg = State::turn_on_led;
            break;
        }
  }          
};

void serialTask(State &stateArg, Stepper &stepperArg, StepControl &stepControlArg, RotateControl &rotateControlArg){

switch (stateArg) {

  case State::auto_end:
    //Serial.println("Auto end");
    break;
  
  case State::stop:

    stepControlArg.stop();
    //Serial.println("Stop");
    stateArg = State::auto_end;
    break;
  
  case State::move:
    stepControlArg.moveAsync(stepperArg);
    //Serial.println("Move");
    break;

  case State::turn_on_led:
    digitalWrite(led, (!digitalRead(led)));
    //Serial.println("LED");
    State::auto_end;
    break;

  }
};

Stepper stepperObj(step_Pin, dir_Pin);
StepControl stepControlObj;
RotateControl rotateControlObj;
State stateObj;

void setup(){

  Serial.begin(115200);
  
  pinMode(enabPin, OUTPUT);
	pinMode(step_Pin, OUTPUT);
	pinMode(dir_Pin, OUTPUT);
  pinMode(led, OUTPUT);

  stepperObj.setMaxSpeed(25000);
  stepperObj.setAcceleration(10000);
  stepperObj.setTargetRel(LONG_MAX);

  stateObj = State::auto_end;
  Serial.println("Setup ready");
};

void loop(){
  //Serial.println("?");
  stateSwitch(stateObj);
  serialTask(stateObj, stepperObj, stepControlObj, rotateControlObj); 
};`

I got it running, but in a way that I don't understand. If I press 'b' nothing happens, but if I press 'c' right after, the motor starts moving. If I then press 'a' the motor turns off and I'm not able to turn it on again, and if I press 'b' right after 'c' the motor stops abruptly. As of the driver, it definitely manages to move the motor but there might be some damage, I'm using a TMC2130 and they are very fragile in my experience, or at least I always find a way to break them. Apart from that I'm using a Teensy 3.2 that fits into a PCB that I designed myself and I am not a professional. The only cables are the ones of the motor, which go into a molex microfit. Thanks!

acastles91, 2020-01-10:
Sorry for the formatting, I don't know what happened to the indentation

luni64, 2020-01-11:
The problem is that you are calling moveAsync while the motor is already moving. All move commands must never be called on a running motor. 
```c++
case State::move:
    stepControlArg.moveAsync(stepperArg);
    //Serial.println("Move");
    break;
```
Since this seems to be a common misunderstanding I'll add some error checks in the next version. You can work around by checking with stepControlArg.isRunning() before calling moveAsync (for a better strategy see my example below) 

You also have a bug here: 
```c++
case State::turn_on_led:
    digitalWrite(led, (!digitalRead(led)));
    //Serial.println("LED");
    State::auto_end;  // this should be stateArg = State::auto_end
    break;
```

### General remarks: 
You mix up triggers and states in your state machine which makes it difficult to follow. As an example 
how to improve it I rewrote this part of your code and added some comments.  
Please note that I also changed the following:
- Better memorizable command keys  
- Use the built in LED (I don't have one lying around for pin 9)
- Uses rotation controller instead of the step controller
- Added output position and speed 

```c++
#include <Arduino.h>
#include <TeensyStep.h>
#include <limits.h>

const int step_Pin = 3;
const int dir_Pin = 4;
const int enabPin = 7;
const int led = 9;

enum class State {
    stopping,
    moving,
    idle
};

enum class Trigger {
    startRequest,
    stopRequest,
    ledToggleRequest,
    internal
};

Stepper stepperObj(step_Pin, dir_Pin);
StepControl stepControlObj;
RotateControl rotateControlObj;

void showPosition()
{
    static elapsedMillis stopwatch = 0;
    static int oldPos;

    int pos = stepperObj.getPosition();
    int speed = rotateControlObj.isRunning() ? rotateControlObj.getCurrentSpeed() : 0;

    if (stopwatch > 150 && oldPos != pos)
    {
        Serial.printf("pos: %d  speed: %d\n", pos, speed);
        oldPos = pos;
        stopwatch = 0;
    }
}

// checks the input (here a stream) and generates triggers for the state machine
// You can simply change that to other input sources, have more than one source etc.
Trigger checkInput(Stream& stream)
{
    switch (stream.read()) // returns -1 if nothing available, no need to check before reading
    {
        case 's':  return Trigger::stopRequest;
        case 'm':  return Trigger::startRequest;
        case 'l':  return Trigger::ledToggleRequest;
        default:   return Trigger::internal;
    }
};

// this is the actual state machine.
// it changes state on external or internal triggers
void switchState(Trigger trigger, Stepper& stepperArg, StepControl& stepControlArg, RotateControl& rotateControlArg)
{
    static State state = State::idle; // no need for a global state variable

    switch (trigger)
    {
        case Trigger::startRequest:
            if (state != State::moving)
            {
                Serial.println("startRequest");
                //stepControlArg.moveAsync(stepperArg);   // will work but simpler to use the rotate controller here
                rotateControlArg.rotateAsync(stepperArg); // start rotation
                state = State::moving;
            }
            break;

        case Trigger::stopRequest:
            if (state == State::moving)
            {
                Serial.println("stopRequest");
                //stepControlArg.stop();             //stop will wait until motor stops which is not good in a state machine...
                rotateControlArg.stopAsync();        //...use the async version instead to immediately return;
                state = State::stopping;
            }
            break;

        case Trigger::ledToggleRequest:
            Serial.println("LED");
            digitalWriteFast(LED_BUILTIN, !digitalReadFast(LED_BUILTIN));
            break;

        case Trigger::internal: // handle state changes without external trigger here
            if (state == State::stopping)
            {
                if (!rotateControlArg.isRunning()) state = State::idle; // check if the stopping sequence is done
            }

            //... other state changes

            break;

        default:
            Serial.println("Unexpected trigger");
            break;
    }
};

void setup()
{
    //Serial.begin(115200);            // doesn't do any harm but is not necessary at all... (USB)
    while (!Serial);                   //...but you should wait until the PC is ready otherwise you'll miss the starting message

    pinMode(enabPin, OUTPUT);
    // pinMode(step_Pin, OUTPUT);      // doesn't do any harm but is not necessary (handled by the stepper class)
    // pinMode(dir_Pin, OUTPUT);
    pinMode(LED_BUILTIN, OUTPUT);

    stepperObj.setMaxSpeed(25000);
    stepperObj.setAcceleration(10000);
    stepperObj.setTargetRel(LONG_MAX); // if you use the rotate controller you don't need the dummy target

    Serial.println("Setup ready");
    Serial.println("m: start, s: stop, l: toggle led");
};

void loop()
{
    Trigger trigger = checkInput(Serial);
    switchState(trigger, stepperObj, stepControlObj, rotateControlObj);  // call this as often as possible
    showPosition();
};
```

acastles91, 2020-01-11:
Thanks a lot for your input, I'll follow your suggestions and integrate them to my code as soon as I fully understand everything I need to change. I just compiled it and the first thing that I notice is that if the motor is moving and I turn on the led it slows down significantly and when I turn it off while running the motor will be suddenly much faster and will get stuck. Could this be solved by changing the state back to `moving` like this?

```
case Trigger::ledToggleRequest:
            Serial.println("LED");
            digitalWriteFast(LED_BUILTIN, !digitalReadFast(LED_BUILTIN));
            state = State::moving;
            break;
```

luni64, 2020-01-11:
I can not observe this. And no! Why would you change the state of the motor if you toggle an LED??

First trace is the step signal, second trace the led signal 
[image]

Zoomed in on first led transition: 
[image]

Zoomed in further: 
[image]

You should definitely check your hardware....

acastles91, 2020-01-13:
Thanks a lot again, I went through the my hardware and found the cause of the issue. It all works now, it's a great library. I have another question regarding other topics but I think it will be better to make another thread.

closed 2020-01-15 by luni64