luni64/TeensyStep #64: changing target location on the fly

dansoskin, 2020-03-03:
something weird is going on, 
if i change the target to a smaller value than the previous one, it works ok.
if i change the target to a bigger value than the previous one, it only reaches the first value.

if i change direction on the fly, it will do emergency brake, then go back to starting position, stops, after that it will start moving to the new location.

also changing max speed on the fly acting weird (step controller)

luni64, 2020-03-03:
You can not change target or speed or anything else while the motors are running. 
See #59 and some other closed issues

The key feature of TeensyStep is to move a bunch of motors synchronized. If you would change the motor parameters of one of the motors during a move you'd necessarily loose sync between the motors.

dansoskin, 2020-03-03:
thanks alot for the quick reply

closed 2020-03-03 by dansoskin