luni64/TeensyStep #65: Define a stepper array using dynamically changeable arrays

AlfaKeNTAvR, 2020-04-20:
Hace a nice day,

First of all, your project is awesome, it provides many opportunities for stepper control and is the best of its kind, I think. However, I am having an issue with my application. In my robotic arm, driven by 4 stepper motors I have 3 rotational joints and one linear. Two motors run at similar velocities, another one is slower and the linear one is much slower than others. And there are several modes for controlling these motors: simultanious movement and in order: rotational first, linear after. Talking about order of movement there are absolutely no problems, because I can simply run controller.move() two times. But considering simultanious movement I would like to form an array of those motors, whose Target is not 0, so they would not slow down the others. For instance, if the slowest rorational joint targetRel position is 0 these is no need to take in it in consideration for controller.move() and the other two can rotate at maximum velocity.

So, is there any possibility to use dynamically changeable arrays for motor array forming? Or maybe there is a way to make controller stop calculating velocities based on the parameters of the motor with 0 target?

I think this might be useful for other applications aswell.
Thanks in advance!

Best wishes, Nikita

luni64, 2020-04-20:
All motors passed to a rotate or move command will always be synchronized, i.e. they will reach their targets at the same time. Ich you want to run them unsynchronized but at the same time you can use more than one controller at the same time. Is this what you want to achieve?

AlfaKeNTAvR, 2020-04-20:
Not exactly, I would like to have an ability to form a stepper motor array dynamically based on current Target positions. So there is just one call of controller.moveAsync(stepper array), but the amount of actual stepper motors is different each time.
If motor target is not 0, then it is added to the array, if 0, not.

luni64, 2020-04-20:
Ok, let me see if I got it right:  You want to run an array of motors  synchronized. The motor array may change from call to call. Additionally, motors with zero target should be excluded from running, right?

AlfaKeNTAvR, 2020-04-20:
Yes, exactly!

luni64, 2020-04-20:
Good, you certainly can not do this directly (without changing the lib) but it should not be too hard to write a wrapper function. I'll do some tests and post it here later today.

AlfaKeNTAvR, 2020-04-20:
Thank you in advance!

luni64, 2020-04-20:
This is a bit tricky since the library is not prepared for such special things. However, here an example how you can do it.  There are two helper functions. 

- The `setNewTarget` function takes a motor and a target. After setting the target it checks if some condition is fulfilled. Here you can choose any condition you like. I chose to exclude motors which are already at the target. You can also exclude motors with target = 0 or whatever you like. 
- The `moveMotors` function takes 4 targets and moves all motors which fulfill the conditions 

The following example shows how to use this. It first moves the motors to positions 200, 600, 100, and 50. Then to 0, 600, 0 and 700. since m2 is already at 600 it will be excluded. After that it moves all motors to 0. Since m1 and m3 are already at 0 they are excluded. 

The sketch uses an intervaltimer to print out the current motor positions in the background. 

```c++
#include "TeensyStep.h"

StepControl ctrl1;
Stepper m1(0,1), m2(2,3), m3(4,5), m4(5,6);

void setup()
{
  // print out motor positions in the background every 100ms for easy testing
  (new IntervalTimer())->begin([] { Serial.printf("%4d %4d %4d %4d\n", m1.getPosition(), m2.getPosition(), m3.getPosition(), m4.getPosition()); }, 100'000);

  delay(150); 

  Serial.println("\nSTART -------------------");
  moveMotors(200, 600, 100, 50);

  delay(100);

  Serial.println("\n-------------------------");
  moveMotors(0, 600, 0, 700);            // ignore m2 (already at target)
  delay(100);

  Serial.println("\n-------------------------");
  moveMotors(0, 0, 0, 0);                // ignore m1 and m2 (already at target)

  Serial.println("DONE --------------------\n\n");
}

void loop()
{
}

//Helpers ===================================================================================

// sets target of passed motor and returns true if the motor should be used
bool setNewTarget(Stepper &motor, int target)
{
  motor.setTargetAbs(target);
  //return target != 0;                  // Change this to any condition you need for excluding the motor from movement
  return target != motor.getPosition();  // This is probably what you really want?
}

// moves motors to targets t1..t4. Doesn't move if the conditions of setNewTarget are not fullfilled.
void moveMotors(int t1, int t2, int t3, int t4)
{
  Stepper *motors[] = {nullptr, nullptr, nullptr, nullptr};

  int cnt = 0;
  if(setNewTarget(m1, t1)) motors[cnt++] = &m1;
  if(setNewTarget(m2, t2)) motors[cnt++] = &m2;
  if(setNewTarget(m3, t3)) motors[cnt++] = &m3;
  if(setNewTarget(m4, t4)) motors[cnt++] = &m4;

  switch (cnt)
  {
  case 1:
    ctrl1.move(*(motors[0]));
    break;
  case 2:
    ctrl1.move(*(motors[0]), *(motors[1]));
    break;
  case 3:
    ctrl1.move(*(motors[0]), *(motors[1]), *(motors[2]));
    break;
  case 4:
    ctrl1.move(m1, m2, m3, m4);
    break;
  }
}
```
Here the printout: 
```
   0    0    0    0

START -------------------
   2    7    1    0
  13   38    6    3
  31   93   15    7
  57  170   28   14
  83  250   41   20
 110  330   55   27
 137  410   68   34
 163  490   81   40
 184  552   92   46
 196  588   98   49
 200  599  100   50
 200  600  100   50

-------------------------
 194  600   97   69
 181  600   91  111
 161  600   81  178
 136  600   68  258
 112  600   56  338
  87  600   44  418
  62  600   31  498
  38  600   19  578
  17  600    9  644
   5  600    3  684
   0  600    0  699
   0  600    0  700

-------------------------
   0  588    0  686
   0  554    0  647
   0  500    0  584
   0  432    0  504
   0  363    0  424
   0  294    0  344
   0  226    0  264
   0  157    0  184
   0   90    0  105
   0   38    0   45
   0    8    0   10
DONE --------------------
```
If I find some time I try to implement a few things in the library making such things more easy to implement. 

Hope that helps

AlfaKeNTAvR, 2020-04-20:
Have a nice evening, thank you for your reply, I will test it and write you back!

AlfaKeNTAvR, 2020-04-20:
By the way, I was trying to do another thing, create a pointer array of 4, after that add 3 motors to it and no movement happened. So, you can't create an array of 4, initialize less elements and use this as an argument for .move() /.moveAsync function? There should be exact the same amount of motors as the size of the array, am I right?
If it was not so, things might go easier, I suppose

luni64, 2020-04-20:
I just reread your first post. I overlooked that you are using relative targets, in my example I used absolute targets. So, your condition target = 0 corresponds to my condition target == current position.  If you prefer relative targets, you can easily adapt that. 

> So, you can't create an array of 4, initialize less elements and use this as an argument for .move() /.moveAsync function?

No, it takes the size of the array and deduces the number of motors from it. While this is convenient usually it is not very flexible. I'll add a standard  move( Stepper*,  unsigned nrOfMotors) later.  
BTW: This is why I had to implement the switch as workaround in movemotors...

AlfaKeNTAvR, 2020-04-20:
> I just reread your first post. I overlooked that you are using relative targets, in my example I used absolute targets. So, your condition target = 0 corresponds to my condition target == current position. If you prefer relative targets, you can easily adapt that.

Well, that absolutely not a problem, I can easily adapt your code, no worries) 
Btw, your code is exactly the same idea I tried to realize yesterday! That means I was on the right way, but have some problems in c++, because I program on python most of the time...

> No, it takes the size of the array and deduces the number of motors from it. While this is convenient usually, it is not very flexible. I'll add a standard move( Stepper*, unsigned nrOfMotors) later.
BTW: This is why I had to implement the switch case block as workaround in movemotors...

I think, the problem was that I tried to use another array initialization method, never saw something like this: Stepper *motors[] = {nullptr, nullptr, nullptr, nullptr};
I used: Stepper *motors[4];
motors[0] = &motor_1; motors[1] = &motor_2; motors[2] = &motor_3; (No 4th one)
And it actually did not work. This was the moment where I stuck. Probably because of poor c++ experience

luni64, 2020-04-20:
> I used: Stepper *motors[4];

This generates an array of 4 pointers to Steppers but doesn't initialize them.  Since you didn't set the 4th there is some garbage in it which probably crashed your program. 

```c++
Stepper *motors[] = {nullptr, nullptr, nullptr, nullptr};
```
initializes the elements to nullptr. You could also do: 

```
Stepper* motors[] = {&motor_1, &motor_2, &motor_3}; 
```
To get an array of 3 initialized pointers. If you pass that to moveAsync it should work flawlessly.

AlfaKeNTAvR, 2020-04-20:
Finally, It works! Thank you so much for your great support!

AlfaKeNTAvR, 2020-04-22:
Hello again,

I've made several tests and it seems to not working correctly in situations when some of the targets are equal to 0. Positions are calculating as they should, however the actual movement of stepper motors is incorrect. I still can't find the problem and fully describe it to you, but it seems like the steppers are not working simultaneously (only one stepper moves and its actual position somehow is a combination of all other steppers targets) For instance, I have 2 steppers out of 4 whose target is not 0: stepper1 = 90 deg, stepper2 = 90, however, the actual positions of the steppers after the calling of .moveAsync(*motors[0], *motors[1]) are: stepper1 = 45, stepper2 = 0. I will try to make more test tomorrow and finally find any regularity.

luni64, 2020-04-23:
Can you provide a minimal example showing the effect? Ideally based on the example I posted above?

AlfaKeNTAvR, 2020-04-23:
Well, I understood, I thought I was making the same thing as you in your code, but actually I did not. Can you please explain to me what I am doing wrong? With your code it perfectly work, however, mine does not...
You code:
`bool setNewTarget(Stepper &motor, int target)
{
  motor.setTargetAbs(target);
  //return target != 0;                  // Change this to any condition you need for excluding the motor from movement
  return target != motor.getPosition();  // This is probably what you really want?
}

  if(setNewTarget(joint_1.stepper_motor, x_abs)) motors[stepper_motor_counter++] = &joint_1.stepper_motor;
  //if(setNewTarget(joint_2.stepper_motor, t2)) motors[cnt++] = &m2;
  if(setNewTarget(joint_3.stepper_motor, y_abs)) motors[stepper_motor_counter++] = &joint_3.stepper_motor;
  if(setNewTarget(joint_4.stepper_motor, alpha_abs)) motors[stepper_motor_counter++] = &joint_4.stepper_motor;

switch (stepper_motor_counter)
      {
      case 1:
        position_controller.move(*(motors[0]));
        break;
      case 2:
        position_controller.move(*(motors[0]), *(motors[1]));
        break;
      case 3:
        position_controller.move(*(motors[0]), *(motors[1]), *(motors[2]));
        break;
      case 4:
        position_controller.move(joint_1.stepper_motor, joint_3.stepper_motor, joint_4.stepper_motor);
        break;
      }`

My code:
`if(joint_1.get_position() != x_abs)
  {
    joint_1.set_target_abs(x_abs); //Actually calls setTargetAbs()) with a convertation from deg to steps
    motors[stepper_motor_counter++] = &joint_1.stepper_motor; 
  }
  
  if(joint_3.get_position() != y_abs)
  {
    joint_3.set_target_abs(y_abs);
    motors[stepper_motor_counter++] = &joint_3.stepper_motor; 
  }

  if(joint_4.get_position() != alpha_abs)
  {
    joint_4.set_target_abs(alpha_abs);
    motors[stepper_motor_counter++] = &joint_4.stepper_motor; 
  }

    if(stepper_motor_counter == 1) position_controller.move(*(motors[0]));
    else if(stepper_motor_counter == 2) position_controller.move(*(motors[0]), *(motors[0])); 
    else if(stepper_motor_counter == 3) position_controller.move(*(motors[0]), *(motors[1]), *(motors[2])); 
    else position_controller.move(joint_1.stepper_motor, joint_3.stepper_motor, joint_4.stepper_motor);`

I am just curious why does not my code work.
Thanks in advance!

luni64, 2020-04-23:
In your code for condition stepper_motor_counter == 2  the second motor in  move should be motors[1] not motors[0]

AlfaKeNTAvR, 2020-04-23:
Well, yes, sorry for that. That's totally my inattention...
Thank you for you support, that's all)

closed 2020-04-23 by AlfaKeNTAvR

luni64, 2020-04-23:
No problem, let me know if you need further support

AlfaKeNTAvR, 2020-04-24:
Sure, thanks! After finishing, I can provide you some photo/video materials about my project for your library portfolio, if you want. That is a 4-axis industrial Scara robot prototype for a cardbox facility)

luni64, 2020-04-24:
> I can provide you some photo/video materials about my project for your library portfolio, if you want.

Definitely, I'm always looking for nice examples for the web page.

AlfaKeNTAvR, 2020-04-24:
Hello again,

I know, that's not a question to you, but since you are constantly working with Teensy, maybe you can help? I've already posted several questions on official PJRC forum, but it seems like nothing helped yet... 
I was controlling my robot using Serial port, everything was okay, after that I decaided to continue programming and in 30 minutes tried to uploade the codeto Teensy, but it was no longer recognized by Windows. So, since early morning I am trying to make a connection, but with no results... Just get sometimes "unrecognized USB-device" and that's all. Have you ever had something like this?

luni64, 2020-04-25:
I read the forum post, not much I can add here other then recommend to install the tyTools. https://github.com/Koromix/tytools/release

Maybe the information from the WIKI https://github.com/TeensyUser/doc/wiki/FAQ helps. Have a look the USB-Serial Communication section. There is some information on how to delete the Ports (Windows screwed up my ports) which might help

AlfaKeNTAvR, 2020-04-27:
Yeah, thank you very much, I've tried everything, but unfurtunately there's something wrong with the hardware. Already ordered another Teensy, that's so sad