luni64/TeensyStep #66: acceleration/deceleration resets to zero.

JonaldJohnston, 2020-04-23:
im having a weird issue where the acceleration/deceleration stages constantly reset to zero for some reason. 
here's a quick sketch of the observed motion that i am seeing:
[image]
It seems to reset between 60000 and 70000 steps/s
Im using a Teensy 3.6
It happens on all pins i have tried, though some pin pairs don't work for some reason, like 2,3 or 7,8

luni64, 2020-04-23:
Can you post a minimal program showing this behavior?

AlfaKeNTAvR, 2020-04-23:
It looks like you have to try pulsePolarity setting of stepper motor. Test .setStepPinPolarity(HIGH);  and .setStepPinPolarity(LOW);

JonaldJohnston, 2020-04-23:
code: 
```
#include "TeensyStep.h"

Stepper motor(9,10);       // STEP pin: 2, DIR pin: 3
StepControl controller;    // Use default settings 

void setup()
{
  motor.setMaxSpeed(70000);
  motor.setAcceleration(20000);
  
}

void loop() 
{
  motor.setTargetRel(6400*100);  // Set target position to 1000 steps from current position
  controller.move(motor);    // Do the move
  delay(2000);
}
```
i tried AlfaKeNTAvR's suggestion and it doesn't change anything.

luni64, 2020-04-23:
This is super strange. I can reproduce the behaviour with this code: 
```c++
#include "Arduino.h"
#include "TeensyStep.h"

Stepper motor(9,10);
StepControl controller;

void printPos()
{
  Serial.printf("%d\t%d\t%d\n", millis(), motor.getPosition(), controller.getCurrentSpeed());
}

void setup()
{
  while(!Serial){}

  motor.setMaxSpeed(70000);
  motor.setAcceleration(20000);

  (new IntervalTimer())->begin(printPos, 100'000);

  motor.setTargetAbs(6400 * 100);
  controller.move(motor);
}

void loop()
{
}
```
A plot of the generated data gives this. 

[image]

I'll have a closer look at it tomorrow.

JonaldJohnston, 2020-04-23:
hey look, my crappy graph was accurate lmao! but yeah, super strange. glad to know its not my motor/driver.

luni64, 2020-05-02:
Should be fixed, can you try v2.0.1? 
Thanks for spotting that.

closed 2020-05-09 by luni64