luni64/TeensyStep #67: controller.overrideSpeed() negative numbers only changes one of the two motors.

300bar, 2020-04-27:
Hi,
I am experiencing a strange behavior I don´t see the reason why, I am using your library to make  a X,Y router table control and I am very happy so far, all manual moves are perfect, input is done through 2 encoders (X,Y) and a PID loop structure for both axes with 2 rotating controllers. All is running smooth and well. So next step is synchronizing  both axes movements, I am starting trying to move both:
        static boolean hecho = false;
        if (!hecho) { giroX.rotateAsync(mX,mY);hecho = true;}
        giroX.overrideSpeed(factorX); 
and what happens is that both axes move in sync in one direction but with negative factorX both should change direction and only one change the other remains moving in the same directions as having a positive factorX... 
I must be ignoring something but I can´t see what.
Thank you in advance for the help.

luni64, 2020-05-02:
Can you please post a minimal example showing the effect?

300bar, 2020-05-02:
> Can you please post a minimal and compiling example showing the effect?

Good morning! Thank you for answering. As your request I did write the most simple sketch and it works as it should!!  so as I was supposing I am missing or doing something wrong... probably it has to do with using the rotation objects wrongly in different parts of the sketch, which it is quite long now, so my question is if I use the rotation object, lets say with only one motor, in a function and I want to use the same rotation object in a different  function with two motors this time once the first function is over how I make sure it can be done with no interfering each other? Sorry if this very basic stuff!.

luni64, 2020-05-02:
It is quite common to have a global RotationController and use it in different places. Just make sure that you don't start a new movement while the controller is still moving other motors. Best to always check with .isRunning() before starting a new movement.

300bar, 2020-05-02:
Ok finally I have been able to isolate my problem, simple sketch to show it, basically it is all about initiating the motors with   mZ.setInverseRotation(true) then strange things happen:

#include "TeensyStep.h"
Stepper mX(31,32);
Stepper mZ(29,30);   
RotateControl giroX;
RotateControl giroZ;

void setup() {
    initMotors(); 
    ceroMaq();
}

void loop() {

}

void initMotors() {
    mZ.setMaxSpeed(6000);
    mZ.setAcceleration(20000);
    mZ.setInverseRotation(true);
    giroZ.rotateAsync (mZ);
    giroZ.overrideSpeed(0);
    mX.setMaxSpeed(6000);
    mX.setAcceleration(20000);
    mX.setInverseRotation(false);  
    giroX.rotateAsync(mX);
    giroX.overrideSpeed(0);     
}

void ceroMaq(){  
    const int tSalida = 300;
    const int tExitLim = 500;

    giroX.overrideSpeed(0.3);
    delay (tSalida);
    giroX.stop();
    mX.setPosition(0);
    
    giroX.overrideSpeed(-0.3);
    delay (tSalida);
    giroX.stop();
    mX.setPosition(0);
 
    giroZ.overrideSpeed(0.3);
    delay (tSalida);
    giroZ.stop();
    mZ.setPosition(0);   

    giroZ.overrideSpeed(-0.3);
    delay (tSalida);
    giroZ.stop();
    mZ.setPosition(0);   
} 

As you can see I initiate the motors, then ceroing the machine...  the two motors should move a little back and a little forward, the X axis do it ok but the Z axis only moves in one direction!!
I am using your last library and the board is a Teensy 3.6

I hope this will help.

luni64, 2020-05-02:
I can reproduce that. Looks like you found a bug. It is related to the inverse rotation you set for mZ. I need to dig deeper into that to see what root cause is. 

**Workaround:** Do not use the inverse rotation feature. You get the same effect by simply swapping the cables of one of the two phases of mZ.

Here an LA output of the generated signals without inverse rotation of mZ. Looks like it does what it is supposed to do:
[image]

**Please note** Even if it works without inverse rotation you have a conceptual bug in your code.  If you stop a motor you can not restart it by override the speed. This only works in the current library since this version doesn't release the grabbed timers. The next version will release the internal timers on a stop() so that your code will not work anymore. You will have to restart the controller after a stop().

300bar, 2020-05-02:
Thank you!
I take note of your comments. I really appreciate your help and your awesome library which is the best you can find.

luni64, 2020-06-29:
Added this to the to do list

closed 2020-06-29 by luni64