luni64/TeensyStep #68: Fix compilation error with newer gcc

noonien, 2020-04-28:
```
In file included from TeensyStep/src/timer/teensy3/TeensyStepFTM.h:2,
                 from TeensyStep/src/timer/teensy3/TeensyStepFTM.cpp:3:
TeensyStep/src/timer/teensy3/config.h:145:81: error: ‘reinterpret_cast<TeensyStepFTM::FTM_t*>(1073971200)’ is not a constant expression
  145 |     constexpr volatile FTM_t* timer = __builtin_constant_p((FTM_t*)timerAddr) ? (FTM_t*)timerAddr : (FTM_t*)timerAddr; // base address for register block of selected timer
      |                                                                                 ^~~~~~~~~~~~~~~~~
TeensyStep/src/timer/teensy3/config.h:149:36: error: non-constant condition for static assertion
  149 |     static_assert(timer != nullptr && irq != 0, "Board does not support choosen timer");  //Generate compiler error in case the board does not support the selected timer
```

Command I ran:

```
"arm-none-eabi-g++" "-O0" "-ffunction-sections" "-fdata-sections" "-g" "-fno-omit-frame-pointer" "-mthumb" "-march=armv7e-m" "-I" "TeensyStep/src" "-I" "teensy3-core" "-Wall" "-c" "-Wno-error=narrowing" "-fno-exceptions" "-fpermissive" "-felide-constructors" "-std=gnu++14" "-fno-rtti" "-fsingle-precision-constant" "-D__MK20DX256__" "-DTEENSYDUINO=147" "-DARDUINO=10809" "-DF_CPU=72000000" "-DUSB_SERIAL" "-DLAYOUT_US_ENGLISH" "-o" "/home/george/Projects/icu-ventilator/ICUVent/Controller/controller-firmware-v2/target/thumbv7em-none-eabi/debug/build/controller-firmware-3fef2048a3d33fc8/out/TeensyStep/src/timer/teensy3/TeensyStepFTM.o" "-c" "TeensyStep/src/timer/teensy3/TeensyStepFTM.cpp"
```

```
$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (GCC) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Running the exact same command with teensyduino's toolchain (gcc 5.4) works just fine. This might be a gcc regression, but, still, it is invalid c++14 code.

luni64, 2020-05-02:
Can you give the branch  `hotfix/GCC9_Compatibility` a try?

noonien, 2020-05-02:
I've tested it, the code seems to compile fine, however, there's a problem that causes a linker error: with these constexprs here:
<https://github.com/luni64/TeensyStep/blob/00100ecbaf38bc1c377935d8cdd8373b767a011a/src/Stepper.h#L8-L11>

Here's the linker error:
```
Stepper.o: in function `Stepper::setAcceleration(unsigned long)':
          TeensyStep/src/Stepper.cpp:62: undefined reference to `Stepper::aMax'
  Stepper.o: in function `Stepper::setMaxSpeed(long)':
          TeensyStep/src/Stepper.cpp:69: undefined reference to `Stepper::vMaxMax'
```

Replacing `static constexpr` with just `const` fixes the issue.

I think the problem is explained here: <https://stackoverflow.com/a/28846608>

I'm using -std=gnu++14 to compile, since it's the same one Arduino uses.

luni64, 2020-05-02:
Strange, works here, can you post a short sketch showing this?

noonien, 2020-05-02:
Sadly, I don't have a sketch. While I am using Arduino and teensy libraries/code, the project is now built using cargo/rust.

However, the relevant parts are built, and linked with gcc using gcc 9.2.

luni64, 2020-05-02:
Ok, seems to be related to some build system differences. Using a standard makefile here, also gnu++14. 

I assume a `static const` does not link as well?  Can't really test it since here everything works :-)

luni64, 2020-05-02:
Changed it to static const and did the definition in stepper.cpp. This is supposed to work I'd say...

noonien, 2020-05-02:
Perfect! Everything works just fine now!

Thank you!

luni64, 2020-05-02:
Merged into master. Thanks for spotting this

closed 2020-05-02 by luni64