luni64/TeensyStep #73: StepContol isRunning() method block

AlfaKeNTAvR, 2020-06-28:
Hey Luni,

After 2.1 realease one of my functions stopped working. It is the homing function and it is utilized for one-by-one homing sequence. After homing of one axis is finished it moves to its zero position using moveAsync(). MoveAsync() is used instead of move() because during axis movement the Emergency Button pin is observed, so ErrorRountine can be triggered if it is pushed. That is why I used special construction with isRunning(), which you can see in code below, however, in 2.1 it is not working anymore. The isRunning() always return True even after the end of the movement. I saw, you made some changes to this function, that now it should not return before the movement is finished, so there is definately something wrong with it.

```
#include "TeensyStep.h"

int enable_pin_4 = 8;

Stepper stepper_motor(31, 30);       // STEP pin: 2, DIR pin: 3
StepControl controller;    // Use default settings 

void setup()
{
  pinMode(31, OUTPUT);
  pinMode(30, OUTPUT);
  pinMode(29, OUTPUT);
  pinMode(28, OUTPUT);
  pinMode(27, OUTPUT);
  pinMode(26, OUTPUT);
  pinMode(25, OUTPUT);
  pinMode(24, OUTPUT);

  
  pinMode(enable_pin_4, OUTPUT);
  digitalWrite(enable_pin_4, HIGH);
  
  stepper_motor
            .setMaxSpeed(20800 * 4)
            .setAcceleration(10400 * 4)
            .setInverseRotation(true)
            .setStepPinPolarity(HIGH);
}

void loop()
{
  String mode = Serial.readStringUntil('_'); 
  
  if(mode == "2")
  {
    String position = Serial.readStringUntil('_'); 
    stepper_motor.setTargetAbs(position.toInt());  
    controller.moveAsync(stepper_motor);    // Do the move
    while(controller.isRunning())
    {
      // Returns if uncommented
      //Serial.println(controller.isRunning());
    }
  }
}
```

luni64, 2020-06-29:
Calling isRunning in a tight loop is quite unfriendly to the library. You'll check the status on a sub microsecond rate which generates a significant load. Just add a delay(1) in the while loop and you should be fine. 

```c++
while(controller.isRunning())
{
     delay(1);

     // Returns if uncommented
     //Serial.println(controller.isRunning()); 
}
```
Regardless of the obvious fix, the behavior is strange. Maybe some race condition. Investigating this further will take some time which I currently don't have. But I put it on the list of open issues.

luni64, 2020-06-29:
BTW: it is not necessary to set the pin mode of the DIR/STP pins. This is handled by TeensyStep automatically.

Mike118, 2020-11-15:
I am not sure that it is the same problem but I manage to have similar issue using the moveAsync ( and relative target )  with a frequent call to controller.isRunning()... 
I somehow manage to get a strange state where my program is not blocked in a loop, but the motor stops to move and stepperMotors.getPosition() return a changing value as if it was moving the motor. But the controller seems to never manage to get the target position ... 
Interresting fact, when I am in this state I can change the target position and manage to have stepperMotors.getPosition() changing from increasing to decreasing ...  But the gpio are not modifier and the controller  seems to never manage to " go to the requested position " even if it obviously manage to go over the requested position according with the stepperMotors.getPosition()  result ...
Unfortunately, I don't have an " easy to reproduce " case to share yet but if I manage to find a way to easy reproduce this behavior I will share it. Despite this problem I encounter, this library is a great job luni64! 

Not sure that it can help but this is the function I call in my loop  using the isRunning method call: 

`void checkStepperController() {  
  static uint32_t oldSpeed[NBSTEPPERS] = {0};
  for (uint8_t i = 0; i < NBSTEPPERS; i++) {
    if (controller[i].isRunning()) {
      uint32_t motorSpeed = controller[i].getCurrentSpeed();
      if (!constantSpeed[i]) {
        if ( motorSpeed == maxspeed[i]) {
          constantSpeed[i] = true;
          // Display that constant speed is reach
        }
      }
      if (!decelSpeed[i]) {
        if ( motorSpeed < oldSpeed[i]) {
          decelSpeed[i] = true;
          // Display that deceleration started
        }
        oldSpeed[i] = motorSpeed;
      }
    } else {
     if( !arret[i]) {
      arret[i] = true;
      constantSpeed[i] = false;
      decelSpeed[i] = false;
      oldSpeed[i] = 0;
      // display that the motor have stopped
     }
    }
  }
}`

Is there something wrong in there that may cause the problem?
(BTW is there a simpler and more precise way to check whether the motor is at a constant speed or in the deceleration phase?)

I also read that we can't change a motor target while moving, I try to not do that but maybe my problem can come from there if I did not handle well this case ...

closed 2021-11-24 by luni64