luni64/TeensyStep #74: Teensy4 define predicate does not work on release boards

CZEMacLeod, 2020-07-16:
I believe the beta version of the Teensy4 used an IMXRT1052, but the release version is based on an IMXRT1062 (and this is what is defined in boards in the latest teensyduino install).
As such I think the predicate  in [TeensyStep.h](https://github.com/luni64/TeensyStep/blob/06c0eaba91d79a800a3b5e9fced64e21b0d75b6c/src/TeensyStep.h#L20) needs to be changed to
```
// TEENSY 4 ================================================================================================

#elif defined(__IMXRT1052__) || defined(__IMXRT1062__)
#include "timer/teensy4/TimerField.h"
```

luni64, 2020-07-16:
That's true, but it won't help you since TeensyStep currently doesn't support Teensy 4.x.  Sorry, this is all long overdue but I need to find a time slot to work on it which currently is quite difficult.

raghukolla, 2020-12-28:
I'm trying to understand why does the code written for 3.5 not work for 4.0. One reason I could think of is that if the ports and registers are different in both the boards and the code directly accesses them then it would not work. Are there any other reasons?
Do you have any pointers for converting an arduino code to work with teensy or a teensy 3.5 code to teensy 4.0.
Your response would be super helpful. thanks

luni64, 2021-01-02:
Sorry, didn't see your post. 

>I'm trying to understand why does the code written for 3.5 not work for 4.0. One reason I could think of is that if the ports and registers are different in both the boards and the code directly accesses them then it would not work. 

T4.0 has a completely different set of hardware timers. That needs to be ported but I didn't find time yet. Anyway, the T4.0 peripherals are quite slow, so the performance will not be as good as with the T3.6. 

>Do you have any pointers for converting an arduino code to work with teensy or a teensy 3.5 code to teensy 4.0.
Your response would be super helpful. thanks

As long as your Arduino code doesn't directly access hardware it usually runs out of the box. There also are a lot of dedicated Teensy libraries. 
 
Hope that helps

LegacyMecha, 2021-01-05:
Should actually be "defined(ARDUINO_TEENSY40)" maybe?

closed 2021-01-14 by luni64