luni64/TeensyStep #76: Using RotateControl and StepControl objects "simultaneously" for same motors

drewhamiltonasdf, 2020-09-28:
Hi Luni,

I really appreciate your work. I've been having quite a bit of fun noodling around with your library to get a 6-axis robot arm I've been working on to move around for me with little monetary investment.
   
I have, however, run into issues trying to use both RotateControl and StepControl interchangeably in the same program with the same axes. I was using RotateControl to manually jog the axes around with very basic PID type control that can track changes in the target position, and then logging various waypoints based on serial commands. This was working decently, so I tried to use a globally defined StepControl object to move all 6 axes between the waypoints. Unfortunately, StepControl doesn't seem to work very well if there is already a RotateControl already initialized.

I just wanted to run this by you to see if this was a known issue, or if there are any workarounds. I am currently initializing RotateControl and StepControl objects globally.

When I use StepControl after RotateControl, the axes will **only move in a positive direction**. 

I'll post some more extensive code if this sounds outlandish, but here's a snippet:
-----------------------------
IN MAIN LOOP:
Do some manual jogging with 6 rotary encoder knobs. Works ok.
Then, if serial gets a command, say "MOVETOPOS1", run below code:
-----------------------------

            rotate_controller.stop();
            delay(100);
            rotate_controller.emergencyStop();
      
            int moveStep = -(64*8);         //<--- This line rotates the same direction, no matter the sign (+/-)
            
            JT[5]->m->setTargetRel(moveStep);     //<---Same effect if I use setTargetAbs()
            JT[5]->m->setMaxSpeed(9000);       
            JT[5]->m->setAcceleration(400);

            step_controller.move(*JT[5]->m);
            step_controller.emergencyStop();
            rotate_controller.overrideSpeed(0);

-Return to Main Loop for Manual Jogging

If it wasn't clear from the above, I had put together an array of class "Joint", each with a Stepper pointer. Here, I was keeping it simple and just trying to move Joint 5.

Any help would be greatly appreciated.

luni64, 2020-09-28:
Step and rotate controller should work together without problems. 

Can you generate a small sketch which I can compile to show the effect? Ideally the sketch should not require any additional hardware.  (Well, I can attach a pot for jogging or something like that...)

drewhamiltonasdf, 2020-09-28:
This is all a bit of a hot mess at this point. I've just been noodling around, so it's pretty dirty. Forgive me.

[Quick_Robo_3__ROTATE_CONTROL_all_six.zip](https://github.com/luni64/TeensyStep/files/5290274/Quick_Robo_3__ROTATE_CONTROL_all_six.zip)

drewhamiltonasdf, 2020-09-28:
[EDIT: *.ino slightly cleaned up]

This is all a bit of a hot mess at this point. I've just been noodling around, so it's pretty dirty. Forgive me. There is some garbage in there from my previous attempts at capturing positions. After some frustration I just put it all in serialEvent() when user enters "ZZZ".

[Quick_Robo_3__ROTATE_CONTROL_all_six_For_Luni.zip](https://github.com/luni64/TeensyStep/files/5290366/Quick_Robo_3__ROTATE_CONTROL_all_six_For_Luni.zip)

luni64, 2020-09-28:
Thanks, I'll have a look later today

drewhamiltonasdf, 2020-09-28:
I spent a remarkable number of hours working on this, trying various things, and continued working on it after I submitted this ticket. Just now I tried moving setMaxSpeed and setAcceleration before setTarget and the problems seem to vanish.

before I had:
            Joint[5]->setTargetRel(moveStep);
            Joint[5]->setMaxSpeed(9000);       
            Joint[5]->setAcceleration(200);
            step_control.move(*Joint[5]);

now I have:
            Joint[5]->setTargetRel(moveStep);
            Joint[5]->setMaxSpeed(9000);       
            Joint[5]->setAcceleration(200);
            step_control.move(*Joint[5]);

luni64, 2020-09-28:
Ok, the explanation is quite simple. `setMaxSpeed` sets the direction pin according to the sign of the speed parameter. This is because the sign of the speed parameter determines the rotation direction for the rotateControllers. So, calling `setMaxSpeed(...)` after  `setTargetRel(...)` will overwrite the direction originally set by `setTargetRel(...)` . 

-> This could be changed of course by  storing the direction and use the correct one depending on the type of the controller. I'll put it on the list of enhancements. But the workaround is rather simple so it doesn't seem urgent right? 

Here a few comments on your code which I made while trying to understand what you do: 

```c++
Stepper m0(0, 1);
Stepper m1(2, 3);
Stepper m2(4, 5);
Stepper m3(6, 7);
Stepper m4(8, 9);
Stepper m5(10, 11);
//                         motor   knobPin1    knobPin2
Joint* JT[6] = { new Joint(m0,     22,         23),
                 new Joint(m1,     20,         21),
                 new Joint(m2,     18,         19),
                 new Joint(m3,     16,         17),
                 new Joint(m4,     14,         15),
                 new Joint(m5,     39,         13)
                 };
```
I think it would be much more elegant to encapsulate the stepper in your Joint class.  That would get rid of the ugly global Stepper objects. 

```c++
Joint::Joint(Stepper &motor, int knobPin1, int knobPin2)
{
	m = &motor;  // 
        ...  
}
```
Why passing a reference to the stepper and then store a pointer to it? I'd simply store the reference but thats a matter of taste. As mentioned above, best would be to not pass in the stepper at all but encapsulate it in the class...

```c++
  noInterrupts();
  difference = JT[active_jog_joint]->encoderPosCount * 8 - JT[active_jog_joint]->m->getPosition();
  Input = JT[active_jog_joint]->m->getPosition();
  Setpoint = JT[active_jog_joint]->encoderPosCount * 8;

  myPID.SetTunings(2, 0, 0); //Kp, Ki, Kd
  myPID.Compute();
  controller.overrideSpeed(float(Output / 10000));

  if (difference == 0)
  {
    controller.overrideSpeed(0);
  }
  interrupts();
```
I don't see any reason to disable interrupts here? Disabling interrupts for such a long time might interfere with the pulse generation. 

It would be much nicer to integrate the complete PID code to the Joint class. 

Best
Lutz

drewhamiltonasdf, 2020-09-28:
These are great comments. Thanks. Sometimes I throw a noInterrupt in there if I'm having trouble with something. It's a bad habit. I also appreciate the comment about storing the reference and encapsulating the stepper in the Joint class. This is one of those "a little knowledge is a dangerous thing" kind of things. 

I agree that this is a relatively low priority fix, but it's good to have some record of my issue on GitHub/forums so other mis-fortunate souls can find searchable answers if they end up stuck on the same issue.

Thanks for the quick replies. I'll likely be back in some form or another when I try my hand at writing some other custom arbitrary function accelerators.

Thanks for this fantastic library!

Best,
Drew

closed 2020-10-05 by luni64