luni64/TeensyStep #78: Explanation wanted

TadyTheFish, 2020-10-06:
Hello
Can someone explain how this library handles timers?
In my project I used 1x StepControl and 1x RotateControl and TimerOne.h.
Then problems struck when I wanted to use PWM on pins 21, 22 ,23. These pins are driven with FTM0. After hours of digging I fount that I can assign which timer to use. I used `#define USE_TIMER TIMER_FTM3` (I'm using Teensy 3.6) And the PWM pins started to work.
But I tought that RotateControl and StepControl each use one timer and the definition is only for one timer?? Which is the other timer or did I not understand something correctly?

Thank you

luni64, 2020-10-07:
- Here the documentation regarding the used timers: https://luni64.github.io/TeensyStep. Open 'Home', scroll down to the chapter "Used ressources". 
- And here some info about how the used FTM module influences PMW pins  #51

> But I tought that RotateControl and StepControl each use one timer and the definition is only for one timer?? Which is the other timer or did I not understand something correctly?

Each controller uses one PIT (aka IntervalTimer) and two FTM *channels*. In the config you choose which FTM *module* (not channel) the library uses. If you use two controllers you need to make sure that you choose an FTM module with at least 4 channels which is FTM0 or FTM3 for a T3.6.  

> In my project I used 1x StepControl and 1x RotateControl and TimerOne.h.

Not related to your issue but TimerOne is only there to make some old Arduino libraries work with Teensy. I'd not use it for Teensy code. If you only have two controllers you have two intervaltimers left which might be a better option than TimerOne.  You can also have a look at the TeensyTimerTool: https://github.com/luni64/TeensyTimerTool

TadyTheFish, 2020-10-07:
Thank you so much!
I will change the code to use your library
Thank you

closed 2020-10-07 by TadyTheFish