luni64/TeensyStep #81: stm32f4 and teensy3.5 pulse comparison

ramboerik, 2020-10-08:
Hi @luni64 ,

I used an external trigger signal to both my boards (NUCLEO-F429ZI and teensy3.5) to trigger the stepper move command at the exact same time and observed a difference in the  generated pulses. It seems like the pulses are genereated 10ms after the trigger for the teensy board. Is that correct behavior? My implementation for the stm32f4 boards starts to generate the pulses as soon as the call to the move command is issued.

Here is a recording of 10 pulses with acc/dec:
[image]

luni64, 2020-10-08:
I remember somebody having issues when starting a second move right after the first one finished. This led to the problem that the first puls of the second move immediately followed the last pulse of the first move which the motor of course couldn't follow. 

I think (need to have a look later today) that I inserted a delay corresponding to the pull in speed at the start of the movement to avoid this. Can you do the same measurement using adjacent  moves?

ramboerik, 2020-10-08:
Here you go:
[image]
That 10ms delay between moves seem to be true for teensy and then I at least know why the pulses differs

closed 2021-09-28 by luni64