luni64/TeensyStep #84: STM32F411 Compile errors on not declared TIM6/TIM7/TIM8/TIM12/TIM14

Steamerzone, 2020-11-03:
Hello,

I get the following compile errors on a STM32F411 target, this device only has 7 general purpose timers.
For now I just removed the affected timers, haven't tested library on the STM32F411 yet.

lib\TeensyStep\src\timer\stm32\TimerField.cpp:4:80: error: 'TIM6' was not declared in this scope; did you mean 'TIM1'?
    4 | TIM_TypeDef* TimerField::timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM6, TIM7, TIM8, TIM9, TIM10, TIM11, TIM12, TIM14 };
 
lib\TeensyStep\src\timer\stm32\TimerField.cpp:4:86: error: 'TIM7' was not declared in this scope; did you mean 'TIM1'?
lib\TeensyStep\src\timer\stm32\TimerField.cpp:4:92: error: 'TIM8' was not declared in this scope; did you mean 'TIM1'?
lib\TeensyStep\src\timer\stm32\TimerField.cpp:4:118: error: 'TIM12' was not declared in this scope; did you mean 'TIM1'?                                                                                                 
lib\TeensyStep\src\timer\stm32\TimerField.cpp:4:125: error: 'TIM14' was not declared in this scope; did you mean 'TIM1'?

Greetings,
Emile

luni64, 2020-11-04:
Actually, the STM compatibility was done by @ramboerik. I'm not familiar with those processors at all. Maybe he can answer your questions.

ramboerik, 2020-11-04:
Hi @Steamerzone ,

The stm32f4 port has only been verified for the stm32f429i board and my assumption was that the whole f4 serie had the same number of interrupts, which clearly isn't the case. I will have a look at it later, thanks 😃 

Your temporary fix should be fine!

closed 2020-11-13 by luni64