luni64/TeensyStep #85: stm32f4 missing/delayed pulses when accelerating / stepper motor sounding rough

Steamerzone, 2020-11-07:
Hi @ramboerik,

I'm randomly having problems with missing/delayed pulses when accelerating, and the stepper is sounding rough when it encounters those issues. Some acceleration runs run better than others with the same settings, so it seems random.

Any idea what it could be? 

[image]

[image]

ramboerik, 2020-11-08:
Hi @Steamerzone !

There is an open pull request with a change that might solve your issue, https://github.com/luni64/TeensyStep/pull/82/files.  
The line you want to add:
```diff
bool TimerField::begin()
{
+ pulseTimer.setPreloadEnable(false);
  return true;
}
```

Steamerzone, 2020-11-08:
Hi @ramboerik,

I already had the line ''pulseTimer.setPreloadEnable(false);''  in bool TimerField::begin().
Played around a bit, and if I remove the step timer preload, the missing/delayed pulses are gone.

bool TimerField::begin()
{
  //stepTimer.setPreloadEnable(false);
  accTimer.setPreloadEnable(false);
  pulseTimer.setPreloadEnable(false);
  return true;
}

ramboerik, 2020-11-09:
I'm glad it works now 😃 

There are currently two issues (that I'm aware of) with the stm32 code, one is the one you encountered and the other one is that the  `endAfterPulse` isn't implemented. The `endAfterPulse` might cause the stepper to step a few extra steps depending on the timing with the acceleration interrupt and the step interrupt. The code is aware of the extra steps and will take them into account for the next movement but it should be fixed.

@luni64 have you tested https://github.com/luni64/TeensyStep/pull/82 😄 ? Apart from the new feature it also contains a important fix for @Steamerzone issue.

I will provide a pull request with a fix for  `endAfterPulse`  when it's merged.

luni64, 2020-11-10:
> @luni64 have you tested #82 😄 ? Apart from the new feature it also contains a important fix for @Steamerzone issue.
Sorry, actually, I'm a bit under water currently. Too much projects in parallel... 

I just created a STM32 branch where I can merge into whatever you want. I hope that I find some time later this year to test everything finally merge it into master. Would that work for you?

ramboerik, 2020-11-10:
Haha, that sounds great 😃 

I guess we can close this issue then as the fix is to use the stm32 branch for now

closed 2020-11-13 by luni64